eruption-project/eruption #1: Service doesn't start after reboot.

Aimen06, 2019-07-30:
Hi,
I have managed to compile the project without error.
But nothing happen after reboot.
When i type systemctl start eruption.service i get Failed to start eruption.service: Unit eruption.service not found.
I use debian unstable.
Do you know what is the problem ?
Thx

X3n0m0rph59, 2019-07-30:
Thanks for your bug report!

Seems like systemd did not find the unit file of the eruption service.
The file `support/systemd/eruption.service` has to be copied from the project's directory to the systemd unit directory `/lib/systemd/system/`.

I'm going to prepare packages for Debian very soon! Then it will be much easier for you to deploy.

Aimen06, 2019-07-31:
Thanks for the reply i will try :)

closed 2020-05-08 by X3n0m0rph59