eruption-project/eruption #4: Support QWERTZ layout

poldi1405, 2020-05-07:
Hello,
my Keyboard is in the german QWERTZ layout and as soon as I start eruption it is set to QWERTY, making working with it a lot harder.

X3n0m0rph59, 2020-05-07:
This could be a problem with the eruption virtual keyboard device being set to the default layout.

Eruption uses the Linux `uinput` subsystem to create a virtual keyboard device that mirrors all keys pressed on the "real" device and additionally can inject/replace keystroke sequences ("macros").

I suspect that the layout of the virtual keyboard is set to "en_US".

Could you please post the output of these commands:

`
$ localectl
`

`
$ xinput list
`

Does this repair the problem (temporarily)?

`
$ setxkbmap de
`

poldi1405, 2020-05-08:
```
$ localectl
   System Locale: LANG=de_DE.UTF-8
       VC Keymap: de-latin1
      X11 Layout: n/a
$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ROCCAT ROCCAT Nyth Mouse                	id=15	[slave  pointer  (2)]
⎜   ↳ ROCCAT ROCCAT Nyth Consumer Control     	id=16	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ ROCCAT ROCCAT Vulcan AIMO               	id=8	[slave  keyboard (3)]
    ↳ ROCCAT ROCCAT Vulcan AIMO               	id=12	[slave  keyboard (3)]
    ↳ Conexant Sennheiser Main Audio          	id=13	[slave  keyboard (3)]
    ↳ Conexant Sennheiser Main Audio Consumer Control	id=14	[slave  keyboard (3)]
    ↳ ROCCAT ROCCAT Nyth System Control       	id=17	[slave  keyboard (3)]
    ↳ ROCCAT ROCCAT Nyth                      	id=18	[slave  keyboard (3)]
    ↳ ROCCAT ROCCAT Nyth Consumer Control     	id=21	[slave  keyboard (3)]
    ↳ Eruption Virtual Keyboard               	id=9	[slave  keyboard (3)]
```

and yes, `setxkbmap` fixes the issue until eruption is restarted

X3n0m0rph59, 2020-05-08:
Hello, 

```
$ localectl
   System Locale: LANG=de_DE.UTF-8
       VC Keymap: de-latin1
      X11 Layout: n/a
```

I suspect that this is the culprit: X11 Layout: **n/a**

To set the system-wide default keyboard layout to `de`, you may want to execute this command:
```
$ sudo localectl set-x11-keymap de
```

You may need to restart at least the X server for the changes to take effect.

poldi1405, 2020-05-08:
That did it. Thanks a lot.

closed 2020-05-08 by poldi1405