eruption-project/eruption #5: Only AIMO default lighting effects are applied

poldi1405, 2020-05-07:
Hello,
I installed eruption using the AUR and none of the effects is applied. No matter whether I'm running it with `sudo eruption impact.lua` or `sudo eruption -c /etc/eruption/eruption.conf` with the following eruption.conf:
```
[global]
profile_dir = "/var/lib/eruption/profiles/"
profile = "fireworks"

script_dir = "/usr/share/eruption/scripts/"
script_files = ["organic.lua"]

# select your keyboard variant
keyboard_variant = "ISO"

[frontend]
enabled = true
theme = "eruption"
```

Same when starting with systemd (basically same as the second command.

poldi1405, 2020-05-07:
Have now tried it with `sudo eruption -p fireworks` and `sudo eruption -p /var/lib/eruption/profiles/fireworks.profile`

Without success

X3n0m0rph59, 2020-05-07:
Hello, thanks for your bug report!

It seems that the eruption daemon is unable to initialize your device. Could you please try to update your keyboard's firmware to the latest revision using ROCCAT Swarm?

Could you also please show some log output of eruption?

You may want to run eruption with additional output enabled:

`sudo RUST_LOG=debug RUST_BACKTRACE=full eruption -c /etc/eruption/eruption.conf`

poldi1405, 2020-05-07:
The firmware was updated just yesterday so I just assume that it is still the most recent version (I don't have a Windows machine here, so no way of checking).

Log:
```
 INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.5
 DEBUG eruption > Loading saved state...
 INFO  eruption > Loaded profile: Default
 INFO  eruption::rvdevice > Found Control interface: "0001:0003:01": ROCCAT Vulcan AIMO
 INFO  eruption::rvdevice > Found LED interface: "0001:0003:03": ROCCAT Vulcan AIMO
 INFO  eruption           > Opening devices...
 INFO  eruption           > Initializing devices...
 INFO  eruption           > Configuring LEDs...
 INFO  eruption           > Initializing D-Bus API...
 INFO  eruption           > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption                 > Spawning input thread...
 INFO  eruption                 > Loading Lua scripts...
 DEBUG eruption                 > Registering filesystem watcher...
 DEBUG eruption::dbus_interface > Message: Message { Type: Signal, Path: "/org/freedesktop/DBus", Interface: "org.freedesktop.DBus", Member: "NameAcquired", Sender: "org.freedesktop.DBus", Destination: ":1.797", Serial: 2, Args: [":1.797"] }
 DEBUG eruption::dbus_interface > Message: Message { Type: Signal, Path: "/org/freedesktop/DBus", Interface: "org.freedesktop.DBus", Member: "NameAcquired", Sender: "org.freedesktop.DBus", Destination: ":1.797", Serial: 3, Args: ["org.eruption"] }
 INFO  eruption::plugins::keyboard > Now listening on: /dev/input/event2
 INFO  eruption::plugins::keyboard > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
 INFO  eruption::plugins::keyboard > Driver version: 10001
 INFO  eruption::plugins::keyboard > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
 INFO  eruption::plugins::keyboard > Physical location: usb-0000:01:00.0-6/input0
 INFO  eruption::plugins::keyboard > Grabbing the device exclusively
 DEBUG eruption::scripting::script > Applying parameter Float { name: "animation_delay", description: "Step animation every [n] frames", default: 1.0 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "time_scale", description: "Scale of time axis (3rd coordinate of the noise function)", default: 250.0 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "coord_scale", description: "Scale of coordinates (1st and 2nd coordinates of the noise function)", default: 15.0 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "color_saturation", description: "Saturation in the range 0.0 .. 1.0", default: 1.0 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "color_lightness", description: "Lightness in the range 0.0 .. 0.5; above 0.5 == whiten color", default: 0.5 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "color_divisor", description: "Divide resulting hue angle by color_divisor. Final value will be mapped to the interval [0..360)", default: 1.0 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "color_offset", description: "Offset resulting hue angle by color_offset. Final value will be mapped to the interval [0..360)", default: 0.0 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "opacity", description: "Opacity value in the range 0.0 .. 1.0, where 1.0 is fully opaque", default: 1.0 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "afterglow_step", description: "Step afterglow animation every [n] frames", default: 2.0 }
 DEBUG eruption::scripting::script > Applying parameter Color { name: "color_afterglow", description: "Afterglow color", default: 4294901760 }
 DEBUG eruption::scripting::script > Applying parameter Color { name: "color_step_afterglow", description: "Color value that gets subtracted from afterglow every [n] frames", default: 252313600 }
 DEBUG eruption::scripting::script > Applying parameter Color { name: "color_shockwave", description: "Color of the shockwave", default: 4294901760 }
 DEBUG eruption::scripting::script > Applying parameter Color { name: "color_step_shockwave", description: "Color value that gets subtracted from the shockwave every [n] frames", default: 168427520 }
 DEBUG eruption::scripting::script > Applying parameter Float { name: "impact_step", description: "Step impact animation every [n] frames", default: 1.0 }
 DEBUG eruption::scripting::script > Applying parameter Color { name: "color_impact", description: "Color of impact effect", default: 4294901760 }
 DEBUG eruption::scripting::script > Applying parameter Int { name: "alpha_step_impact", description: "Value that gets subtracted from the impact effects alpha channel every n frames", default: 5 }
 DEBUG eruption                    > FPS: 29
 DEBUG eruption                    > FPS: 30
[... cut ...]
 DEBUG eruption::plugins::keyboard > Key event: InputEvent { time: TimeVal { tv_sec: 1588875857, tv_usec: 743961 }, event_type: EV_KEY, event_code: EV_KEY(KEY_C), value: 0 }
 DEBUG eruption::plugins::keyboard > Key event: InputEvent { time: TimeVal { tv_sec: 1588875857, tv_usec: 743961 }, event_type: EV_SYN, event_code: EV_SYN(SYN_REPORT), value: 0 }
 INFO  eruption                    > Closing devices...
 DEBUG eruption                    > Saving state...
 INFO  eruption                    > Exiting now
```

X3n0m0rph59, 2020-05-07:
Hello, thank you for showing the debug logs. 

So apparently the initialization of the device is working as expected, and therefore there is no problem with the firmware revision or the hardware itself.

I was able reproduce a problem with the profile selection code, and can confirm that the command line handling is currently broken in the daemon proper. The state file currently overrides all other ways to set a profile. This clearly is a bug. Until this is fixed you may may want to edit the file `/var/lib/eruption/eruption.state` to set the profile before starting the daemon.
```
$ sudoedit /var/lib/eruption/eruption.state
```

```
profile = 'profile1.profile'
enable_sfx = false
brightness = 100
```

When the daemon is running you may want to use this command to switch profiles at runtime:
```
$ dbus-send --print-reply --system --dest=org.eruption /org/eruption/profile org.eruption.Profile.SwitchProfile string:"profile1.profile"
```
This will switch to `profile1.profile` at runtime.

If you use GNOME 3 you may want to use this extension:
https://extensions.gnome.org/extension/2621/eruption-profile-switcher/

X3n0m0rph59, 2020-05-08:
I just pushed eruption version `0.1.7` which fixes the bug in the profile selection code.

It should now be possible to override all other ways to set a profile, by appending `-p` or  `--profile` on the command line.

Please let me know if this fixes the problem.

X3n0m0rph59, 2020-05-08:
P.s.:
The effect you were seeing the whole time is specified in the `default.profile`. This is not to be confused with the keyboard's built in effects (the hardware based rainbow gradient).

The so called "AIMO" effect is only implemented in the ROCCAT Swarm Windows driver.

> INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.5
 DEBUG eruption > Loading saved state...
 **INFO  eruption > Loaded profile: Default**
 INFO  eruption::rvdevice > Found Control interface: "0001:0003:01": ROCCAT Vulcan AIMO
 INFO  eruption::rvdevice > Found LED interface: "0001:0003:03": ROCCAT Vulcan AIMO
 INFO  eruption           > Opening devices...

poldi1405, 2020-05-10:
Yes, It works now… Thank you very much. Now for tikering together a theme…

closed 2020-05-10 by poldi1405