eruption-project/eruption #6: unable to automatically compile under Arch

poldi1405, 2020-05-19:
Due to the sysinfo dependency requiring a development-version the PKGBUILD is currently not able to build the package automatically. Now for the question: Should I modify the PKGBUILD and open a pull request (not very elegant as this would probably modify the users rust environment) or is there anything you can do?

X3n0m0rph59, 2020-05-19:
Hi, thank you for your report! 

Which version of `cargo` and `rustc` do you have installed? The current stable version of rust is able to successfully compile Eruption.

```sh
$ cargo --version
cargo 1.43.0 (2cbe9048e 2020-05-03)
```

```sh
$ rustc --version
rustc 1.43.1 (8d69840ab 2020-05-04)
```
I think you may be using rust nightly, which is currently broken wrt. compiling `sysinfo`...

Could you please try to switch to rust stable by running the following command:

```sh
$ rustup default stable
```
and then, please try to compile Eruption again!

poldi1405, 2020-05-19:
It seems to work now, thanks.

closed 2020-05-19 by poldi1405