eruption-project/eruption #7: Problems Running from Source on Ubuntu 18.04

erockce, 2020-06-18:
Hello.

I've not had success using the program with the sample configurations after compiling from source.

I compiled from source because the PPA doesn't work with Bionic (I wasn't ready to upgrade yet to the next distro version).

After a quick crash course with `cargo` and Rust, I got all the dependencies and the `eruption` program compiled.  But executing fails and kills the lighting effects. Only resolution is to "power cycle" the keyboard by un/re-plugging in.

Here is information on my set up.

These are my Roccat software and firmware versions (used through a Windows 10 guest via VirtualBox 6.1
```
Roccat Swarm Version: v1.9368
Roccat Driver Version: v1.36
Roccat Vulcan AIMO 120, Firmware Version: v1.0029
```

Additional information:
```
[ 1159.165357] input: ROCCAT ROCCAT Vulcan AIMO as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/0003:1E7D:3098.0007/input/input20
[ 1159.224438] hid-generic 0003:1E7D:3098.0007: input,hidraw0: USB HID v1.11 Keyboard [ROCCAT ROCCAT Vulcan AIMO] on usb-0000:00:14.0-1/input0
[ 1159.225781] input: ROCCAT ROCCAT Vulcan AIMO Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.1/0003:1E7D:3098.0008/input/input21
[ 1159.284181] input: ROCCAT ROCCAT Vulcan AIMO Consumer Control as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.1/0003:1E7D:3098.0008/input/input22
[ 1159.284257] input: ROCCAT ROCCAT Vulcan AIMO System Control as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.1/0003:1E7D:3098.0008/input/input23
[ 1159.284425] hid-generic 0003:1E7D:3098.0008: input,hiddev0,hidraw1: USB HID v1.11 Mouse [ROCCAT ROCCAT Vulcan AIMO] on usb-0000:00:14.0-1/input1
[ 1159.287780] input: ROCCAT ROCCAT Vulcan AIMO as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/0003:1E7D:3098.0009/input/input25
[ 1159.344151] hid-generic 0003:1E7D:3098.0009: input,hidraw2: USB HID v1.11 Keypad [ROCCAT ROCCAT Vulcan AIMO] on usb-0000:00:14.0-1/input2
[ 1159.344604] input: ROCCAT ROCCAT Vulcan AIMO as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.3/0003:1E7D:3098.000A/input/input26
[ 1159.404426] hid-generic 0003:1E7D:3098.000A: input,hidraw3: USB HID v1.11 Device [ROCCAT ROCCAT Vulcan AIMO] on usb-0000:00:14.0-1/input3
```

Distro Information:
```
erockce@nuc:~$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.4 LTS"
```

Tried executing both as user and using `sudo`. My user account is a member of the relative groups `usb` and `plugdev`.

Build Information:
```
erockce@nuc:~$ cargo -V
cargo 1.41.0
rustc 1.41.0
```

Executed with command (with ideas from [issue #5](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/issues/5#issuecomment-625202682)):
```
$ sudo RUST_LOG=full RUST_BACKTRACE=full /home/erockce/.cargo/bin/eruption -c support/config/eruption-dev.conf 2>&1 | tee > fail-to-work-$(date +%G%m%d-%H%M%S).log
```

Error outputs attached. First as user, second with `sudo`.

FWIW, I had the same loss of lights with another project and reported here: simonhuwiler/roccatvulcan/issues/6.

Attachments:
[fail-to-work-20200615-233510.log](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/4796268/fail-to-work-20200615-233510.log)
[fail-to-work-20200615-233808.log](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/4796269/fail-to-work-20200615-233808.log)

X3n0m0rph59, 2020-06-18:
Hello, thank you for your bug report!

It seems that eruption is unable to initialize the D-Bus interface, maybe because the interface configuration file is missing. Could you please try to run the following commands:

```sh
$ sudo cp support/dbus/org.eruption.control.conf /etc/dbus-1/system.d/org.eruption.control.conf
$ sudo systemctl reload dbus
```

And then please try to re-execute eruption.

I will look into supporting Ubuntu Bionic.

X3n0m0rph59, 2020-06-18:
Hello, I just have prepared a package for Ubuntu Bionic. Could you please try out the PPA?

erockce, 2020-06-22:
Thank you @X3n0m0rph59 for making a `bionic` package for installation!

I had a few tweaks to the getting started instructions.

```
$ sudo systemctl status eruption
Unit eruption.service could not be found.
$ dpkg -L eruption-roccat-vulcan-git | grep servic
/usr/lib/systemd/system/eruption.service
```

The `eruption.service` file was not immediately available - I chose not to reboot. The systemd script is set `disabled` by default.  So I enabled. 

```
$ sudo systemctl enable eruption.service
Created symlink /etc/systemd/system/basic.target.wants/eruption.service → /usr/lib/systemd/system/eruption.service.
```

Now the service is visible
```
$ sudo systemctl status eruption.service
● eruption.service - Driver for ROCCAT Vulcan 100/12x series keyboards
   Loaded: loaded (/usr/lib/systemd/system/eruption.service; enabled; vendor preset: disabled)
   Active: inactive (dead)
     Docs: man:eruption(8)
           man:eruption.conf(5)

Jun 21 16:48:24 mysystem systemd[1]: /usr/lib/systemd/system/eruption.service:9: Failed to parse service type, ignoring: exec
```

According to the man page for `systemd.service` on my system the value `exec` is not an available option for `Type=` (version `systemd 237`).  But for the latest `systemd 245` (systemd.service)[https://www.freedesktop.org/software/systemd/man/systemd.service.html] online documentation shows it is supported.  I keep my system up-to-date weekly, but since `bionic` is on LTS, `systemd` likely will not get updated too much further.  I will look into upgrading or maybe I'll write a separate issue/PR to update the `.service` file appropriate for `bionic`.  I'm not familiar developing Debian packages for PPAs (if your packaging system can support having different configurations possible between distribution packages).

```
$ sudo systemctl status eruption.service
● eruption.service - Driver for ROCCAT Vulcan 100/12x series keyboards
   Loaded: loaded (/usr/lib/systemd/system/eruption.service; enabled; vendor preset: disabled)
   Active: active (running) since Sun 2020-06-21 21:36:09 MDT; 7s ago
     Docs: man:eruption(8)
           man:eruption.conf(5)
 Main PID: 2769 (eruption)
    Tasks: 15 (limit: 4915)
   CGroup: /system.slice/eruption.service
           └─2769 /usr/bin/eruption -c /etc/eruption/eruption.conf

Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption                 > Spawning input thread...
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption                 > Switching to profile: profile1.profile
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption::plugins::keyboard > Now listening on: /dev/input/event3
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption::plugins::keyboard > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption::plugins::keyboard > Driver version: 10001
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption::plugins::keyboard > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption::plugins::keyboard > Physical location: usb-0000:00:14.0-1/input0
Jun 21 21:36:11 mysystem eruption[2769]:  INFO  eruption::plugins::keyboard > Grabbing the device exclusively
```

Neat! I have cool colors and reactions to typing now. Next will be experimenting with the profiles and scripts.

I consider this issue closed.

P.S. Thank you for starting this awesome project. I will certainly see if I can help out. One thing I noticed should be a "feature" is that if the service is stopped with the associated `eruption` program ended, the configuration before the application/service started should be restored or go back to factory default should be restored.  When I stopped the `eruption` service, the colors just froze and the brightness control did not work.  I had to power-cycle to restore settings.

closed 2020-06-22 by erockce