eruption-project/eruption #10: All leds are dark following update to 0.1.11

apfeiffertraum, 2020-07-25:
Following an update from eruption-roccat-vulcan-git.x86_64 0.1.10-1.fc32 to eruption-roccat-vulcan-git.x86_64 0.1.11-2.fc32, when the eruption service starts, all leds on my Vulcan 121 AIMO go dark. All was functioning beautifully before the update.

Curiously, there are no error messages or complaints from eruption and the eruption profile switcher seems to be able to change from one profile to another - but all leds remain dark.

SoundFX works.

X3n0m0rph59, 2020-07-26:
Hi! Thanks for your bug report, and sorry for the inconvenience this has caused you.

I am currently unable to reproduce this issue on my Fedora 32 VM.

This may very well be a bug in the Fedora package, caused by non up-to-date config files after the update. If that is true, the problem would be fixed by resetting the state of eruption. If you did any customizations like user macros, then please back up the customized files before performing the steps below!

To reset eruption to a clean state, please execute these commands:
```sh
sudo dnf erase eruption-roccat-vulcan-git
sudo rm -fr /etc/eruption/
sudo rm -fr /var/lib/eruption/
sudo rm -fr /usr/share/eruption/
sudo dnf install eruption-roccat-vulcan-git
```
Then please reboot your system.

If eruption still does not work as expected then could you please send me the debug logs generated by the eruption daemon, by running these commands:
```sh
$ sudo systemctl stop eruption.service
$ sudo RUST_LOG=debug eruption
```

apfeiffertraum, 2020-07-26:
I followed your directions to wipe and reinstall the package, and it is working again. Thank you for responding so quickly and for sharing your work! _eruption_ is wonderful!

My problem may have had something to do with brightness control. Before I purged the old version, there was a file named **/var/lib/eruption/eruption.state** that contained this:

```
active_slot = 2
profiles = [
    'batique.profile',
    'profile2.profile',
    'profile2.profile',
    'batique.profile',
]
enable_sfx = false
brightness = 0

```

At the same time, the brightness control slider in the Gnome shell extension didn't alter the brightness (leds remained completely dark). After reinstalling, I notice that this file is no longer there and the brightness slider works as expected.

X3n0m0rph59, 2020-07-26:
Great to hear that! Thanks for investigating! That seems to be the culprit:
>```
>brightness = 0
>```

>At the same time, the brightness control slider in the Gnome shell extension didn't alter the brightness (leds remained completely dark).

Will look into this...

I will close this issue now. Please feel free to open another issue if you encounter further problems!

closed 2020-07-26 by X3n0m0rph59