eruption-project/eruption #12: How do I disable the "Eruption Virtual Mouse" feature?

erockce, 2020-07-28:
Since 0.11, Eruption is trying to grab my mouse instead of letting me use a kernel Logitech driver (for my M325).

I use XFCE4 and with the previous Eruption version 0.10, I could simply select and uncheck the "Enable this device" for "Eruption Virtual Mouse" in the "Mouse and Touchpad" (I believe provided by the `xfsettingsd` process).

Now if I want mouse functionality, I'm forced to use the "Eruption Virtual Mouse" and disable "Logitech M325"

```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.4 LTS"
```

At one point just after I updated eruption, I was able to disable the "Eruption Virtual Mouse" and "Logitech M325" was enabled.  This promptly sent systemd-journal to use very high CPU usage and repeat the following message (seen when using `journalctl -xe`):
```
Jul 27 21:25:19 nuc eruption[875]:  ERROR eruption::plugins::mouse               > Could not get mouse events
Jul 27 21:25:49 nuc systemd-journald[361]: Suppressed 3567672 messages from eruption.service
```

X3n0m0rph59, 2020-07-28:
Hi! Thanks for bringing this issue to my attention!

I think that your issue is caused by Eruption grabbing your mouse device exclusively.
It is possible to completely opt-out of mouse support by adding:
```toml
[global]
# ...
grab_mouse = false
```
to the main configuration file: `/etc/eruption/eruption.conf`. (This feature has recently been promoted from experimental.)

Could you please describe in more detail which desired features you lose while using the `Eruption Virtual Mouse` device? Maybe I am able to fix the root cause then!

It may be the case that you are running another user-mode driver software that wants to grab the device too!?
Then either the driver software or Eruption will fail to open the device.

erockce, 2020-07-29:
@X3n0m0rph59 thank you for confirming.  It appears I was on the right track.

Last night when I re-read the `README.md` after filing the issue, I set the `grab_mouse` parameter before finishing for the day.

```
$ grep mouse /etc/eruption/eruption.conf
grab_mouse = false
```

Today I didn't give it much thought about the issue.  I just checked the XFCE4 "mouse and touchpad" control panel application and it shows the `Eruption Virtual Mouse` is disabled and `Logitech M325` enabled.

As far as having a potential bug, would some of the following information help?
```
$ dmesg | grep -i mouse
[    0.718873] mousedev: PS/2 mouse device common for all mice
[    2.007789] input: ROCCAT ROCCAT Vulcan AIMO Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.1/0003:1E7D:3098.0002/input/input4
[    2.067556] hid-generic 0003:1E7D:3098.0002: input,hiddev0,hidraw1: USB HID v1.11 Mouse [ROCCAT ROCCAT Vulcan AIMO] on usb-0000:00:14.0-2/input1
[    2.189756] input: Logitech USB Receiver Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.4/2-1.4.2/2-1.4.2:1.1/0003:046D:C52B.0006/input/input11
[    2.308039] hid-generic 0003:046D:C52B.0006: input,hiddev1,hidraw5: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-1.4.2/input1
[    2.592539] input: Logitech Unifying Device. Wireless PID:400a Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.4/2-1.4.2/2-1.4.2:1.2/0003:046D:C52B.0007/0003:046D:400A.0008/input/input15
[    2.594568] hid-generic 0003:046D:400A.0008: input,hidraw5: USB HID v1.11 Mouse [Logitech Unifying Device. Wireless PID:400a] on usb-0000:00:14.0-1.4.2/input2:1
[    3.713926] logitech-hidpp-device 0003:046D:400A.0008: input,hidraw5: USB HID v1.11 Mouse [Logitech M325] on usb-0000:00:14.0-1.4.2/input2:1
[    5.931915] input: Eruption Virtual Mouse as /devices/virtual/input/input29
```

And output from `journalctl -xe eruption`:
```
Jul 28 20:46:18 <HOSTNAME_REDACTED> systemd[1]: Started Driver for ROCCAT Vulcan 100/12x series keyboards.
-- Subject: Unit eruption.service has finished start-up
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit eruption.service has finished starting up.
-- 
-- The start-up result is RESULT.
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.11
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption > Enumerating connected devices...
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::hwdevices > Found Control interface: "0002:0003:01": ROCCAT Vulcan AIMO
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::hwdevices > Found LED interface: "0002:0003:03": ROCCAT Vulcan AIMO
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption            > Opening devices...
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption            > Initializing devices...
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption            > Configuring LEDs...
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption            > Initializing D-Bus API...
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption            > Registering plugins...
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption                 > Spawning system monitor thread...
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption                 > Spawning keyboard input thread...
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption                 > Mouse support is DISABLED by configuration
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption                 > Switching to profile: my-preset-blue-red.profile
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugins::keyboard > Now listening on: /dev/input/event3
Jul 28 20:46:19 <HOSTNAME_REDACTED> eruption[848]:  INFO  eruption::plugins::keyboard > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
[... cut ...]
Jul 29 00:35:09 <HOSTNAME_REDACTED> eruption[848]:  WARN  eruption                    > This means that we dropped at least one frame
Jul 29 00:35:09 <HOSTNAME_REDACTED> eruption[848]:  WARN  eruption                    > Loop took: 133 milliseconds, goal: 41
```

Also `eruption` is typically in my top five high CPU processes (between ~`2.85-3.35%`) according to my `conky` widget process listing. `eruption` uses more CPU than Google chrome with 3 tabs open.  I have an Intel NUC with a Core i7-5557U w/16GiB RAM and a Samsung EVO SSD.

X3n0m0rph59, 2020-07-30:
Hi! Thanks for the response!

I decided to split the configuration option `grab_mouse` into two distinct options, `enable_mouse` and `grab_mouse`. This change will likely make it into Eruption `0.1.12`:

If `enable_mouse` is set to `true`, Eruption will process mouse events (the mouse device is opened in "shared mode")
In addition to this, if `grab_mouse` is set to `true` as well, then Eruption will be able to inject mouse events (the mouse device is opened in "exclusive mode").
`grab_mouse` set to `true` will open the "real" mouse exclusively, hide it from the (input) system, and replace it with the "Eruption Virtual Mouse" device. That means if you want to use other software that needs to access the "real" hardware mouse device, you may want to set `grab_mouse` to `false`. This way you will only lose the ability to inject virtual mouse events (no Macros/Easy Shift+), but Eruption may at least process mouse events and show some effects.

Concerning the CPU load, please feel free to open up new issues any time!

closed 2020-08-02 by erockce