eruption-project/eruption #13: Screensaver Won't Activate when Eruption Is Running

erockce, 2020-08-09:
Hi @X3n0m0rph59 

Since the new version was installed today, my screen saver is now not activating if the `eruption` service is running.

If I stop the service, the screen saver (simple blanking) works again.  I did not have this issue with prior `0.1.11` version.

When I try `xset s activate` the screen blanks for 3 seconds, but then comes back.

If the `eruption` service is not running, the screen saver will activate at the specified timeout. I have also reset all the profile slots to `default.profile` and restarted the `eruption` service multiple times but the problem still occurs.

Please let me know if you need any more information.

Eruption Version
```
$ dpkg -s eruption-roccat-vulcan-git | grep "^Version"
Version: 0.1.12-ubuntu1
```
Distribution:
```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.5 LTS"
```

Service configuration
```
$ cat /etc/eruption/eruption.conf
# Eruption - Linux user-mode driver for the ROCCAT Vulcan 100/12x series keyboards
# Main configuration file

[global]
profile_dir = "/var/lib/eruption/profiles/"
script_dir = "/usr/share/eruption/scripts/"

# select your keyboard variant
# keyboard_variant = "ANSI"
keyboard_variant = "ISO"

# Mouse handling
enable_mouse = false
grab_mouse = false

# "Away from keyboard" handling
afk_profile = "matrix.profile"
afk_timeout_secs = 300
```

Eruption state file:
```
$ cat /var/lib/eruption/eruption.state 
active_slot = 3
profiles = [
    'default.profile',
    'default.profile',
    'default.profile',
    'default.profile',
]
enable_sfx = false
brightness = 20
```

Here is my `xset` parameters
```
$ xset q
Keyboard Control:
  auto repeat:  on    key click percent:  0    LED mask:  00000002
  XKB indicators:
    00: Caps Lock:   off    01: Num Lock:    on     02: Scroll Lock: off
    03: Compose:     off    04: Kana:        off    05: Sleep:       off
    06: Suspend:     off    07: Mute:        off    08: Misc:        off
    09: Mail:        off    10: Charging:    off    11: Shift Lock:  off
    12: Group 2:     off    13: Mouse Keys:  off
  auto repeat delay:  500    repeat rate:  20
  auto repeating keys:  00ffffffdffffbbf
                        fadfffefffedffff
                        9fffffffffffffff
                        fff7ffffffffffff
  bell percent:  50    bell pitch:  400    bell duration:  100
Pointer Control:
  acceleration:  0/1    threshold:  1
Screen Saver:
  prefer blanking:  yes    allow exposures:  yes
  timeout:  10    cycle:  60
Colors:
  default colormap:  0x22    BlackPixel:  0x0    WhitePixel:  0xffffff
Font Path:
  /usr/share/fonts/X11/misc,/usr/share/fonts/X11/Type1,built-ins
DPMS (Energy Star):
  Standby: 1800    Suspend: 0    Off: 3600
  DPMS is Enabled
  Monitor is On
```

Relevant `journalctl -xe -u eruption`:
```
-- The start-up result is RESULT.
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.12
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption > Loading saved state...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  WARN  eruption::state > Brightness configuration value is set very low, the LEDs will probably stay dark!
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption        > Registering plugins...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Plugins loaded and initialized successfully
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Enumerating connected devices...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::hwdevices            > Found Control interface: "0002:0003:01": ROCCAT Vulcan AIMO
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::hwdevices            > Found LED interface: "0002:0003:03": ROCCAT Vulcan AIMO
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Device enumeration completed
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Spawning keyboard input thread...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Mouse support is DISABLED by configuration
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Waiting for tasks to complete...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Opening keyboard device...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugins::keyboard    > Now listening on: /dev/input/event3
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugins::keyboard    > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugins::keyboard    > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugins::keyboard    > Physical location: usb-0000:00:14.0-2/input0
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption::plugins::keyboard    > Grabbing the keyboard device exclusively
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Initializing keyboard device...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Configuring keyboard LEDs...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Opening mouse device...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > This mouse is currently not supported, falling back to the generic driver
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Performing late initializations...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Initializing D-Bus API...
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Late initializations completed
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Startup completed
Aug 08 22:25:27 <hostname redacted> eruption[29602]:  INFO  eruption                       > Switching to profile: default.profile
```

X3n0m0rph59, 2020-08-09:
Hi! Thanks for your bug report!

Sounds like `eruption` somehow injects input events that prohibit the screensaver from being run.

I have tried on a Ubuntu 18.04 virtual machine as well as on my development box, but was unable to reproduce this issue.
On both machines the screensaver/blanking works as expected:
```sh
$ xset s 10
```
Wait 10 seconds...

Screen will blank, as expected after 10 seconds, and stay this way until I move the mouse or press a key.
This works with `eruption` running, as well as with `eruption.service` being masked and stopped.

```sh
$ xset s activate
```
fails to blank the screen most of the time. This one will work though:
```sh
$ sleep 1; xset s activate
```

I have tested all permutations of configuration parameters (assigned `true` or `false`, `0` or `5`):
`enable_mouse`, `grab_mouse`, `afk_timeout_secs`

To verify that spurious input events are the culprit in deed, could you please run `sudo evtest`:
```sh
$ sudo apt install evtest
$ sudo evtest
```
and select the _Eruption Virtual Keyboard/Mouse_ devices. If you see events being logged while you don't perform any input actions at all, then this will most likely be the cause of the issue.

X3n0m0rph59, 2020-08-09:
I can reproduce it now on my development box! It seems that something is resetting the screensaver timeout. So spurious input events are not the root cause of this issue.

I will investigate...

X3n0m0rph59, 2020-08-09:
I just set up an Ubuntu 18.04 VM using the xfce4 desktop environment. Furthermore, I did install the xscreensaver package and configured a timeout of 1 minute, using the screensaver configuration GUI (Xfce4 menu item: Settings / Screensaver). This way, the screensaver seems to work alright.

Could you please tell me how you set the timeout of the screensaver? Did you set it using the xscreensaver GUI or via `xset s <secs>`. The `xset s <secs>` value often gets reset for a yet to be discovered reason, on both of my test machines.

erockce, 2020-08-09:
I wasn't expecting you to already attempt (and in one case) reproduce my problem!

I'm sorry for the quick knee-jerk issue I created yesterday but it appears things with eruption might be working after all.

More in-depth investigation and monitoring is required. Today my screensaver is working as expected.  I did not change anything from yesterday.  I let my system sit idle for 5 minutes (300 seconds) then the screen blanked and Eruption AFK mode kicked into the `matrix` display as defined by the configuration.

The only item that appears to have changed is that `cycle` for screen saver is now `0` which I did not modify myself but maybe there are some defaults that get set if something is not correct.

```
$ xset q
Keyboard Control:
  auto repeat:  on    key click percent:  0    LED mask:  00000002
  XKB indicators:
    00: Caps Lock:   off    01: Num Lock:    on     02: Scroll Lock: off
    03: Compose:     off    04: Kana:        off    05: Sleep:       off
    06: Suspend:     off    07: Mute:        off    08: Misc:        off
    09: Mail:        off    10: Charging:    off    11: Shift Lock:  off
    12: Group 2:     off    13: Mouse Keys:  off
  auto repeat delay:  500    repeat rate:  20
  auto repeating keys:  00ffffffdffffbbf
                        fadfffefffedffff
                        9fffffffffffffff
                        fff7ffffffffffff
  bell percent:  50    bell pitch:  400    bell duration:  100
Pointer Control:
  acceleration:  0/1    threshold:  1
Screen Saver:
  prefer blanking:  yes    allow exposures:  yes
  timeout:  300    cycle:  0
Colors:
  default colormap:  0x22    BlackPixel:  0x0    WhitePixel:  0xffffff
Font Path:
  /usr/share/fonts/X11/misc,/usr/share/fonts/X11/Type1,built-ins
DPMS (Energy Star):
  Standby: 1800    Suspend: 0    Off: 3600
  DPMS is Enabled
  Monitor is On
```

Background on my system installation is from Xubuntu 18.04 and I use Light DM with XFCE4. As far as display and screen related packages, I do not have the `xscreensaver` package installed and instead just let "screen saving" be a simple screen blank provided by the X server (specifically is turning the HDMI signal OFF to the monitor).  I do have `light-locker` installed, a program called `caffeine` (which wasn't activated), and `redshift` and `redshift-gtk`.

Here is `evtest` for the keyboard and mouse... 

`evtest` for keyboard

```
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Sleep Button
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	ROCCAT ROCCAT Vulcan AIMO
/dev/input/event7:	ROCCAT ROCCAT Vulcan AIMO
/dev/input/event8:	Logitech M325
/dev/input/event10:	Video Bus
/dev/input/event11:	HDA Intel HDMI HDMI/DP,pcm=3
/dev/input/event12:	HDA Intel HDMI HDMI/DP,pcm=7
/dev/input/event13:	HDA Intel HDMI HDMI/DP,pcm=8
/dev/input/event14:	HDA Intel HDMI HDMI/DP,pcm=9
/dev/input/event15:	HDA Intel HDMI HDMI/DP,pcm=10
/dev/input/event16:	HDA Intel PCH Mic
/dev/input/event17:	HDA Intel PCH Headphone
/dev/input/event18:	Eruption Virtual Keyboard
/dev/input/event19:	Eruption Virtual Mouse
Select the device event number [0-19]: 18
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x59 product 0x123 version 0x1
Input device name: "Eruption Virtual Keyboard"
Supported events:
[... 222 lines cut ...]
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
    Event code 277 (BTN_FORWARD)
    Event code 278 (BTN_BACK)
    Event code 279 (BTN_TASK)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 2 (REL_Z)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (?)
    Event code 12 (?)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
 ^C
```

No issues or "ghosts in the machine" detected! ;-)

Side question...although I have the mouse options set to `false`, why is there still activation of the `Eruption Virtual Mouse`?  Why would I need this if I'm relying on the mouse driver provided by the kernel?

erockce, 2020-08-10:
For what it is worth I restored all of my previous custom profiles to slots (which right now are nothing more than renamed presets (`preset-blue-red.profile`, `preset-red-yellow.profile`, `profile3.profile`, and `spectrum-analyzer.profile`) to see if there was something strange in the profiles or scripts they used to cause the reported issue.  My computer has been left on all day and the screen has blanked and Eruption goes to AFK mode after 5 minutes as expected without any issues.

Maybe another recent update or removal of an X library caused the temporary issue.  I will have to go through the apt logs for details.

At this point if you do not feel there is any issue or cannot reproduce the reported problem further, I think this issue can be closed.

X3n0m0rph59, 2020-08-10:
Hi! Thanks for the quick reply!

>I'm sorry for the quick knee-jerk issue I created yesterday but it appears things with eruption might be working after all.

Not a problem at all! Please feel free to open up issues anytime!
Your `sudo evtest` output is looking good, so we can be reasonably sure that spurious events aren't involved here.

I noticed the same during testing yesterday. The values set by `xset s <num> <num>` get magically reset.
I suspect that we aren't supposed to use the low-level configuration stuff of the X server, but instead something like `xfce4-power-manager-settings` instead.

>Side question...although I have the mouse options set to false, why is there still activation of the Eruption Virtual Mouse? Why would I need this if I'm relying on the mouse driver provided by the kernel?

I decided to not completely remove the `Eruption Virtual Mouse` device when `enable_mouse = false` and `grab_mouse = false`, so that we are still able to inject mouse events to the input system. Said virtual events just wouldn't originate from "real" mouse events anymore. The only time you would see events on the virtual mouse in that case would be, when a Lua script like `user-macros.lua` injects mouse events by calling `inject_mouse_button(button_index, down)` or `inject_mouse_wheel(direction)`, e.g. as part of a custom "Easy Shift" (keyboard) macro.

I'm closing this issue now, but please re-open it if you find that eruption is involved somehow!

closed 2020-08-10 by X3n0m0rph59