eruption-project/eruption #15: eruption cannot connect to pulseaudio

Nanowires, 2020-08-27:
Keyboard: Roccat Vulcan 121 AIMO
Distro: Fedora 32
Issue: Volume controls not working
Read and followed all instructions in the entire README
Followed "Support for Audio Playback and Capture" instructions without any resolution.
journalctl outputs these errors from eruption:
ERROR eruption::plugins::audio           > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
ERROR eruption::plugins::audio           > Could not start the audio grabber: Pulse Audio error: Could not open Pulse Audio: Connection refused

Any ideas? ;-)
Thanks

X3n0m0rph59, 2020-08-28:
Hi! Could you please paste the contents of the following files:

`~/.config/pulse/default.pa`
`/root/.config/pulse/client.conf`

And please show the output of the following command:
```
$ journalctl --user -u pulseaudio.service -b
```

Nanowires, 2020-08-28:
Hi,

thanks for your fast response :-)
BTW: Very nice tool

FYI: I'm now able to grab the audio (i.e. visualize the audio) and play audio via keystrokes. The problem was, that the wrong audio source was selected for eruption in the pulseaudio mixer.

But i still cannot control the volume with the knob, or mute/unmute the audio.

~/.config/pulse/default.pa:

```
#!/usr/bin/pulseaudio -nF
#
# This file is part of PulseAudio.
#
# PulseAudio is free software; you can redistribute it and/or modify it
# under the terms of the GNU Lesser General Public License as published by
# the Free Software Foundation; either version 2 of the License, or
# (at your option) any later version.
#
# PulseAudio is distributed in the hope that it will be useful, but
# WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
# General Public License for more details.
#
# You should have received a copy of the GNU Lesser General Public License
# along with PulseAudio; if not, see <http://www.gnu.org/licenses/>.

# This startup script is used only if PulseAudio is started per-user
# (i.e. not in system mode)

.fail

### Automatically restore the volume of streams and devices
load-module module-device-restore
load-module module-stream-restore
load-module module-card-restore

### Automatically augment property information from .desktop files
### stored in /usr/share/application
load-module module-augment-properties

### Should be after module-*-restore but before module-*-detect
load-module module-switch-on-port-available

### Load audio drivers statically
### (it's probably better to not load these drivers manually, but instead
### use module-udev-detect -- see below -- for doing this automatically)
#load-module module-alsa-sink
#load-module module-alsa-source device=hw:1,0
#load-module module-null-sink
#load-module module-pipe-sink

### Automatically load driver modules depending on the hardware available
.ifexists module-udev-detect.so
load-module module-udev-detect
.else
### Use the static hardware detection module (for systems that lack udev support)
load-module module-detect
.endif

### Automatically connect sink and source if JACK server is present
.ifexists module-jackdbus-detect.so
.nofail
load-module module-jackdbus-detect channels=2
.fail
.endif

### Automatically load driver modules for Bluetooth hardware
[... 97 lines cut ...]
$ journalctl --user -u pulseaudio.service -b:

```
-- Logs begin at Wed 2020-08-26 17:33:03 CEST, end at Fri 2020-08-28 08:36:46 CEST. --
Aug 28 05:55:08 <PC> systemd[2136]: Starting Sound Service...
Aug 28 05:55:08 <PC> pulseaudio[2599]: W: [pulseaudio] alsa-ucm.c: Path Speaker is not a volume control
Aug 28 05:55:08 <PC> systemd[2136]: Started Sound Service.
Aug 28 05:55:33 <PC> pulseaudio[2599]: E: [pulseaudio] bluez5-util.c: GetManagedObjects() failed: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Aug 28 06:35:34 <PC> pulseaudio[2599]: E: [alsa-sink-ALC1220 Analog] alsa-sink.c: ALSA weckte uns auf, um neue Daten auf das Gerät zu schreiben, doch es gab nichts zum Schreiben!
Aug 28 06:35:34 <PC> pulseaudio[2599]: E: [alsa-sink-ALC1220 Analog] alsa-sink.c: Dies ist höchstwahrscheinlich ein Fehler im ALSA-Treiber »snd_hda_intel«. Bitte melden Sie diesen Fehler den ALSA-Entwicklern.
Aug 28 06:35:34 <PC> pulseaudio[2599]: E: [alsa-sink-ALC1220 Analog] alsa-sink.c: Wir wurden durch das POLLOUT-Set geweckt, allerdings lieferte ein anschließender snd_pcm_avail() den Wert 0 oder einen anderen Wert < min_avail.
Aug 28 07:50:04 localhost.localdomain pulseaudio[2599]: E: [alsa-sink-USB Audio] alsa-sink.c: Error opening PCM device front:0: Das Argument ist ungültig
Aug 28 07:50:04 localhost.localdomain pulseaudio[2599]: W: [alsa-sink-ALC1220 Analog] alsa-util.c: Unable to set sw params: Keine Berechtigung
Aug 28 07:50:04 localhost.localdomain pulseaudio[2599]: E: [alsa-sink-ALC1220 Analog] alsa-sink.c: Failed to set software parameters: Keine Berechtigung
Aug 28 07:51:16 <PC> systemd[2136]: pulseaudio.service: Succeeded.
Aug 28 07:51:16 <PC> systemd[2136]: pulseaudio.service: Consumed 9min 41.813s CPU time.
Aug 28 07:51:17 <PC> systemd[2136]: Starting Sound Service...
Aug 28 07:51:18 <PC> pulseaudio[5189]: W: [pulseaudio] alsa-ucm.c: Path Speaker is not a volume control
Aug 28 07:51:18 <PC> systemd[2136]: Started Sound Service.
Aug 28 07:51:43 <PC> pulseaudio[5189]: E: [pulseaudio] bluez5-util.c: GetManagedObjects() failed: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Aug 28 07:55:34 <PC> pulseaudio[5189]: E: [alsa-sink-ALC1220 Analog] alsa-sink.c: ALSA weckte uns auf, um neue Daten auf das Gerät zu schreiben, doch es gab nichts zum Schreiben!
Aug 28 07:55:34 <PC> pulseaudio[5189]: E: [alsa-sink-ALC1220 Analog] alsa-sink.c: Dies ist höchstwahrscheinlich ein Fehler im ALSA-Treiber »snd_hda_intel«. Bitte melden Sie diesen Fehler den ALSA-Entwicklern.
Aug 28 07:55:34 <PC> pulseaudio[5189]: E: [alsa-sink-ALC1220 Analog] alsa-sink.c: Wir wurden durch das POLLOUT-Set geweckt, allerdings lieferte ein anschließender snd_pcm_avail() den Wert 0 oder einen anderen Wert < min_avail.

```

X3n0m0rph59, 2020-08-28:
Nice, halfway there!

When the mute button is pushed on the keyboard, or the volume knob is rotated, eruption will inject keystrokes on the virtual keyboard device that need to be handled by your desktop environment.

The following lines are taken from `scripts/macros.lua`

```Lua
        -- process other HID events
	if event_type == 3 then
		-- Mute button event
		if key_code == 1 then
			inject_key(113, true) -- KEY_MUTE (audio) (down)
			set_status_led(1, true)
		else
			inject_key(113, false) -- KEY_MUTE (audio) (up)
		end
	elseif event_type == 4 then
		-- Volume/brightness dial knob rotation
		local event_handled = false
		if on_dial_knob_rotate_left ~= nil and on_dial_knob_rotate_right ~= nil then
			if key_code == 0 then
				-- default behaviour may be overridden by a user macro
				event_handled = on_dial_knob_rotate_right(key_code)
			else
				-- default behaviour may be overridden by a user macro
				event_handled = on_dial_knob_rotate_left(key_code)
			end
		end

		if not event_handled then
			-- default action is to adjust volume
			overlay_state = VOLUME_OVERLAY
			overlay_ttl = overlay_max_ttl

			if key_code == 1 then
				inject_key(114, true) -- VOLUME_DOWN (down)
				inject_key(114, false) -- VOLUME_DOWN (up)
			else
				inject_key(115, true) -- VOLUME_UP (down)
				inject_key(115, false) -- VOLUME_UP (up)
			end
		end
	end
```

The relevant lines are: `inject_key(<code>, up_or_down)`

So your desktop environment needs to be told to handle these key events on the virtual keyboard, and subsequently adjust the master volume.

Which desktop environment are you currently using?

You may want to debug the input subsystem using `evtest`, and select the `Eruption Virtual Keyboard` device, to see if eruption sends the right events.

The output should look like this:
```
Event: time 1598601495.058400, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1598601495.058400, -------------- SYN_REPORT ------------
Event: time 1598601496.556592, type 1 (EV_KEY), code 113 (KEY_MUTE), value 1
Event: time 1598601496.556592, -------------- SYN_REPORT ------------
Event: time 1598601496.657587, type 1 (EV_KEY), code 113 (KEY_MUTE), value 0
Event: time 1598601496.657587, -------------- SYN_REPORT ------------
Event: time 1598601498.258566, type 1 (EV_KEY), code 113 (KEY_MUTE), value 1
Event: time 1598601498.258566, -------------- SYN_REPORT ------------
Event: time 1598601498.409578, type 1 (EV_KEY), code 113 (KEY_MUTE), value 0
Event: time 1598601498.409578, -------------- SYN_REPORT ------------
Event: time 1598601500.112406, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1598601500.112406, -------------- SYN_REPORT ------------
Event: time 1598601500.112447, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1598601500.112447, -------------- SYN_REPORT ------------
Event: time 1598601500.314794, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1598601500.314794, -------------- SYN_REPORT ------------
Event: time 1598601500.314834, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1598601500.314834, -------------- SYN_REPORT ------------
Event: time 1598601500.865084, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1598601500.865084, -------------- SYN_REPORT ------------
Event: time 1598601500.865130, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1598601500.865130, -------------- SYN_REPORT ------------
Event: time 1598601501.365104, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1598601501.365104, -------------- SYN_REPORT ------------
Event: time 1598601501.365144, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1598601501.365144, -------------- SYN_REPORT ------------
```

Nanowires, 2020-08-28:
Thanks, this may be the problem:
I'm using the minimalistic XFCE4 environment.

At least evtest shows, that the key strokes are recognised correctly. :-)

Nanowires, 2020-08-28:
Ok, sorry for filing an issue...

Thanks for your time and support :-)

closed 2020-08-28 by Nanowires

MrYdse, 2021-11-13:
Hi, this happens to me as well on KDE 5.23.3 on Arch Linux with kernel 5.15.2-arch1-1 on X11. I get the same journalctl errors repeated several times per second, flooding the log:
`ERROR eruption::plugins::audio > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
ERROR eruption::plugins::audio > Could not start the audio grabber: Pulse Audio error: Could not open Pulse Audio: Connection refused`

I have replaced ALSA, PulseAudio and Jack with Pipewire. Does eruption not work with Pipewire?

I want to either ask to reopen this issue, or I can create a new one, though it is about the same error message.

X3n0m0rph59, 2021-11-13:
Hi @MrYdse!

Eruption is compatible with Pipewire, it will use the PulseAudio interface of Pipewire. There currently is additional configuration required though. Please find the respective steps in [SETUP_AUDIO.md](https://github.com/X3n0m0rph59/eruption/blob/master/docs/SETUP_AUDIO.md)

If the problem persists, please open up a new issue!

MrYdse, 2021-11-13:
Ok, that helped my poor journald. Will the changes persist? Do I have to redo the steps if I disconnect the sink that I connected Eruption Audio Grabber to?

Are these steps possible to automate? I wish to be able to use eruption without much configuration after install if possible.

Also, I noticed that eruption runs as root, which sounds a bit scary. What are the reasons it runs as root? Is it because it is a service/daemon?

Sorry for all the questions.

MrYdse, 2021-11-13:
Btw, if I disconnect the sink (it is a bluetooth speaker) and use the volume knob on my Roccat Vulkan 120, this error pops up:
`mod.protocol-pulse: client 0x560dd82c6620 [libcanberra]: ERROR command:18 (PLAY_SAMPLE) tag:8 error:25 (Input/output error)`

X3n0m0rph59, 2021-11-13:
@MrYdse 

> Ok, that helped my poor journald. Will the changes persist? Do I have to redo the steps if I disconnect the sink that I connected Eruption Audio Grabber to?

Great! I suspect that the changes will persist. Pipewire/Wireplumber should restore these settings respectively.

> I wish to be able to use eruption without much configuration after install if possible.

I am working on a solution for this right now... I am planning to introduce a user session daemon called `eruption-audio-proxy` that listens on a Pipewire sink and send the recorded samples to the Eruption daemon for further processing like e.g. Fourier transform for the spectrum analyzer. Got a prototype implementation working already!

> Are these steps possible to automate?

This would be an even more clumsy solution than what we currently have... I prefer the aforementioned architectural change to Eruption.

MrYdse, 2021-11-13:
Alright, thanks for the informative answers! I appreciate this project a lot, it is very useful.