eruption-project/eruption #16: Profile Using stripes.lua + audioviz3.lua Cause Lights to Occasionally Freeze

erockce, 2020-08-29:
Changes between the current and previous Eruption release (or I never noticed) affects the processing of audio and will occasionally freeze the lights on the keyboard when using `stripes.lua` and `audioviz3.lua`.  After a few seconds the audio signal is processed again, but then the lights freeze again after a few seconds.  Sometimes the freeze is for a significant length of time until I switch profiles and then back.

Perhaps Eruption does not like my taste in music :-)  For what it is worth...the issue also happens for speech audio such as while watching Twitch streams.

The LUA scripts `stripes.lua` and `audioviz3.lua` are together in the default `profile3.profile`.

Here is my "profile3" profile (which is a copy of `profile3.profile`):
```
$ cat /var/lib/eruption/profiles/my-profile3.profile 
id = '5dc62fa6-e965-45cb-a0da-e87d29713097'
name = "My Profile 3 (Spectrum Analyzer)"
description = "My Profile #3"
active_scripts = [
  'stripes.lua',
  'audioviz3.lua',
  'afterglow.lua',
  'macros.lua',
]

[[config."Audio Visualizer #3 (Spectrum Analyzer)"]]
type = 'float'
name = 'opacity'
value = 0.75
```

I've confirmed that when `stripes.lua` is commented out, there is no freezing.  But you loose the effect of a default color background until audio is registered.  I believe an earlier of Eruption did have `stripes.lua` in `profile3.profile`.  Before installing new releases, I try to keep a backup of the `/var/lib/eruption/profile` directory so I can compare and update my custom profiles in case there are new features.  The prior release had `stripes.lua` included.  I can't remember if there was freezing with the prior release.

I have another profile (a copy of the included, `spectrum-analyzer.profile`) using `audiovis2.lua` which does not have any freezing issues.

I have not done systematic testing using combinations of other `lua` scripts with `audioviz3.lua` to see if the same problem happens.

The LUA scripts are unmodified from what is provided.

Eruption package version installed
```
$ dpkg -l eruption-roccat-vulcan-git  | grep "^ii"
ii  eruption-roccat-vulcan-git 0.1.13-ubuntu0 amd64        Linux user-mode driver for the ROCCAT Vulcan 100/12x series keyboards
```
System Information
```
$ sudo inxi -C -m
CPU:       Dual core Intel Core i7-5557U (-MT-MCP-) cache: 4096 KB
           clock speeds: max: 3400 MHz 1: 2270 MHz 2: 2083 MHz 3: 2144 MHz 4: 1819 MHz
Memory:    Used/Total: 1739.5/15836.8MB
           Array-1 capacity: 16 GB devices: 2 EC: None
           Device-1: CHANNEL A DIMM0 size: 8 GB speed: 1867 MT/s type: DDR3
           Device-2: CHANNEL B DIMM0 size: 8 GB speed: 1867 MT/s type: DDR3
```
Ubuntu Release (Xubuntu)
```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.5 LTS"
```

There are no messages in the `eruption` system log when the freezing happens.

What other information can I provide for you?

X3n0m0rph59, 2020-08-29:
Hi! Thanks again for your bug report!

I am able to reproduce this issue on a Ubuntu 18.04 VM. I am working on a fix/improving the spectrum analyzer code now...

As a temporary workaround you may comment out line `37` in `audioviz3.lua`.

Please change:
```Lua
  if ticks % 2 ~= 0 then return end
```

to:
```Lua
-- if ticks % 2 ~= 0 then return end
```
Please tell me if this fixes the issue completely.

X3n0m0rph59, 2020-08-29:
I just published Eruption version `0.1.14` which should fix this issue!

erockce, 2020-08-29:
Thanks @X3n0m0rph59. I'm trying to understand the change as it relates to the commit message.

I reviewed commit (e53abdb261e4f4e0dacc7e3b6308a893754ce5f3) for the "fix".

I'm no Lua expert but looks like you removed an optimization for conditions when the `tick` counter is an even number to exit the function.

With your comment 
>...at least with the old spectrum analyzer code for versions <= 0.1.13

I see this statement applies to the version of Eruption.

Does the issue not happen because 1) it was never reported, so you wouldn't have known 2) not an issue with other Linux distributions or versions (such as with Ubuntu 20.04) due to use of different Rust or Lua engine versions?

Does this have to do with the Lua engine version or implementation for Ubuntu 18.04.x versus a newer distribution?

X3n0m0rph59, 2020-08-29:
Hi @erockce!

The early return condition is buggy in the sense that it inverts logic (does not test for `... == 0`, but for `~= 0`), which in combination with the ticks counter (and seemingly the old `get_audio_spectrum()` code) tends to cause fluctuating hangs.

I was able to reproduce this in both of my Ubuntu VMs as well as a Fedora 32 VM. This did not happen on my Arch-based development system maybe because of different timings, I suppose.

The fix just removed the buggy condition. 

Eruption version `0.1.14` includes one additional changeset: The spectrum analyzer code is no longer sensitive to timing issues of the caller, since the code now gets executed in the audio grabber thread as soon as we receive new audio data.

I am currently evaluating two optimization opportunities:
1. To reintroduce `if ticks % 2 == 0 then return end` into `audioviz3.lua`. Will need further tests if it's looking smooth enough.
2. Execute Spectrum Analyzer/FFT code only when `get_audio_spectrum()` was called at least once.

>Does the issue not happen because 1) it was never reported, so you wouldn't have known 2) not an issue with other Linux distributions or versions (such as with Ubuntu 20.04) due to use of different Rust or Lua engine versions?

* It was never reported, I did not know it existed.
* I could reproduce it on Fedora 32 (no long hangs just stuttering sometimes)
* It hangs on too on Ubuntu 20.04 

>Does this have to do with the Lua engine version or implementation for Ubuntu 18.04.x versus a newer distribution?

Seems that thread scheduling plays a much bigger part.

@erockce Thanks for bringing this to my attention.

erockce, 2020-08-29:
Fantastic @X3n0m0rph59. I have confirmed the issue is resolved with `0.1.14`.

I'm very grateful for your dedication and prompt responses to issues for this project!

closed 2020-08-29 by erockce