eruption-project/eruption #17: Eruption crashes on startup

bthiebault, 2020-09-04:
I installed eruption yesterday and it worked very well.
This morning however, it crashed at startup.

The error message from `systemctl status eruption` command line is the following:

`Could not initialize the keyboard plugin: Could not get the name of the evdev device from udev`

Restarting the service via `sudo systemctl restart eruption.service` works, though.

I'm using eruption-roccat-vulcan-git version 0.1.14-0 from AUR on Arch Linux, with a Roccat vulcan 120 AIMO

bthiebault, 2020-09-04:
The full journalctl output is the following:

`
-- Logs begin at Fri 2019-07-19 16:23:42 CEST, end at Fri 2020-09-04 11:10:30 CEST. --
Sep 04 09:10:31 Solgaleo systemd[1]: Starting Driver for ROCCAT Vulcan 100/12x series keyboards...
Sep 04 09:10:31 Solgaleo systemd[1]: Started Driver for ROCCAT Vulcan 100/12x series keyboards.
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.14
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption > Loading saved state...
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption > Registering plugins...
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption                       > Plugins loaded and initialized successfully
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption                       > Enumerating connected devices...
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::hwdevices            > Found Control interface: "0007:0003:01": ROCCAT Vulcan AIMO
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::hwdevices            > Found LED interface: "0007:0003:03": ROCCAT Vulcan AIMO
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption::hwdevices::roccat_vulcan > Bound driver: ROCCAT Vulcan
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption                           > Device enumeration completed
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption                           > Spawning keyboard input thread...
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption                           > Mouse support is DISABLED by configuration
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption                           > Waiting for tasks to complete...
Sep 04 09:10:31 Solgaleo eruption[585]:  INFO  eruption                           > Opening keyboard device...
Sep 04 09:10:31 Solgaleo eruption[585]:  ERROR eruption                           > Could not initialize the keyboard plugin: Could not get the name of the evdev device from udev
Sep 04 09:10:31 Solgaleo eruption[585]: The application panicked (crashed).
Sep 04 09:10:31 Solgaleo eruption[585]: Message:  explicit panic
Sep 04 09:10:31 Solgaleo eruption[585]: Location: eruption/src/main.rs:290
Sep 04 09:10:31 Solgaleo eruption[585]: Backtrace omitted.
Sep 04 09:10:31 Solgaleo eruption[585]: Run with RUST_BACKTRACE=1 environment variable to display it.
Sep 04 09:10:31 Solgaleo eruption[585]: Run with RUST_BACKTRACE=full to include source snippets.
Sep 04 09:10:32 Solgaleo systemd[1]: eruption.service: Main process exited, code=killed, status=6/ABRT
Sep 04 09:10:32 Solgaleo systemd[1]: eruption.service: Failed with result 'signal'.
`

X3n0m0rph59, 2020-09-04:
Hi and thanks again for your bug report!

Sounds like some kind of race condition. I am so far unable to reproduce this. Will need to dig deeper...

Can you tell me how often this does happen? Is it reproducible?
Could you please post your `/etc/eruption/eruption.conf` file?

bthiebault, 2020-09-04:
I only rebooted twice... and it happened twice. I can try rebooting more if necessary.

My .conf file is pretty basic:
```
# Eruption - Linux user-mode driver for the ROCCAT Vulcan 100/12x series keyboards
# Main configuration file

[global]
profile_dir = "/var/lib/eruption/profiles/"
script_dir = "/usr/share/eruption/scripts/"

# select your keyboard variant
# keyboard_variant = "ANSI"
keyboard_variant = "ISO"

# Mouse handling
enable_mouse = false
grab_mouse = false

# "Away from keyboard" handling
afk_profile = "rainbow-wave.profile"
afk_timeout_secs = 0
```
(As an aside, I don't know what options are available for the keyboard variant... I have an AZERTY keyboard, not sure it makes a difference).

I guess this may be due to the order in which the systemd service is started (probably too early), but I'm not sure exactly what the code that crashes is trying to do.

I read (most of) the rust book, but your code is too advanced for a rust beginner like me to read.

X3n0m0rph59, 2020-09-05:
Hi! I am now able to reproduce this in an ArchLinux VM. It seems like the USB-HID code is completely broken on such setups. 
Eruption somehow is unable to open/read from the USB control device. (The evdev layer is working though)

>(As an aside, I don't know what options are available for the keyboard variant... I have an AZERTY keyboard, not sure it makes a difference).

I don't think that this is a problem.

>I guess this may be due to the order in which the systemd service is started (probably too early), but I'm not sure exactly what the code that crashes is trying to do.

Yes guessed that too at first, but as it turns out the code does not work even if Eruption is started from a shell.

I am working on a fix now...

X3n0m0rph59, 2020-09-05:
>Hi! I am now able to reproduce this in an ArchLinux VM. It seems like the USB-HID code is completely broken on such setups.
Eruption somehow is unable to open/read from the USB control device. (The evdev layer is working though)

That turned out to be a red herring. USB device forwarding to the VM did not work right.

So I am still unable to reproduce this...

I think we should rule out some obvious things first:
* Could you please verify that you are using the most recent firmware that is available for your device, by updating using ROCCAT Swarm?
* If you are using a USB-hub could you please plug the keyboard directly into a plug on the mainboard?

bthiebault, 2020-09-05:
The keyboard is plugged directly to the motherboard, so it's not that.
In theory, I updated the keyboard firmware when I received it earlier this week. I'm gonna restart on Windows and double check though.
I think this has to do with the timing for the startup of the systemd service. When I start it within my session, it works perfectly. A workaround would be to restart it automatically once I'm logged in (in my bspwmrc file for instance). However, the systemd service currently is launched by root, so it requires a password. Does eruption service actually need to be started as root?

X3n0m0rph59, 2020-09-05:
>The keyboard is plugged directly to the motherboard, so it's not that.
In theory, I updated the keyboard firmware when I received it earlier this week. I'm gonna restart on Windows and double check though. (Edit: I just checked, I've got hte last version of the firmware).

Thank you for verifying this!

>I think this has to do with the timing for the startup of the systemd service. 

Yes, I suppose so too. I just have changed the udev related device enumeration code to implement a 'back-off and wait' feature, when it encounters errors. Will push it out soon...

Do you want to build from source and test this change yourself, or do you want to wait for a new release of Eruption?

>When I start it within my session, it works perfectly. A workaround would be to restart it automatically once I'm logged in (in my bspwmrc file for instance). However, the systemd service currently is launched by root, so it requires a password. 

Ahh, did you do something like ```systemctl enable eruption.service```? The daemon is supposed to get started via an udev rule.

>Does eruption service actually need to be started as root?

It once was able to run as a user service a long time ago. I doubt that it currently works correctly as a non-root user.

bthiebault, 2020-09-05:
> Ahh, did you do something like `systemctl enable eruption.service`? The daemon is supposed to get started via an udev rule.

No, I only restarted the service with systemctl restart.

> Do you want to build from source and test this change yourself, or do you want to wait for a new release of Eruption?

I can try building from source... I'll do it as soon as the kids give me a minute :)

X3n0m0rph59, 2020-09-05:
To build from sources you may want to `git clone` the repository and then simply change to the directory `support/pkg/arch/`. You then should be able to run `makepkg -si`. 

After the package has been installed:

```
 $ eruption -V
```
should print version `0.1.15`.

If you want to make local changes to the sources you should build Eruption using `cargo build` and run it with `sudo target/debug/eruption`.

bthiebault, 2020-09-05:
I compiled eruption as instructed. `eruption -V` returned `Eruption 0.1.15`

I rebooted... and it works now. For information, the output of journalctl is now:
```
-- Logs begin at Fri 2019-07-19 16:23:42 CEST, end at Sun 2020-09-06 00:05:25 CEST. --
Sep 05 22:05:26 Solgaleo systemd[1]: Starting Driver for ROCCAT Vulcan 100/12x series keyboards...
Sep 05 22:05:26 Solgaleo systemd[1]: Started Driver for ROCCAT Vulcan 100/12x series keyboards.
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.15
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption > Loading saved state...
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption > Registering plugins...
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Sep 05 22:05:26 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                       > Plugins loaded and initialized successfully
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                       > Enumerating connected devices...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::hwdevices            > Found Control interface: "0007:0003:01": ROCCAT Vulcan AIMO
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::hwdevices            > Found LED interface: "0007:0003:03": ROCCAT Vulcan AIMO
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::hwdevices::roccat_vulcan > Bound driver: ROCCAT Vulcan
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Device enumeration completed
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Spawning keyboard input thread...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Mouse support is DISABLED by configuration
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Waiting for tasks to complete...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Opening keyboard device...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::plugins::keyboard        > Now listening on: /dev/input/event22
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::plugins::keyboard        > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::plugins::keyboard        > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::plugins::keyboard        > Physical location: usb-0000:0b:00.3-2/input0
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption::plugins::keyboard        > Grabbing the keyboard device exclusively
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Initializing keyboard device...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Configuring keyboard LEDs...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Opening mouse device...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > This mouse is currently not supported, falling back to the generic driver
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Performing late initializations...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Initializing D-Bus API...
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Late initializations completed
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Startup completed
Sep 05 22:05:27 Solgaleo eruption[600]:  INFO  eruption                           > Switching to profile: profile1.profile
```

closed 2020-09-05 by bthiebault

X3n0m0rph59, 2020-09-05:
Great! Thank you very much for your patience! Please feel free to open up new issues anytime!