eruption-project/eruption #18: Volume control does not work

bthiebault, 2020-09-04:
The volume control knob does not work. When I turn it, the LED lightning goes white for a few seconds before going back to the selected profile. The volume does not change. The mute button and the FX button don't do anything. When pressing the FX button, it is turned on and the volume button is turned off, but the behaviour of the volume knob does not change.

I have followed the README instructions regarding PulseAudio. However, I'm not sure this was necessary. The README says it needs to be done `If you want Eruption to be able to play back sound effects, or use one of the audio visualizer Lua scripts`, which I actually don't want. But I figured that if my keyboard must adjust the volume it might better have access to the audio.

X3n0m0rph59, 2020-09-04:
Hi! Thank you for your bug report!

Could you please tell me which distro and what version of it you are using? And what is your current desktop environment?

> I have followed the README instructions regarding PulseAudio. However, I'm not sure this was necessary. The README says it needs to be done If you want Eruption to be able to play back sound [...]

If you only want the media keys and volume control knob/mute button to work, then this was not necessary. Eruption translates rotation of the volume control knob to key press events that subsequently need to be handled by your desktop environment. Eruption currently does not adjust the system volume.

bthiebault, 2020-09-04:
I use Arch.

I have no DE, only a WM, namely BSPWM with SXHKD. What are the key pressed events I need to listen to to translate to volume modification?

X3n0m0rph59, 2020-09-04:
The respective events are:

code 113 (KEY_MUTE)
code 114 (KEY_VOLUMEDOWN)
code 115 (KEY_VOLUMEUP)

You can see the exact key codes that are emitted by the `Eruption Virtual Keyboard` device using the `evtest` utility.

closed 2020-09-04 by bthiebault