eruption-project/eruption #20: Eruption crash on startup

bthiebault, 2020-09-11:
Maybe this is related to #17 
Eruption crashed on startup with the following message:

```
-- Logs begin at Sat 2019-07-20 18:20:28 CEST, end at Fri 2020-09-11 15:09:45 CEST. --
Sep 11 13:09:46 Solgaleo systemd[1]: Starting Driver for ROCCAT Vulcan 100/12x series keyboards...
Sep 11 13:09:46 Solgaleo systemd[1]: Started Driver for ROCCAT Vulcan 100/12x series keyboards.
Sep 11 13:09:46 Solgaleo eruption[599]:  INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.15
Sep 11 13:09:46 Solgaleo eruption[599]:  INFO  eruption > Loading saved state...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption > Registering plugins...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                       > Plugins loaded and initialized successfully
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                       > Enumerating connected devices...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::hwdevices            > Found Control interface: "0007:0003:01": ROCCAT Vulcan AIMO
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::hwdevices            > Found LED interface: "0007:0003:03": ROCCAT Vulcan AIMO
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::hwdevices::roccat_vulcan > Bound driver: ROCCAT Vulcan
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Device enumeration completed
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Spawning keyboard input thread...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Mouse support is DISABLED by configuration
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Waiting for tasks to complete...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Opening keyboard device...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugins::keyboard        > Now listening on: /dev/input/event25
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugins::keyboard        > Input device name: "ROCCAT ROCCAT Vulcan AIMO System Control"
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugins::keyboard        > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugins::keyboard        > Physical location: usb-0000:0b:00.3-2/input1
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugins::keyboard        > Grabbing the keyboard device exclusively
Sep 11 13:09:47 Solgaleo eruption[599]:  ERROR eruption::plugins::keyboard        > Fatal: Keyboard device went away: No such device (os error 19)
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Initializing keyboard device...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Configuring keyboard LEDs...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Opening mouse device...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > This mouse is currently not supported, falling back to the generic driver
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Performing late initializations...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Initializing D-Bus API...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Late initializations completed
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Startup completed
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Switching to profile: profile1.profile
Sep 11 13:09:47 Solgaleo eruption[599]:  ERROR eruption                           > Channel error: channel is empty and sending half is closed
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption::plugins::persistence     > Storing persistent state data to disk...
Sep 11 13:09:47 Solgaleo eruption[599]:  INFO  eruption                           > Closing devices...
Sep 11 13:09:48 Solgaleo eruption[599]:  INFO  eruption                           > Exiting now
Sep 11 13:09:48 Solgaleo systemd[1]: eruption.service: Succeeded.
```

I could not reproduce the bug, so I guess this was some sort of race condition that rarely happen. I just wanted to list it here for future reference

X3n0m0rph59, 2020-09-11:
Hi, thanks for your bug report!

I can confirm that there is a startup issue still lurking somewhere. The exact same crash happened to me yesterday. It seems to be very rare though. I will look into it soon...

X3n0m0rph59, 2020-11-12:
This bug seems to have vanished. I never was able to reproduce it. Please reopen this issue if the problem persists!

closed 2020-11-12 by X3n0m0rph59