eruption-project/eruption #21: Keyboard stops responding, panics after I press a button on a second keyboard (Ryos).

GreatGlobox, 2020-10-07:
Eruption is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 Eruption is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

 INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.16
 INFO  eruption > Loading saved state...
 WARN  eruption > Could not parse state file: Could not load global runtime state: configuration file "/var/lib/eruption/eruption.state" not found
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 WARN  eruption                       > Could not load persisted state: No such file or directory (os error 2)
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 0
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 1
 INFO  eruption::hwdevices            > Found Control interface: "0005:0003:01": ROCCAT Vulcan 100 AIMO
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 2
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 3
 INFO  eruption::hwdevices            > Found LED interface: "0005:0003:03": ROCCAT Vulcan 100 AIMO
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 0
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 1
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 2
 DEBUG eruption::hwdevices            > Device: C-Media Electronics Inc. USB Audio Device, interface: 3
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Ryos MK FX, interface: 0
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Ryos MK FX, interface: 1
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Ryos MK FX, interface: 2
 INFO  eruption::hwdevices::roccat_vulcan > Bound driver: ROCCAT Vulcan
 INFO  eruption                           > Device enumeration completed
 INFO  eruption                           > Spawning keyboard input thread...
 INFO  eruption                           > Spawning mouse input thread...
 INFO  eruption                           > Waiting for tasks to complete...
 INFO  eruption                           > Opening keyboard device...
 INFO  eruption                           > Initializing keyboard device...
 INFO  eruption::plugins::keyboard        > Now listening on: /dev/input/event19
 INFO  eruption::plugins::keyboard        > Input device name: "ROCCAT ROCCAT Vulcan 100 AIMO"
 INFO  eruption::plugins::keyboard        > Input device ID: bus 0x3 vendor 0x1e7d product 0x307a
 INFO  eruption::plugins::keyboard        > Physical location: usb-0000:0f:00.3-4/input0
 INFO  eruption::plugins::keyboard        > Grabbing the keyboard device exclusively
 INFO  eruption::plugins::mouse           > Now listening on: /dev/input/event20
 INFO  eruption::plugins::mouse           > Input device name: "ROCCAT ROCCAT Ryos MK FX"
 INFO  eruption::plugins::mouse           > Input device ID: bus 0x3 vendor 0x1e7d product 0x2fda
 INFO  eruption::plugins::mouse           > Physical location: usb-0000:02:00.0-3/input0
 INFO  eruption::plugins::mouse           > Grabbing the mouse device exclusively
thread 'events/mouse' panicked at 'called `Result::unwrap()` on an `Err` value: Could not map an evdev event code to a key or button

Location:
    /rustc/04488afe34512aa4c33566eb16d8c912a3ae04f9/src/libcore/convert/mod.rs:560:9', eruption/src/plugins/mouse.rs:208:80
stack backtrace:
   0:     0x559b98d37438 - backtrace::backtrace::libunwind::trace::h14d338b30b3ea0a7
                               at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/libunwind.rs:86
   1:     0x559b98d37438 - backtrace::backtrace::trace_unsynchronized::h73ea91d74a3fd67f
                               at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/mod.rs:66
   2:     0x559b98d37438 - std::sys_common::backtrace::_print_fmt::hd42948c952866e12
                               at src/libstd/sys_common/backtrace.rs:78
   3:     0x559b98d37438 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::ha8f928866ff7571e
                               at src/libstd/sys_common/backtrace.rs:59
   4:     0x559b98bbcf9c - core::fmt::write::he0c1e5f7426d2718
                               at src/libcore/fmt/mod.rs:1076
   5:     0x559b98d36b06 - std::io::Write::write_fmt::hf3afc6cfd57d0033
                               at src/libstd/io/mod.rs:1537
   6:     0x559b98d3650a - std::sys_common::backtrace::_print::hfc0110703f3696fd
                               at src/libstd/sys_common/backtrace.rs:62
   7:     0x559b98d3650a - std::sys_common::backtrace::print::h3f77c6990ddfaa22
                               at src/libstd/sys_common/backtrace.rs:49
   8:     0x559b98d3650a - std::panicking::default_hook::{{closure}}::heae49580a8d62d75
                               at src/libstd/panicking.rs:198
   9:     0x559b98d35d91 - std::panicking::default_hook::hecc34e3f729e213c
                               at src/libstd/panicking.rs:217
  10:     0x559b98d35d91 - std::panicking::rust_panic_with_hook::he82f5d0644692441
                               at src/libstd/panicking.rs:526
  11:     0x559b98d35a1a - rust_begin_unwind
                               at src/libstd/panicking.rs:437
  12:     0x559b98bba860 - core::panicking::panic_fmt::h09c929f06bb87c98
                               at src/libcore/panicking.rs:85
  13:     0x559b98bbdf92 - core::option::expect_none_failed::h188f17af6c9f404b
                               at src/libcore/option.rs:1269
  14:     0x559b98ad4c6a - eruption::spawn_mouse_input_thread::{{closure}}::h955630c5936e7322
  15:     0x559b98a32871 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h39d628bf9fbdc4a8
  16:     0x559b98d497f5 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::hd2b3bc04af94a84f
                               at /rustc/04488afe34512aa4c33566eb16d8c912a3ae04f9/src/liballoc/boxed.rs:1081
  17:     0x559b98d497f5 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::h1044417e186e567a
                               at /rustc/04488afe34512aa4c33566eb16d8c912a3ae04f9/src/liballoc/boxed.rs:1081
  18:     0x559b98d497f5 - std::sys::unix::thread::Thread::new::thread_start::h276e6ca033938925
                               at src/libstd/sys/unix/thread.rs:87
  19:     0x7fa914325ef7 - start_thread
                               at /home/steo/Downloads/glibc-2.31/nptl/pthread_create.c:477
  20:     0x7fa91423b23f - __GI___clone
                               at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
  21:                0x0 - <unknown>
Aborted

X3n0m0rph59, 2020-10-07:
Hi! Thank you for submitting this bug report!

This seems to be a problem with the device selection code for the mouse device. It wrongly selects your `ROCCAT Ryos MK FX` instead of the `Cooler Master Technology Inc. MM710 Gaming Mouse` device.

Could you please try to change `/etc/eruption/eruption.conf`:

```toml
[global]
# ...

# Mouse handling
enable_mouse = false
grab_mouse = false
```

GreatGlobox, 2020-10-07:
Thanks, I just tried that, then unplugged the Ryos and tried to run it again. It seems to say that it finds the keyboard, but then just halts again while it doesn't respond.

 INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.16
 INFO  eruption > Loading saved state...
 WARN  eruption > Could not parse state file: Could not load global runtime state: configuration file "/var/lib/eruption/eruption.state" not found
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 WARN  eruption                       > Could not load persisted state: No such file or directory (os error 2)
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 0
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 1
 INFO  eruption::hwdevices            > Found Control interface: "0005:0003:01": ROCCAT Vulcan 100 AIMO
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 2
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 3
 INFO  eruption::hwdevices            > Found LED interface: "0005:0003:03": ROCCAT Vulcan 100 AIMO
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 0
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 1
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 2
 DEBUG eruption::hwdevices            > Device: C-Media Electronics Inc. USB Audio Device, interface: 3
 INFO  eruption::hwdevices::roccat_vulcan > Bound driver: ROCCAT Vulcan
 INFO  eruption                           > Device enumeration completed
 INFO  eruption                           > Spawning keyboard input thread...
 INFO  eruption                           > Mouse support is DISABLED by configuration
 INFO  eruption                           > Waiting for tasks to complete...
 INFO  eruption                           > Opening keyboard device...
 INFO  eruption                           > Initializing keyboard device...
 INFO  eruption::plugins::keyboard        > Now listening on: /dev/input/event19
 INFO  eruption::plugins::keyboard        > Input device name: "ROCCAT ROCCAT Vulcan 100 AIMO"
 INFO  eruption::plugins::keyboard        > Input device ID: bus 0x3 vendor 0x1e7d product 0x307a
 INFO  eruption::plugins::keyboard        > Physical location: usb-0000:0f:00.3-4/input0
 INFO  eruption::plugins::keyboard        > Grabbing the keyboard device exclusively

X3n0m0rph59, 2020-10-07:
Could you please try to update the firmware of your Vulcan keyboard to the latest revision by using the ROCCAT Swarm software?

GreatGlobox, 2020-10-07:
Since I don't have Windows, I'll have to get someone else's computer in order to do so, but I'll try this and then get back to you.

GreatGlobox, 2020-10-08:
After the firmware update, it throws me a Lua error:

 INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.16
 INFO  eruption > Loading saved state...
 WARN  eruption > Could not parse state file: Could not load global runtime state: configuration file "/var/lib/eruption/eruption.state" not found
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 WARN  eruption                       > Could not load persisted state: No such file or directory (os error 2)
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 0
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 1
 DEBUG eruption::hwdevices            > Device: Cooler Master Technology Inc. MM710 Gaming Mouse, interface: 2
 DEBUG eruption::hwdevices            > Device: C-Media Electronics Inc. USB Audio Device, interface: 3
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 0
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 1
 INFO  eruption::hwdevices            > Found Control interface: "0001:0007:01": ROCCAT Vulcan 100 AIMO
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 2
 DEBUG eruption::hwdevices            > Device: ROCCAT ROCCAT Vulcan 100 AIMO, interface: 3
 INFO  eruption::hwdevices            > Found LED interface: "0001:0007:03": ROCCAT Vulcan 100 AIMO
 INFO  eruption::hwdevices::roccat_vulcan > Bound driver: ROCCAT Vulcan
 INFO  eruption                           > Device enumeration completed
 INFO  eruption                           > Spawning keyboard input thread...
 INFO  eruption                           > Mouse support is DISABLED by configuration
 INFO  eruption                           > Waiting for tasks to complete...
 INFO  eruption                           > Opening keyboard device...
 INFO  eruption                           > Initializing keyboard device...
 INFO  eruption::plugins::keyboard        > Now listening on: /dev/input/event19
 INFO  eruption::plugins::keyboard        > Input device name: "ROCCAT ROCCAT Vulcan 100 AIMO"
 INFO  eruption::plugins::keyboard        > Input device ID: bus 0x3 vendor 0x1e7d product 0x307a
 INFO  eruption::plugins::keyboard        > Physical location: usb-0000:02:00.0-10/input0
 INFO  eruption::plugins::keyboard        > Grabbing the keyboard device exclusively
 INFO  eruption                           > Configuring keyboard LEDs...
 INFO  eruption                           > Opening mouse device...
 INFO  eruption                           > This mouse is currently not supported, falling back to the generic driver
 INFO  eruption                           > Keyboard device firmware revision: "136"
 INFO  eruption                           > Performing late initializations...
 INFO  eruption                           > Initializing D-Bus API...
 DEBUG eruption                           > Registering filesystem watcher...
 INFO  eruption                           > Late initializations completed
 INFO  eruption                           > Startup completed
 DEBUG eruption                           > Entering the main loop now...
 INFO  eruption                           > Switching to profile: profile1.profile
 DEBUG eruption::scripting::script        > Applying parameter Bool { name: "mouse_events", description: "React on mouse events", default: true }
 DEBUG eruption::scripting::script        > Applying parameter Color { name: "color_mouse_click_flash", description: "Color of flash effect shown on mouse button clicks", default: 553626282 }
 DEBUG eruption::scripting::script        > Applying parameter Color { name: "color_mouse_wheel_flash", description: "Color of flash effect shown on mouse wheel scroll", default: 1084926719 }
 DEBUG eruption::scripting::script        > Applying parameter Color { name: "color_afterglow", description: "Afterglow color", default: 4294967295 }
 DEBUG eruption::scripting::script        > Applying parameter Color { name: "color_step_afterglow", description: "Color value that gets subtracted from afterglow every [n] frames", default: 134217728 }
[... cut ...]
        no file '/usr/local/share/lua/5.1/themes/default/init.lua'
        no file '/usr/share/eruption/scripts/lib/themes/default'
        no file '/usr/share/eruption/scripts/lib/themes/default.lua'
        no file './themes/default.so'
        no file '/usr/local/lib/lua/5.1/themes/default.so'
        no file '/usr/local/lib/lua/5.1/loadall.so'
stack traceback:
        ok: in ?
        ok: in function 'require'
        ok:58: in main chunk
        UnknownError
 ERROR eruption                           > Script execution failed
Aborted

X3n0m0rph59, 2020-10-08:
It is looking for `/usr/share/eruption/scripts/lib/themes/default.lua`. Could you please double check that this file exists in your filesystem? The directory has to contain `gaming.lua` as well.

```sh
$ ls /usr/share/eruption/scripts/lib/themes/                                                                                                                                                            
default.lua  gaming.lua
```

And please be sure that you don't run a debug build, it has to be compiled with `cargo build --release`. Otherwise the default include search paths would differ.

GreatGlobox, 2020-10-08:
Oh, that seems to be what the issue was. For some reason, my system had created /usr/share/eruption/scripts/lib/themes as a file instead of a directory, I guess I had made a mistake while copying the files. Anyway, it works now, thanks a lot!

X3n0m0rph59, 2020-10-08:
Great that it works now! Please feel free to open up new issues and feature requests anytime!

closed 2020-10-08 by X3n0m0rph59