eruption-project/eruption #22: Roccat Kova AIMO Mouse...Supported?

erockce, 2020-10-20:
I just purchased this RGB mouse to match my Vulcan keyboard.

The lighting on the wheel and feet shift through colors between red, yellow, and orange, and on click of the left side back extra button changes the color briefly to blue, but otherwise doesn't appear to match anything with the keyboard despite the profile selected. Some of the newest profiles containing `mouse` like `swirl-voronoi.profile` move colors around on the keyboard when the mouse changes position - neat! Other than the movement with `swirl-voroni`, the only color coordination from the mouse is when either primary and secondary mouse button is clicked or wheel rotated, the keyboard lights briefly "blinks".

Is there a profile to shift the RGB lights on the mouse like the keyboard? Or do you plan to support this AIMO mouse in the future?

Here's my information

```
$ dpkg -s eruption-roccat-vulcan-git | grep -E '^Package|^Version'
Package: eruption-roccat-vulcan-git
Version: 0.1.16-ubuntu2
```

`/etc/eruption.conf` (for mouse)
```
$ grep mouse /etc/eruption/eruption.conf 
enable_mouse = true
grab_mouse = false
```
* Assumed to be configured properly. From interpreting the information in the `eruption.conf` man page are these options mutually exclusive?

Eruption `journalctl` log on (re)start
```
Oct 19 22:33:51 <hostname redacted> eruption[30367]:  INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.16
Oct 19 22:33:51 <hostname redacted> eruption[30367]:  INFO  eruption > Loading saved state...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption > Registering plugins...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                       > Plugins loaded and initialized successfully
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                       > Enumerating connected devices...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::hwdevices            > Found Control interface: "0002:000d:01": ROCCAT Vulcan AIMO
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::hwdevices            > Found LED interface: "0002:000d:03": ROCCAT Vulcan AIMO
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::hwdevices::roccat_vulcan > Bound driver: ROCCAT Vulcan
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Device enumeration completed
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Spawning keyboard input thread...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Spawning mouse input thread...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Waiting for tasks to complete...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Opening keyboard device...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::keyboard        > Now listening on: /dev/input/event3
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::keyboard        > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::keyboard        > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::keyboard        > Physical location: usb-0000:00:14.0-3.4.3/input0
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::keyboard        > Grabbing the keyboard device exclusively
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::mouse           > Now listening on: /dev/input/event9
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::mouse           > Input device name: "ROCCAT ROCCAT Kova Aimo Mouse"
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::mouse           > Input device ID: bus 0x3 vendor 0x1e7d product 0x2cf1
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption::plugins::mouse           > Physical location: usb-0000:00:14.0-3.4.4/input0
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Initializing keyboard device...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Configuring keyboard LEDs...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Opening mouse device...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > This mouse is currently not supported, falling back to the generic driver
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Keyboard device firmware revision: "136"
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Performing late initializations...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Initializing D-Bus API...
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Late initializations completed
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Startup completed
Oct 19 22:33:52 <hostname redacted> eruption[30367]:  INFO  eruption                           > Switching to profile: my-custom-colors.profile
Oct 19 22:44:13 <hostname redacted> eruption[30367]:  INFO  eruption                           > Switching to profile: my-profile3.profile
```
Unsure if the message..."`This mouse is currently not supported, falling back to the generic driver`" is an indicator of limited functionality.  This new mouse is the only one plugged in now.

Relevant output from `usb-devices`:
```
T:  Bus=02 Lev=03 Prnt=12 Port=02 Cnt=01 Dev#= 13 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1e7d ProdID=3098 Rev=01.00
S:  Manufacturer=ROCCAT
S:  Product=ROCCAT Vulcan AIMO
C:  #Ifs= 4 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
I:  If#= 2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
I:  If#= 3 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs

T:  Bus=02 Lev=03 Prnt=12 Port=03 Cnt=02 Dev#= 14 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1e7d ProdID=2cf1 Rev=01.00
S:  Manufacturer=ROCCAT
S:  Product=ROCCAT Kova Aimo
C:  #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=01 Driver=usbhid
```

I am still using Ubuntu 18.04.5 at the moment.

Sorry for the upfront data dump but I figured the more you have initially the better.

Is there something else I need configured?

I admit that I have yet to use Swarm to update mouse firmware. I am currently having an issue in the Windows VM when I give Windows control of the Roccat mouse with Swarm running, the cursor pointer disappears. I cannot use keyboard hot keys to navigate to the mouse icon to update.

X3n0m0rph59, 2020-10-20:
Hi! Thanks for the excellent and detailed issue report! This mouse is currently not supported, but I am definitely interested in adding support for it to Eruption. Since I am lacking the hardware it would be really nice if you could send me some packet dumps of the initialization sequence of your Kova.

>Unsure if the message..."This mouse is currently not supported, falling back to the generic driver" is an indicator of limited functionality. This new mouse is the only one plugged in now.

Yes, that means there only will be the basic HID device functionality, as configured by the kernel. No DPI switching or AIMO LEDs are supported by the generic driver code, as this requires a specific driver plugin for the Kova. The colors you are currently seeing are probably generated by the mouse hardware itself. 

The easiest way to update the firmware would be to plug in a second mouse device. This way you can use the second mouse while your Kova is routed to the Windows 10 VM.

If you are interested in helping out, I can write up a detailed HOWTO, maybe in a Wiki article, on how to save a packet dump of the device init sequence.

erockce, 2020-10-21:
@X3n0m0rph59 , of course I will help!

No need for a detailed HOWTO right now. I work in IT and have been using UNIX/Linux variants for 25 years, but I'm not a kernel device driver developer!

Please point me in the right direction to start (e.g. strace, lsof, etc) and will get you some data!

X3n0m0rph59, 2020-10-21:
>@X3n0m0rph59 , of course I will help!

Great! Glad to hear that!

Basically I need a *.pcapng file of the init sequence of the Kova, as sent by ROCCAT Swarm, as well as a sample where it sets the colors while in AIMO LED mode.

You need this:
* Loaded the `usbmon` kernel module
* A Windows 10 VM with ROCCAT Swarm installed
* Wireshark: Wireshark-qt works great
* Fully updated device firmware and latest Swarm software would be nice
* A second mouse or other pointing device would be nice

Please perform the following steps in roughly that order:

* Load the `usbmon` Linux kernel module that creates a virtual network device which allows for capturing USB traffic using Wireshark
```sh
$ sudo modprobe usbmon
```
* Start up Wireshark

* Boot the Windows 10 VM

* Start capturing the traffic on the correct USB bus, you may want to note what your keyboard device and mouse devices bus IDs are

* Enable USB forwarding of the Kova to the Windows 10 VM

* Start ROCCAT Swarm: This will sent the full initialization sequence to the forwarded Kova (please use Swarm factory settings for the Kova. The LED mode has to be AIMO Lighting)

* After the device has been initialized, you should see Swarm changing the colors on the Kova

* Capture a few seconds of these 'color setting' packets

* In Wireshark: Please save the capture as pcapng file

With the captured data, I should be able to reverse engineer a subset of the protocol and get the device initialized and the LEDs up and running.

Thanks in advance!

1drp, 2020-10-27:
Do you know if you'd be able to tweak that process to get what you needed from a bare metal Windows install?  I have a Kone AIMO (not Kova, but still an AIMO device), that I'd like to get the LEDs syncing with the keyboard like they do on Windows.  Similar to the report on Kova performance, some profiles will recognize mouse events.

X3n0m0rph59, 2020-10-28:
Hi! @1drp Yes, it seems that this is possible. Please have a look at https://wiki.wireshark.org/CaptureSetup/USB in section `Windows`. You would need to set-up https://desowin.org/usbpcap/ and Wireshark.
(I have not verified that this actually works, but chances are pretty high).
Please get back to me if you need assistance!

erockce, 2020-11-08:
Sorry this took a while to get to.  I just tried it out today.  Hopefully it has what you need.

I made the capture with only relevant devices like the ROCCAT KOVA mouse attached.  All other devices were generic, and I turned off the Eruption service.

ROCCAT Swarm
Software Version: 1.9379

ROCCAT KOVA
Firmware: 1.03

```
$ lsusb | grep -i roccat
Bus 002 Device 003: ID 1e7d:2cf1 ROCCAT
```

I performed the following steps:
1. Installed wireshark-qt
a. Added my user to `wireshark` group
b. Logged out and back in with new group membership
1. Started `usbmon` kernel module (`sudo modprobe usbmon`)
1. Started Wireshark QT
a. Set input capture option for only `usbmon2` (correspond to the bus with KOVA)
1. Created a test capture to ensure I was able to see packets from the mouse
a. Saved then closed the capture file
1. Started a new capture file
1. Start Windows VM
1. Start Wireshark capture
1. Logged in
1. Forwarded KOVA to Windows
1. Started Swarm
1. Caused a RESET to defaults for the mouse
1. Changed color settings
a. Changed brightness
b. Changed color theme
1. Caused a RESET to defaults for the mouse
1. Stopped Swarm
1. Stopped Wireshark package capture

How should I get the capture to you? I tried direct upload, then zip, and that doesn't work either.  The compressed size is only 217k.

X3n0m0rph59, 2020-11-09:
Hi @erockce, that sounds great! With the packet capture I should be able to implement support for your device.

It would be nice if you could attach the capture file to a message in this thread. Alternatively you could upload it to a cloud storage/file hoster and then just post a link to it here.

> How should I get the capture to you? I tried direct upload, then zip, and that doesn't work either.

If your browser is unable to read the capture file, it could be that this is due to an AppArmor profile blocking access to it. It should work if you paste a file from within the /home/user/Downloads directory. (Copy the *.pcap file to ~/Downloads prior to pasting it here)

Next steps:

* I will implement device initialization code within `eruption-debug-tool` using your capture file
* I will send you a build of  `eruption-debug-tool` so that you can verify that the new code works
* After you confirmed that all worked out right, I will port the code over to the `Eruption` daemon

For further information, please see this thread: #25

Thank you very much! Your help is highly appreciated!

erockce, 2020-11-11:
[usb-roccat-kova-winvm-swarm-reset-test-color-set-reset-capture-erockce-2020NOV08.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5520911/usb-roccat-kova-winvm-swarm-reset-test-color-set-reset-capture-erockce-2020NOV08.zip)

I'm embarassed :(  I could not upload before because an associated Github site was blocked by default by uMatrix.

You should see the attached zip file now.  It is a compressed `.pcapng` file. That was the default from Wireshark QT on Ubuntu 18.04.5. Let me know if you need a different format.

X3n0m0rph59, 2020-11-11:
Thank you very much! This looks good!

I have implemented a driver sub-module in `eruption-debug-tool`.

* It seems like the init sequence of the ROCCAT Kova AIMO is much like that of the ROCCAT Kone Aimo.
* It seems to have two distinct configurable colors
* There is a major difference in the way the colors are set. The Kova AIMO is using a completely different feature report there

Please run the attached `eruption-debug-tool` version `0.0.4` like this:

```sh
$ sudo ./eruption-debug-tool list
```
Note the index of your mouse device, sub-device 0.

```sh
$ sudo ./eruption-debug-tool run-tests <index>
```

If everything works out right, it should show first red then blue lighting effects.

Please mask and stop the `eruption.service` prior to running the attached `eruption-debug-tool`. If your mouse configuration becomes garbled in any way, you should be able to reset it to factory defaults using ROCCAT Swarm. Maybe you need a second mouse device to do this.

[eruption-debug-tool.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5521414/eruption-debug-tool.zip)

erockce, 2020-11-12:
In #25, @1drp mentioned cycling through buttons and the wheel.  I didn't do that in my trace.  Do you need this information to get all the buttons working?

Using `evtest` it seems that some button events are spread across multiple `/dev/input/event*` devices (sorry I'm not a Linux kernel device expert, just have learned enough to get by).

Device 15:
```
$ sudo evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Sleep Button
...
/dev/input/event9:	ROCCAT ROCCAT Vulcan AIMO
/dev/input/event13:	ROCCAT ROCCAT Vulcan AIMO
/dev/input/event15:	ROCCAT ROCCAT Kova Aimo Mouse
...
/dev/input/event17:	ROCCAT ROCCAT Kova Aimo Consumer Control
...
/dev/input/event19:	ROCCAT ROCCAT Kova Aimo
/dev/input/event20:	ROCCAT ROCCAT Kova Aimo
/dev/input/event21:	Eruption Virtual Keyboard
/dev/input/event22:	Eruption Virtual Mouse
Select the device event number [0-22]: 15
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x1e7d product 0x2cf1 version 0x111
Input device name: "ROCCAT ROCCAT Kova Aimo Mouse"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (?)
    Event code 12 (?)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
```
Device 19
```
$ sudo evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Sleep Button
...
/dev/input/event9:	ROCCAT ROCCAT Vulcan AIMO
/dev/input/event13:	ROCCAT ROCCAT Vulcan AIMO
/dev/input/event15:	ROCCAT ROCCAT Kova Aimo Mouse
...
/dev/input/event17:	ROCCAT ROCCAT Kova Aimo Consumer Control
...
/dev/input/event19:	ROCCAT ROCCAT Kova Aimo
/dev/input/event20:	ROCCAT ROCCAT Kova Aimo
/dev/input/event21:	Eruption Virtual Keyboard
/dev/input/event22:	Eruption Virtual Mouse
Select the device event number [0-22]: 19
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x1e7d product 0x2cf1 version 0x111
Input device name: "ROCCAT ROCCAT Kova Aimo"
Supported events:
  Event type 0 (EV_SYN)
  Event type 3 (EV_ABS)
    Event code 40 (ABS_MISC)
[... 132 lines cut ...]
    Event code 125 (KEY_LEFTMETA)
    Event code 126 (KEY_RIGHTMETA)
    Event code 127 (KEY_COMPOSE)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
  Event type 17 (EV_LED)
    Event code 0 (LED_NUML) state 0
    Event code 1 (LED_CAPSL) state 0
    Event code 2 (LED_SCROLLL) state 0
    Event code 3 (LED_COMPOSE) state 0
    Event code 4 (LED_KANA) state 0
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
```
(This appears to be keyboard, yet the device is named "`Kova Mouse`".

Is this possibly because the default Linux kernel mouse driver (or your virtual) is not fully recognizing my Kova mouse and having 9 buttons (including the DPI switch)?

Beyond the primary, secondary, middle, and wheel buttons I can use the two outer buttons for scroll up and down in applications. With the exception of the DPI switch, one button briefly switches the wheel color to blue but other than the buttons mentioned the rest are non-functional.

erockce, 2020-11-12:
By the way I recently updated packages and now have Eruption 0.1.17.

I tried using the debug tool without arguments and having glib dependency issues.
```
$ sudo ./eruption-debug-tool list
./eruption-debug-tool: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./eruption-debug-tool)
```
```
$ ldd ./eruption-debug-tool 
./eruption-debug-tool: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./eruption-debug-tool)
	linux-vdso.so.1 (0x00007ffe56be8000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f534fae2000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f534f8ca000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f534f6ab000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f534f30d000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f534f109000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f534ed18000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f534eafa000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f534ff28000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f534e8f2000)
```

This is Ubuntu 18.04.  I plan to upgrade to 20 maybe in a couple more weeks.

X3n0m0rph59, 2020-11-12:
Hi @erockce, 

> In #25, @1drp mentioned cycling through buttons and the wheel. I didn't do that in my trace. Do you need this information to get all the buttons working?

No, this information is not strictly necessary to write a driver sub-module. Only the device initialization sequence, as well as setting of the colors when the device is in AIMO mode are strictly required.
(To get DPI switch notifications Eruption needs to listen to USB HID events though, for which I maybe need further data gathered via the `eruption-debug-tool`, not via Wireshark)

> Using evtest it seems that some button events are spread across multiple /dev/input/event* devices (sorry I'm not a Linux kernel device expert, just have learned enough to get by).

These are all Linux input subsystem (evdev) devices belonging to the Kova mouse. They serve different purposes:

**Device 15**

This is the mouse device that sends relative motion events as well as button press events, except for the DPI switch. The DPI switch button is only accessible via the USB HID layer. The Linux input subsystem (evdev) does not emit or pass through events originating from the DPI switch button.

**Device 19**

This one sends pointer motion events in absolute coordinate space. Don't know much more to say about it.

**Device 17**

Used for hardware assisted macros as well as Easy Shift+. I believe this one can be used to control wireless hardware, e.g. to switch the state of the radio as well. 

**Device 20**

Used for hardware assisted macros as well as Easy Shift+. Like device 17, this is a keyboard device of your Kova mouse from which macros are send to the system. Like device 17, Eruption neither uses nor implements this. It is all done in software to gain more flexibility than what can be achieved by using the hardware based macro mechanism.

Sorry for the disfunctional build, I will rebuild the `eruption-debug-tool` in a Ubuntu container and post another binary...

X3n0m0rph59, 2020-11-12:
@erockce This one should link to the correct libc:

[eruption-debug-tool.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5528430/eruption-debug-tool.zip)

erockce, 2020-11-14:
```
$ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
Created symlink /etc/systemd/system/eruption.service → /dev/null.
$ sudo ./eruption-debug-tool list
...
Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other subcommands of this tool

Index: 00: ID: 1e7d:2cf1 ROCCAT/ROCCAT Kova Aimo Subdev: 0
Index: 01: ID: 1e7d:2cf1 ROCCAT/ROCCAT Kova Aimo Subdev: 1
Index: 02: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 0
Index: 03: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 1
Index: 04: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 2
Index: 05: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 3

Enumeration completed
```

Index 00: [eruption-debug-tool_index00.txt](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5540304/eruption-debug-tool_index00.txt)

The colors changed from Red to Blue, and stayed blue after the test finished.
Mouse is responsive and usable as expected (e.g. no corruption or lockups)

For a clean test between index numbers, I unplugged/plugged in the mouse and colors/theme resumed to default.

Index 01: [eruption-debug-tool_index01.txt](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5540305/eruption-debug-tool_index01.txt)

Output is mostly similar except for a few lines...

```diff
$ diff -u eruption-debug-tool_index0*
--- eruption-debug-tool_index00.txt	2020-11-13 23:39:09.365454038 -0700
+++ eruption-debug-tool_index01.txt	2020-11-13 23:39:28.929149470 -0700
@@ -1,6 +1,6 @@
-$ sudo ./eruption-debug-tool run-tests 00
+$ sudo ./eruption-debug-tool run-tests 01
 ...
-Index: 00: ID: 1e7d:2cf1 ROCCAT/ROCCAT Kova Aimo Subdev: 0
+Index: 01: ID: 1e7d:2cf1 ROCCAT/ROCCAT Kova Aimo Subdev: 1
 Bound driver: ROCCAT Kova AIMO
 Sending device init sequence...
 Sending control device feature report
@@ -24,9 +24,7 @@
   |09080100 00000000|                   ........         00000000
                                                          00000008
 Waiting for control device to respond...
-  |040300cf|                            ....             00000000
-                                                         00000004
-  |040300cf|                            ....             00000000
+  |04000000|                            ....             00000000
                                                          00000004
   |040300cf|                            ....             00000000
                                                          00000004
```

I hope this is helpful information.

Thank you again for implementing some functionality for this device!

X3n0m0rph59, 2020-11-14:
@erockce Excellent! Thank you very much for trying things out!

So the next release of Eruption will ship with support for the ROCCAT Kova AIMO.

X3n0m0rph59, 2020-11-14:
To enable detection of click events of the DPI switch button (within Eruption), I would need the originating USB HID subdevice and the byte sequence that gets sent as the payload of the interrupt. Please note that this is not strictly needed for the DPI switch feature of your device to work, as switching DPI profiles is handled by the mouse hardware already. It would just complement the feature set, so that Eruption may display a visual notification when the DPI setting is switched via the respective mouse button. It would be great, if you could find the time to gather this information!

To gather the required data, you may use the `trace` subcommand of the `eruption-debug-tool` 

```sh
$ sudo eruption-debug-tool list

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 

Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other subcommands of this tool

Index: 00: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 0
Index: 01: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 1
Index: 02: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 2
Index: 03: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 3
Index: 04: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 0
Index: 05: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 1
Index: 06: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 2
Index: 07: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 3

Enumeration completed
```

By tracing each sub-device, I found that on my mouse, a ROCCAT Kone Pure Ultra, the subdevice 0 sends motion events as well as click events of the main buttons. The DPI switches (up/down) send interrupts originating from the third HID sub-device. So the originating sub-device index would be 2.

Example session:

```sh
$ sudo eruption-debug-tool trace 2
-- Start of trace --
Index: 02: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 2
Initializing...
8 bytes
  |04010000 00000000|                   ........         00000000
                                                         00000008
Entering polling loop:
Instant { tv_sec: 18520, tv_nsec: 853536368 }: 8 bytes
  |0300b004 00000000 00000000 00000000| ................ 00000000
                                                         00000010
Instant { tv_sec: 18521, tv_nsec: 907516478 }: 8 bytes
  |0300b005 00000000 00000000 00000000| ................ 00000000
                                                         00000010
Instant { tv_sec: 18524, tv_nsec: 161524453 }: 8 bytes
  |0300b004 00000000 00000000 00000000| ................ 00000000
                                                         00000010
Instant { tv_sec: 18524, tv_nsec: 391529991 }: 8 bytes
  |0300b003 00000000 00000000 00000000| ................ 00000000
                                                         00000010
Instant { tv_sec: 18524, tv_nsec: 576524540 }: 8 bytes
  |0300b002 00000000 00000000 00000000| ................ 00000000
                                                         00000010
Instant { tv_sec: 18524, tv_nsec: 752525964 }: 8 bytes
  |0300b001 00000000 00000000 00000000| ................ 00000000
                                                         00000010
Instant { tv_sec: 18525, tv_nsec: 825536526 }: 8 bytes
  |0300b002 00000000 00000000 00000000| ................ 00000000
                                                         00000010
Instant { tv_sec: 18527, tv_nsec: 994448139 }: 8 bytes
  |0300b001 00000000 00000000 00000000| ................ 00000000
                                                         00000010

```

The fourth byte encodes the active DPI slot, so this line: `|0300b004 00000000 00000000 00000000|` means that DPI slot 4 is the currently active one, where that line `|0300b001 00000000 00000000 00000000|` tells us that we are in the lowest DPI slot 1.

It would be really great if you could find the sub-device that reports changes of the DPI slot, as well as show the payload that gets send.

erockce, 2020-11-15:
Reports and traces attached.  My apologies if some of it is very redundant, but I wanted to make sure you had enough data.

Reports:
[eruption-debug-tool-report00.txt](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5543034/eruption-debug-tool-report00.txt)
[eruption-debug-tool-report01.txt](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5543035/eruption-debug-tool-report01.txt)

DPI trace:

[eruption-debug-tool-trace00-dpi-switch.txt](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5543029/eruption-debug-tool-trace00-dpi-switch.txt)
[eruption-debug-tool-trace01-dpi-switch.txt](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5543031/eruption-debug-tool-trace01-dpi-switch.txt)

Button (minus DPI) trace:

[eruption-debug-tool-trace00-buttons-only-no-dpi.txt](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5543032/eruption-debug-tool-trace00-buttons-only-no-dpi.txt)
[eruption-debug-tool-trace01-buttons-only-no-dpi.txt](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5543033/eruption-debug-tool-trace01-buttons-only-no-dpi.txt)

Not sure if this is news worthy but between debugging "trace" 0 and 1, I lost functionality (not because I was holding the mouse in the air to prevent movement events! :smile: ).  So I unplugged and re-plugged in the mouse between traces.

I double-clicked each button (minus DPI) for the traces.

Four of the buttons (bottom two on each side) are not recognized by the current driver or X. I think I questioned this before if I need to specify a custom configuration like having a manually configured `xorg.conf` again?  I thought that this was handled dynamically nowadays.

X3n0m0rph59, 2020-11-15:
Great, thank you very much!

So if I interpret things correctly the DPI switch events are sent via sub-device 0. I will add support for these USB HID events to the `ROCCAT Kova AIMO` driver within Eruption.

>Not sure if this is news worthy but between debugging "trace" 0 and 1, I lost functionality (not because I was holding the mouse in the air to prevent movement events! :smile: ). So I unplugged and re-plugged in the mouse between traces.

:smile: This happens on my system too. I did not have time to fix this bug...

>Four of the buttons (bottom two on each side) are not recognized by the current driver or X. I think I questioned this before if I need to specify a custom configuration like having a manually configured xorg.conf again? I thought that this was handled dynamically nowadays.

* Is this a bug within Eruption or are said mouse buttons non-functional, even with the Eruption daemon disabled?
* Did the clicks of the non-functional buttons generate any output in the `trace` run of `eruption-debug-tool`, either on sub-device 0 or sub-device 1? If this is the case, then I can add support for them.

To verify whether the kernel's input subsystem processes clicks of these buttons at all, you could try out `evtest`. The `eruption.service` has to be masked and disabled for this to work.

```sh
$ sudo evtest
```

Then please try out each `/dev/input/event??` evdev device that has the text `ROCCAT Kova AIMO ` in its name. If a button is clicked, it should show up in the output of `evtest` on at least one of the sub-devices.

I suspect that the non-functioning buttons trigger events not on the primary mouse device, but on an associated keyboard sub-device of the Kova. So an additional software layer is needed for this to work. I don't think that auto-configuration is supported for this kind of hardware. I recently added an experimental evdev-based "secondary sub-device" feature to Eruption, exactly to cover this kind of configuration. This is, as of now, completely untested and hidden behind a feature gate.

I suspect that `/dev/input/event20` (from the posts above) may be the device that sends the events for the non-functional buttons. It would be really great to see the output of `evtest` for said buttons!

Thanks in advance!

erockce, 2020-11-16:
Thank you very much for the explainations provided so far.

When I used `evtest` before and pressed the buttons, depending on the `event*` chosen, it results in either an event, a control character, or nothing at all.

It will be better for me to describe the button press tests with a picture

[image]

(Source from [here](https://images.app.goo.gl/LpCgeNNEv4tpLw9T6))

Here is testing the Kova `event*` devices.  I go through pressing each button (keeping the mouse lifted to prevent movement events).

```
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
...
/dev/input/event9:	ROCCAT ROCCAT Kova Aimo Mouse
/dev/input/event10:	ROCCAT ROCCAT Kova Aimo Consumer Control
/dev/input/event11:	ROCCAT ROCCAT Kova Aimo
..
/dev/input/event14:	ROCCAT ROCCAT Kova Aimo
/dev/input/event15:	ROCCAT ROCCAT Vulcan AIMO
/dev/input/event19:	ROCCAT ROCCAT Vulcan AIMO
/dev/input/event21:	Eruption Virtual Keyboard
/dev/input/event22:	Eruption Virtual Mouse
```

`event9`, I get as output:
```
Select the device event number [0-22]: 9
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x1e7d product 0x2cf1 version 0x111
Input device name: "ROCCAT ROCCAT Kova Aimo Mouse"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (?)
    Event code 12 (?)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1605505445.218797, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1605505445.218797, type 1 (EV_KEY), code 272 (BTN_LEFT), value 1
Event: time 1605505445.218797, -------------- SYN_REPORT ------------
Event: time 1605505445.371816, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1605505445.371816, type 1 (EV_KEY), code 272 (BTN_LEFT), value 0
Event: time 1605505445.371816, -------------- SYN_REPORT ------------
Event: time 1605505446.613796, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1605505446.613796, type 1 (EV_KEY), code 273 (BTN_RIGHT), value 1
Event: time 1605505446.613796, -------------- SYN_REPORT ------------
Event: time 1605505446.772821, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1605505446.772821, type 1 (EV_KEY), code 273 (BTN_RIGHT), value 0
Event: time 1605505446.772821, -------------- SYN_REPORT ------------
Event: time 1605505450.337733, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1605505450.337733, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 1
Event: time 1605505450.337733, -------------- SYN_REPORT ------------
Event: time 1605505450.538712, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1605505450.538712, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 0
[... 360 lines cut ...]
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
    Event code 277 (BTN_FORWARD)
    Event code 278 (BTN_BACK)
    Event code 279 (BTN_TASK)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 2 (REL_Z)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (?)
    Event code 12 (?)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
^[[5~^[[6~^C
```
When I pressed button 4 and 5, I get a "control character". Pressing button 7, changes the mouse wheel color briefly to blue on click. The rest of the buttons show nothing.

Let me know if you want more data!

X3n0m0rph59, 2020-11-20:
@erockce I just released version `0.1.18` of Eruption

Please note that I renamed the project, the PPA, and the Eruption (Ubuntu) package to accommodate for the recent changes in the direction of the project. Please see README.md for PPA and package names.

Eruption version `0.1.18` should display info messages in the system log, when the buttons on the Kova are clicked. It should dump hexadecimal output of the HID report payload. (Much like the debug tool did).  I ended up implementing support for the extra buttons on the USB HID layer instead of the Linux input (evdev) layer.

The button events are not yet hard-wired in the core daemon, so they still wont trigger any events besides printing hexdumps.

If all works out right and you can see said info messages when clicking the Buttons on your Kova, could you please create a list of `Button index/number` => `HID report payload (the hexdump)` mappings. With this information I will proceed implementing the missing functionality.

Thanks in advance!

X3n0m0rph59, 2020-11-22:
@erockce I have prepared a pre-release build of the upcoming Eruption `0.1.19` for Ubuntu Bionic, which _should_ support the extra mouse buttons on the Kova AIMO.

The code is mostly untested, since I am still lacking the required hardware. I had to make some guesses, so I hope that it all works out...

Please unzip the file prior to installing it:
[eruption-git_0.1.19-ubuntu0-pre_amd64.deb.zip](https://github.com/X3n0m0rph59/eruption/files/5579687/eruption-git_0.1.19-ubuntu0-pre_amd64.deb.zip)

erockce, 2020-11-30:
I finally had a chance to install the pre-release package. For now I'm still using Ubuntu 18.04. I uninstalled `eruption-roccat-vulcan-git` before installing `eruption-git_0.1.19-ubuntu0-pre_amd64.deb`.  Only file I allowed that was to be overwritten is `/etc/eruption/eruption.conf`. I took backups so it will be easy to replace my custom changes as needed.

After `eruption` was started there were no lights on the mouse. Then I switched keyboard profiles verifying prior settings were intact - my custom copied profile files in `/var/lib/eruption/profiles` were untouched.

Seems that by design and selection of modules some profiles do not vary the colors on the mouse. Not a complaint just an observation.

The colors are varying depending on the profile selected, looks great!

Here is additional button tests you requested with the `eruption-debug-tool` provided from the pre-release package.
 
```
$ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
$ sudo eruption-debug-tool list

 Please stop the Eruption daemon prior to running this tool:
...
Index: 00: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 0
Index: 01: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 1
Index: 02: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 2
Index: 03: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 3
Index: 04: ID: 1e7d:2cf1 ROCCAT/ROCCAT Kova Aimo Subdev: 0
Index: 05: ID: 1e7d:2cf1 ROCCAT/ROCCAT Kova Aimo Subdev: 1

Enumeration completed
```

Index 04:
[eruption-debug-kova-index-04-20201129-201326.txt](https://github.com/X3n0m0rph59/eruption/files/5613637/eruption-debug-kova-index-04-20201129-201326.txt)

1. Moved the mouse varying X and Y coordinates
2. Button sequence (refer to picture in post [above](#issuecomment-727754882)): 1, 2, 3, 3(roll up/down), 4, 5, 6, 7, 8, 9, 10

After escaping to finish the mouse stopped working requiring a power cycle (unplug/re-plug in) 

Kova returned to factory default color cycling and button 7 flashing blue when pressed.

Index 05:
[eruption-debug-kova-index-05-20201129-201709.txt](https://github.com/X3n0m0rph59/eruption/files/5613639/eruption-debug-kova-index-05-20201129-201709.txt)

1. Moved the mouse varying X and Y coordinates
2. Button sequence (refer to picture in post [above](#issuecomment-727754882)): 1, 2, 3, 3(roll up/down), 4, 5, 6, 7, 8, 9, 10

Note the "control character sequence" on the buttons without events.

Please let me know if you need more data!

erockce, 2020-11-30:
Ok, now something weird is happening after reactivating and using eruption. I can put in a separate issue, but figured since I have a pre-release, this version is "use at your own risk"!

After the restart of eruption, the mouse no longer works; no mouse click, lights, or cursor movement. So I manually unplugged and re-plugged back in but still have the same result.  I thought at first it was because I swapped the USB ports, but the same behavior still happens.

I tried different combinations of adjusting `enable_mouse` and `grab_mouse` but that did not make a difference.  To have the ability to use the mouse, I set both options to `false` for now.

Maybe the `eruption-debug-tool` has left the mouse in a strange state?

I have not restarted my computer or tried to re-initialize with SWARM via my Windows VM yet.

I can wait until you have the next official Eruption release and test again.

Just providing some information in case you had a thought to what might be happening.

X3n0m0rph59, 2020-11-30:
@erockce Thanks for testing the pre-release!

> After eruption was started there were no lights on the mouse. Then I switched keyboard profiles verifying prior settings were intact - my custom copied profile files in /var/lib/eruption/profiles were untouched.

> Seems that by design and selection of modules some profiles do not vary the colors on the mouse. Not a complaint just an observation.

This behavior is expected and by design! Previously the mouse "zone" of the LED color map completely overlapped with the "zone" of the keyboard (the numpad area was used for this). So the mouse simply reflected the color state of the keyboard. Since version `0.1.18`, the LED color map is now larger and each device has its own "zone" allocated on that color map. The mouse can now show colors which are independent from that of the keyboard. Each Lua script may decide on its own which parts of the unified LED color map it modifies.

This introduces a change to the way the Lua scripts should iterate over the color maps. The upper bounds of the loops need to be changed from `get_num_keys()` to `get_canvas_size()` since these are now distinct concepts. Otherwise the mouse will just "stay dark" since only the keyboard part of the color map will be modified.

Custom Lua scripts will need to be changed to add support for multiple devices. The vendor supplied scripts should be up-to-date by now. (I updated alle Lua scripts to use `canvas_size` where it made sense. They now only use `get_num_keys()` when just the keyboard "zone" should be modified)

So, new Lua code should look like this example:

```Lua
require "declarations"
-- ....
for i = 0, canvas_size do
    color_map[i] = rgba_to_color(r, g, b, lerp(0, 255, opacity)) -- just an example
end
```

Regarding the mouse buttons:

Does Eruption version `0.1.19-pre` show debug messages in the syslog, as soon as a button on the mouse is clicked?
I am especially looking for the `Macros: HID event (mouse):` messages.

You can enable debug messages like this, by running Eruption from a terminal:
```
$ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
$ sudo RUST_LOG=debug eruption
```
`enable_mouse` and `grab_mouse` have to be set to `true` in the configuration file! Please try to disable third party drivers that could grab the Linux input (evdev) or HID devices of the mouse exclusively.

> After the restart of eruption, the mouse no longer works; no mouse click, lights, or cursor movement. So I manually unplugged and re-plugged back in but still have the same result. I thought at first it was because I swapped the USB ports, but the same behavior still happens.

It should be fixed by rebooting your system. Otherwise, please load factory defaults using Swarm.

> Maybe the eruption-debug-tool has left the mouse in a strange state?

That is entirely possible, though I think that some part of the input stack got confused.

Please tell me whether Eruption `0.1.19-pre` is able to register the mouse button events. As said before, I am especially looking for the `Macros: HID event (mouse):` messages. They should at least get printed when changing the DPI level.

erockce, 2020-12-06:
Upgraded SWARM from `V1.9380` to `V1.9381`.

The KOVA Firmware is still `1.03`

I used SWARM in Windows VM to reset and set Kova to defaults again.

As far as Roccat devices I only have the KOVA plugged in during testing. 

```
$ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
Created symlink /etc/systemd/system/eruption.service → /dev/null.
```

```
$ cat /etc/eruption/eruption.conf | grep -v "#" | grep -v "^$"
[global]
profile_dir = "/var/lib/eruption/profiles/"
script_dir = "/usr/share/eruption/scripts/"
keyboard_variant = "ISO"
enable_mouse = true
grab_mouse = true
afk_profile = "rainbow-wave.profile"
afk_timeout_secs = 0
```

Ran
```
$ sudo RUST_LOG=debug eruption 2>&1 | tee eruption-cli-`date +%Y%m%d-%H%M%S`.txt
```
[eruption-cli-20201206-140403.txt](https://github.com/X3n0m0rph59/eruption/files/5649549/eruption-cli-20201206-140403.txt)

Referencing the picture in post [above](#issuecomment-727754882). Events were received from all except buttons 6, 7, and 10.

There were no messages from the `eruption` command in `/var/log/syslog`. Just messages from the kernel or other OS services.

thelyron, 2021-03-03:
> ```
> $ lsusb | grep -i roccat
> Bus 002 Device 003: ID 1e7d:2cf1 ROCCAT
> ```

Btw I have the same mouse in white. According from sniffing arround in the roccat-tools it seems each color has it's own USB ID.

```
Bus 003 Device 008: ID 1e7d:2cf3 ROCCAT ROCCAT Kova Aimo
```