eruption-project/eruption #23: Unplugging Mouse Causes Eruption to Panic "Could not get mouse events"

erockce, 2020-10-23:
In preparing to update the firmware for my Kova, I unplugged the mouse to re-route the mouse cable.  Then what I thought was unusual was the computer fans increasing in speed!  That's when I noticed in `conky` (and `top`) that `journald` had jumped to the highest CPU usage so I checked the messages. Eruption was spamming journald/syslog. 

Brief output from `journalctl -xe -u eruption`:
```
Oct 22 20:33:18 <hostname redacted> eruption[805]:  ERROR eruption::plugins::mouse           > Could not get mouse events
```
This line is continuously emitted to `journald`.

A similar message also appears in the `syslog`:
```
Oct 22 20:33:28 <hostname redacted> eruption[805]: message repeated 875956 times: [  ERROR eruption::plugins::mouse           > Could not get mouse events]
```

The work-around/resolution was to shutdown `eruption` service, update and set `enable_mouse = false`, and restart eruption.  Now when a mouse is unplugged, there isn't any spamming or message about getting mouse events.

I thought at first it was related to my new Kova mouse.  But I tested with my previous Logitech mouse and the same issue occurred.  With the Logitech there "seemed" like related additional information was being put in the `syslog`, but that maybe just unrelated noise from other services like `systemd`. 

Details from the `syslog`:
```
Oct 22 20:39:24 <hostname redacted> eruption[8621]: message repeated 18654 times: [  ERROR eruption::plugins::mouse           > Could not get mouse events]
Oct 22 20:39:24 <hostname redacted> eruption[8621]:  INFO  eruption::plugins::persistence     > Storing persistent state data to disk...
Oct 22 20:39:24 <hostname redacted> eruption[8621]:  INFO  eruption                           > Closing devices...
Oct 22 20:39:24 <hostname redacted> kernel: [ 1807.620924] input: ROCCAT ROCCAT Vulcan AIMO Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.4/2-3.4.3/2-3.4.3:1.1/0003:1E7D:3098.000E/input/input42
Oct 22 20:39:24 <hostname redacted> kernel: [ 1807.621620] input: ROCCAT ROCCAT Vulcan AIMO Consumer Control as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.4/2-3.4.3/2-3.4.3:1.1/0003:1E7D:3098.000E/input/input43
Oct 22 20:39:24 <hostname redacted> systemd[1]: Requested transaction contradicts existing jobs: Resource deadlock avoided
Oct 22 20:39:24 <hostname redacted> kernel: [ 1807.683262] input: ROCCAT ROCCAT Vulcan AIMO System Control as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.4/2-3.4.3/2-3.4.3:1.1/0003:1E7D:3098.000E/input/input44
Oct 22 20:39:24 <hostname redacted> kernel: [ 1807.683632] hid-generic 0003:1E7D:3098.000E: input,hiddev0,hidraw2: USB HID v1.11 Mouse [ROCCAT ROCCAT Vulcan AIMO] on usb-0000:00:14.0-3.4.3/input1
Oct 22 20:39:24 <hostname redacted> kernel: [ 1807.687696] input: ROCCAT ROCCAT Vulcan AIMO as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.4/2-3.4.3/2-3.4.3:1.3/0003:1E7D:3098.000F/input/input46
Oct 22 20:39:24 <hostname redacted> kernel: [ 1807.747782] hid-generic 0003:1E7D:3098.000F: input,hidraw4: USB HID v1.11 Device [ROCCAT ROCCAT Vulcan AIMO] on usb-0000:00:14.0-3.4.3/input3
Oct 22 20:39:24 <hostname redacted> systemd[1]: message repeated 9 times: [ Requested transaction contradicts existing jobs: Resource deadlock avoided]
Oct 22 20:39:24 <hostname redacted> eruption[8621]:  INFO  eruption                           > Exiting now
Oct 22 20:39:24 <hostname redacted> systemd[1]: Requested transaction contradicts existing jobs: Resource deadlock avoided
Oct 22 20:39:24 <hostname redacted> systemd[1]: message repeated 4 times: [ Requested transaction contradicts existing jobs: Resource deadlock avoided]
Oct 22 20:39:24 <hostname redacted> systemd[1]: Requested transaction contradicts existing jobs: Resource deadlock avoided
Oct 22 20:39:24 <hostname redacted> systemd[1]: message repeated 6 times: [ Requested transaction contradicts existing jobs: Resource deadlock avoided]
Oct 22 20:39:24 <hostname redacted> systemd[1]: Requested transaction contradicts existing jobs: Resource deadlock avoided
Oct 22 20:39:24 <hostname redacted> systemd[1]: Requested transaction contradicts existing jobs: Resource deadlock avoided
Oct 22 20:39:24 <hostname redacted> upowerd[1662]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.4/2-3.4.3/2-3.4.3:1.3/0003:1E7D:3098.000F
Oct 22 20:39:24 <hostname redacted> upowerd[1662]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.4/2-3.4.3/2-3.4.3:1.3
Oct 22 20:39:24 <hostname redacted> upowerd[1662]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.4/2-3.4.3/2-3.4.3:1.1/0003:1E7D:3098.000E
Oct 22 20:39:24 <hostname redacted> upowerd[1662]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.4/2-3.4.3/2-3.4.3:1.1
Oct 22 20:39:24 <hostname redacted> systemd[1]: Requested transaction contradicts existing jobs: Resource deadlock avoided
Oct 22 20:39:24 <hostname redacted> acpid: input device has been disconnected, fd 16
Oct 22 20:39:24 <hostname redacted> systemd[1]: Stopped Driver for ROCCAT Vulcan 100/12x series keyboards.
```

Eruption version and computer information is the same as what I provided in #22.

X3n0m0rph59, 2020-10-23:
Hi! Thanks for your bug report!

It seem that the mouse events thread is spinning in a tight loop as soon as it loses connection to the mouse device.
I was able to reproduce this at first try. It will be fixed in the next release!

Btw.: If the keyboard and the mouse are connected to the same bus, then maybe you should stop the eruption daemon as well as mask the eruption.service while recording the pcapng file, so that the trace does not include as much packets destined to the keyboard.

erockce, 2020-10-23:
My computer has a limited number of USB ports. With the recent number of new wired devices, I use a Belkin USB hub.

When I start data collecting will remember to remove non-essential devices and plug the keyboard and mouse on separate USB ports.

X3n0m0rph59, 2020-11-12:
This should be fixed in Eruption `0.1.17`. Please reopen this issue if the problem persists!

closed 2020-11-12 by X3n0m0rph59