eruption-project/eruption #25: Feature Request: Add support for Kona AIMO

1drp, 2020-10-29:
Please add support for Kona AIMO.  I captured data in a *.pcap similar to what you asked for in the Kove thread (but on a bare-metal Windows install.)  Note that my Vulkan121 was on the same root.  The Kona was on Port20, and the Vulkan was on Port21.  Also similar to the Kove, the keyboard does respond to mouse events as is.  I wiggled the mouse around a bit before and after starting swarm, and I also cycled through all the buttons and wheel manipulations (but I can't remember the order...I can do it again if that's important).

[KonaAIMO.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5457368/KonaAIMO.zip)

X3n0m0rph59, 2020-10-30:
Thank you, the attached pcap file looks good! I have implemented a driver that should be able to initialize the device to AIMO mode and set the colors of the LEDs.

What you should see if all works out:
* The mouse should be initialized to AIMO mode
* All LEDs should briefly turn red
* All LEDs should turn blue
* The debug-tool terminates, LEDs should stay blue

Please run the following commands to try to initialize the device:
* cd to the directory where you downloaded `eruption-debug-tool`
* Please unzip the file and then:

```sh
$ chmod +x ./eruption-debug-tool
$ sudo ./eruption-debug-tool list
$ sudo ./eruption-debug-tool run-tests <device index of your mouse>
```

Be sure to run the correct version of `eruption-debug-tool` (`0.0.2`). The version supplied with `Eruption 0.1.16` will not work with your mouse. 

Please use the `list` sub-command of `eruption-debug-tool` and find the index of your mouse. (Use sub-device 0).

It would be nice if you could paste the console output of the commands above, so that I can check if all works out. If something goes wrong you should be able to restore factory settings using ROCCAT Swarm, e.g: If the DPI settings become garbled or something like that.

[eruption-debug-tool.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5462951/eruption-debug-tool.zip)

1drp, 2020-10-30:
$ sudo ./eruption-debug-tool list
`Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 

Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other subcommands of this tool

Index: 00: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 0
Index: 01: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 1
Index: 02: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 2
Index: 03: ID: 1e7d:3098 ROCCAT/ROCCAT Vulcan AIMO Subdev: 3
Index: 04: ID: 1e7d:2e27 ROCCAT/ROCCAT Kone Aimo Subdev: 0
Index: 05: ID: 1e7d:2e27 ROCCAT/ROCCAT Kone Aimo Subdev: 1

Enumeration completed`

$ sudo ./eruption-debug-tool run-tests 04
`Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 
Index: 04: ID: 1e7d:2e27 ROCCAT/ROCCAT Kone Aimo Subdev: 0
Bound driver: ROCCAT Kone Aimo
Sending device init sequence...
Sending control device feature report
  |040080|                              ...              00000000
                                                         00000003
Invalid result
Waiting for control device to respond...
  |040100ce|                            ....             00000000
                                                         00000004
Sending control device feature report
  |0e060101 00ff|                       ......           00000000
                                                         00000006
Waiting for control device to respond...
  |040100ce|                            ....             00000000
                                                         00000004
Sending control device feature report
  |0d2e0000 00000000 00000000 00000000| ................ 00000000
  |00000000 00000000 00000000 00000000| ................ 00000010
  |00000000 00000000 00000000 0000|     ..............   00000020
                                                         0000002e
Waiting for control device to respond...
  |040300ce|                            ....             00000000
                                                         00000004
  |040300ce|                            ....             00000000
                                                         00000004
  |040100ce|                            ....             00000000
                                                         00000004
Setting LEDs from supplied map...
  |0d2eff00 00ff0000 ff0000ff 0000ff00| ................ 00000000
  |00ff0000 ff0000ff 0000ff00 00ff0000| ................ 00000010
  |ff0000ff 0000ff00 00ff0000 0000|     ..............   00000020
                                                         0000002e
Setting LEDs from supplied map...
  |0d2e0000 ff0000ff 0000ff00 00ff0000| ................ 00000000
  |ff0000ff 0000ff00 00ff0000 ff0000ff| ................ 00000010
  |0000ff00 00ff0000 ff0000ff 0000|     ..............   00000020
                                                         0000002e`
The colors did not end up as you described; I'll do this again and capture a video after my next reboot (I switch back and forth between Linux and Windows to play a certain game with pesky anti-cheat software that blocks it on Linux).  Here's the final state of the LEDs.  (Lengths are approximate, and the color transitions are more gradual than what I depicted in the bitmap...I couldn't get a good picture.)

The DPI did not get scrambled...I cycled through all 5 settings.

[eruption-debug_KoneAIMO.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5463410/eruption-debug_KoneAIMO.zip)

X3n0m0rph59, 2020-10-30:
Great, so initialization basically works! It looks like I got some offsets wrong though. Hope that I can figure it out within a few tries.
Please test again with the new build of `eruption-debug-tool` version `0.0.3`

[eruption-debug-tool.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5463512/eruption-debug-tool.zip)

1drp, 2020-10-30:
Red then blue as described.
```
 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 
Index: 04: ID: 1e7d:2e27 ROCCAT/ROCCAT Kone Aimo Subdev: 0
Bound driver: ROCCAT Kone Aimo
Sending device init sequence...
Sending control device feature report
  |040080|                              ...              00000000
                                                         00000003
Invalid result
Waiting for control device to respond...
  |040100ce|                            ....             00000000
                                                         00000004
Sending control device feature report
  |0e060101 00ff|                       ......           00000000
                                                         00000006
Waiting for control device to respond...
  |040100ce|                            ....             00000000
                                                         00000004
Sending control device feature report
  |0d2e0000 00000000 00000000 00000000| ................ 00000000
  |00000000 00000000 00000000 00000000| ................ 00000010
  |00000000 00000000 00000000 0000|     ..............   00000020
                                                         0000002e
Waiting for control device to respond...
  |040300ce|                            ....             00000000
                                                         00000004
  |040300ce|                            ....             00000000
                                                         00000004
  |040100ce|                            ....             00000000
                                                         00000004
Setting LEDs from supplied map...
  |0d2eff00 00ffff00 00ffff00 00ffff00| ................ 00000000
  |00ffff00 00ffff00 00ffff00 00ffff00| ................ 00000010
  |00ffff00 00ffff00 00ffff00 00ff|     ..............   00000020
                                                         0000002e
Setting LEDs from supplied map...
  |0d2e0000 ffff0000 ffff0000 ffff0000| ................ 00000000
  |ffff0000 ffff0000 ffff0000 ffff0000| ................ 00000010
  |ffff0000 ffff0000 ffff0000 ffff|     ..............   00000020
                                                         0000002e
```

X3n0m0rph59, 2020-10-31:
Great, so support for the ROCCAT Kone Aimo will be included in the next release of Eruption!

Thank you for testing this!

closed 2020-10-31 by X3n0m0rph59