eruption-project/eruption #26: Issue: Missing Key Events, Vulcan 121

1drp, 2020-11-09:
xev does not register any events for Fn-F5 or Fn-F7.  I will map all the keys/combinations next time I boot into Windows and then compare to the same from Linux.  I will update this thread with a complete list of missing/miss-assigned keys that I find.

X3n0m0rph59, 2020-11-10:
> I will update this thread with a complete list of missing/miss-assigned keys that I find.

That would be great! Thanks in advance!

1drp, 2020-11-10:
I've collected the data from the Windows side, but I probably won't be getting to the Linux side until tomorrow night.  For Starters:

Fn-F1 Through Fn-F4 are kept internal to the keyboard (profile switching)
Fn-F5=182 (Computer)
Fn-F6=172 (Default Browser)
Fn-F7=180 (e-mail)
Fn-F8=183 (Calculator)
Fn-F9=177 (Previous Track)
Fn-F10=178 (Stop)
Fn-F11=179 (Play/Pause)
Fn-F12=176 (Next Track)
Fn-PRINTSCR and Fn-BREAK can be assigned in swarm, but default to the un-modified command.
Fn-SCROLL toggles game mode and is kept internal to the keyboard
No other keys have an alternate "Fn" in Swarm

Looking at Swarm, there are many more of these "open this thing" style commands that I will need to assign and capture to fully complete this list.  I'll come back and do that at a later date.

1drp, 2020-11-10:
I'm wondering if Linux doesn't use different keycodes for the "open this thing" and "media control" keys:  I did a quick run of F5-F12, and am getting different keycodes, but correct descriptions for the function.
Fn-F6=158 (XF86WWW)
Fn-F8=148 (XF86Calculator)
Fn-F9=173 (XF86AudioPrev)
Fn-F10=174 (XF86AudioStop)
Fn-F11=172 (XF86AudioPlay)
Fn-F12=171 (XF86AudioNext)

1drp, 2020-11-10:
I think what we are looking for is
F5: `#define XF86XK_Explorer		0x1008FF5D   /* Launch file explorer        */`
F7: `#define XF86XK_Mail		0x1008FF19   /* Invoke user's mail program */`
but I haven't been able to find keycodes for those keysyms

1drp, 2020-11-10:
I should note that I use fluxbox, so I'm not the best for testing if one of the big DEs will automatically interpret the commands correctly.  I have to manually map the keys to commands (which is how I discovered the bug in the first place).

I would say that once we have Fn-F5 and Fn-F7 sending codes correctly, that we will have sufficiently addressed this bug.  Making sure that all options for binding in the Swarm software are available in Eruption would be more appropriate in a separate feature request, and can be deferred to a future release.

X3n0m0rph59, 2020-11-11:
Yes, you are right! With commit 2559364 I did enable some evdev keycodes on the Eruption virtual keyboard device, which were previously missing. I also changed `KEY_WWW` to `KEY_HOMEPAGE`, since this is what the keyboard emits without Eruption being running.

|Key| evdev event | X KeySym |
| --------- | -------------------- | ---------------------- |
| F5 | KEY_FILE | XF86XK_Explorer |
| F6 | KEY_HOMEPAGE | XF86XK_HomePage |
| F7 | KEY_MAIL | XF86XK_Mail |
| F8 | KEY_CALC | XF86XK_Calculator |

These changes will be included in the next release of Eruption.

1drp, 2020-11-18:
Using the beta code from issue #29, I was able to confirm that F5 through F8 are coming through as keycodes 152 (Explorer), 180 (HomePage), 163 (Mail), and 148 (Calculator).

X3n0m0rph59, 2020-11-18:
Thank you, great! I will close this issue now.

closed 2020-11-18 by X3n0m0rph59