eruption-project/eruption #27: Add support for loading configs from user-specific directories

cheesits456, 2020-11-10:
I absolutely love this driver, it works amazingly well and opens up so many opportunities for doing cool stuff with the keyboard - you've done some amazing work :)

A suggestion I have is that the `eruptionctl switch profile` command should also look for profiles stored in a directory such as `~/.config/eruption/profiles`, and the `active_scripts` array should also look for scripts stored in `~/.config/eruption/scripts`. That'll allow each user to make their own configs and stuff unique to that individual user, and it won't require root privileges to modify

X3n0m0rph59, 2020-11-12:
Hi! Thanks for your feature request!

I thought about implementing this too, since user supplied scripts/per-user profiles would be really nice to have! I refrained from implementing it because of the security implications this has. It would require a huge change in the security model of Eruption. The Eruption daemon needs to run with elevated privileges to access the hardware. If it would be possible for a regular user to arbitrarily load and run Lua code, this could be easily exploited in multiple ways to gain root privileges. It would increase the attack surface of Eruption by a huge margin.

One possible solution to this problem is implemented in the `eruption-gui` branch that I am working on. Basically the Eruption GUI uses PolicyKit to authenticate the user as an administrator. When the user changes a configuration value via the GUI, the Eruption GUI process (running as a regular user) calls into Eruption (running as root) via a D-Bus based API and requests a change of that configuration value (or even of Lua code). The Eruption daemon only executes this request if PolicyKit confirms that the user running the GUI process is authenticated with the required permissions. This basically resembles the way other shared hardware resources are managed on Linux, like e.g. printer devices.

Implementing it this way has the following disadvantages:

* It will not deliver per-user separation of profiles and scripts, we only have a single namespace
* It reduces the attack surface of the Eruption daemon but requires authentication of the user

In the current security model, we basically have three levels of permissions: Monitor, Settings and Manage.

Excerpt of the PolicyKit configuration:
```xml
    <!-- Monitor the Eruption daemons state -->
    <action id="org.eruption.monitor">
      <description>Monitor Eruption</description>
      <message>Authentication is required to monitor Eruption</message>
      <defaults>
        <allow_any>yes</allow_any>
        <allow_inactive>yes</allow_inactive>
        <allow_active>yes</allow_active>
      </defaults>
    </action>

    <!-- Change non-privileged settings like e.g.: LED Brightness -->
    <action id="org.eruption.settings">
      <description>Change settings of Eruption</description>
      <message>Authentication is required to change the settings of Eruption</message>
      <defaults>
        <allow_any>yes</allow_any>
        <allow_inactive>yes</allow_inactive>
        <allow_active>yes</allow_active>
      </defaults>
    </action>

    <!-- Change privileged settings like e.g.: Lua script code -->
    <action id="org.eruption.manage">
      <description>Manage the Eruption daemon</description>
      <message>Authentication is required to manage the Eruption daemon</message>
      <defaults>
        <allow_any>auth_admin_keep</allow_any>
        <allow_inactive>auth_admin_keep</allow_inactive>
        <allow_active>auth_admin_keep</allow_active>
      </defaults>
    </action>
```

I will have to think about this some more...

cheesits456, 2020-11-12:
Perhaps you could add a toggleable option to read profiles and scripts from the `.config` directories, and provide a warning about the security implications? Because for people who only use their computers for themselves and only have a single account on the computer, that isn't so much of a problem. Just an idea ¯\\\_(ツ)_/¯

SHU-red, 2021-11-20:
Hey there,
i got my Roccat Vulcan today and i love this project!
Im getting excited to script my own scripts!

In
`/etc/eruption/eruption.conf`
i tried to add my user-.conf folder to the search paths by adding them comma separated like this
```
[global]
profile_dirs = ["/var/lib/eruption/profiles/", "/home/USER/.config/eruption/profiles/"]
script_dirs = ["/usr/share/eruption/scripts/", "/home/USER/.config/eruption/scripts/"]
```

Seems that this does not work and is related to this task
1. Am i doing something wrong?
2. (As others already seem to have requested) it would be nice to have my custom scripts in my .config folder for easily backing up

Im not a very advanced user, so i thought my questions are here at the right place?

X3n0m0rph59, 2021-11-20:
Hi @SHU-red, thanks for your feature request!

Eruption `0.1.23` built from the current `master` branch is able to load scripts and profiles from the user's home directory. I just double checked this.

Support is in place for the following components:
 * The `eruption` daemon proper
 * The companion tools like `eruptionctl` should mostly be able to deal with multiple script and profile paths
 * Eruption profile switcher GNOME 4x Shell Extension has full support for multiple script and profile paths

Missing support or erratic behavior:
 * Eruption GUI does not handle non-standard paths, it will just crash
 * `eruptionctl scripts list` does not show some of the user's scripts
 * Include order of scripts across directories is not tested well, will have to re-visit this soon

Steps to add custom directories:
```shell
mkdir -p /home/USER/.local/share/eruption/{scripts,profiles}
sudoedit /etc/eruption/eruption.conf
```
2. Now please add the directories as applicable to the `[global]` section in `eruption.conf`

3. You need to create a `my-profile.profile` file in the respective profile directory

4. You need to create `script.lua` and a valid `script.lua.manifest` file in the respective script directory. The `.manifest` file is mandatory, without it it will not work

If it still does not work then please post debug logs of `eruption`:
```shell
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
sudo RUST_LOG=debug eruption

# re-enable eruption.service
sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
```

Please open up new issues as soon as you encounter further problems! Btw., please specify the version of Eruption that you are currently using.

SHU-red, 2021-11-20:
Thanks for your friendly support!
I installed eruption-git from AUR-package manager which seems to be (Eruption 0.1.23 (0.1.22.r28.g1ba264c) (release build))

- I think this means that my version is too old and i have do make a build by myself?
- Do i have to place my code into `/home/USER/.local/share/eruption` or can i choose whatever i want? Like `/home/USER/.config/eruption`?

X3n0m0rph59, 2021-11-21:
@SHU-red thanks, you're welcome!

I just built Eruption from git commit 1ba264c and can confirm that this built is able to handle custom scripts and profiles. So there must be something else that is going on.

The location of the directories is not important. You can place the files wherever you like, as long as your user account has read/write access to the files it should be okay.

Could you please post your `.profile` file and/or `.lua` script and `.lua.manifest` file?
Please post debug logs as well!

SHU-red, 2021-11-21:
Now i did acknowledge, that you also created a eruption-gui! Awesome!

__OK as said im not very advanced in all of this stuff but i tried my best:__

I understood, that your commit 1ba264c from 24. Aug should already be contained in the eruption version i use from AUR eruption-git = 1.2.23
Hopefully i did understand you right

_______
Based on that i now did try to crate
- user.profile: Containing 3 provided scripts and 1 i made myself
- destinytwo.lua: My personal script for gaming

Here is what i did:

0. Updated Vulcan firmware with swarm on WIN-PC (could not achieve this in linux) & Re-installed eruption-git from AUR (everything works)
1. /etc/eruption/eruption.conf: Add paths to my custom folders
2. user.profile: copied a random provided profile to ~/.config/eruption/profiles/user.profile and deleted everything after "active_scripts"
3. destinytwo.lua and destinytwo.lua.manifest: copied from gaming* and placed into ~/.config/eruption/scripts/

__Attention:__
For the first step, i only used scripts already provided by eruption in my user.profile, so i did NOT use destinytwo.lua(.manifest)
--> I only used scripts provided by eruption

```
active_scripts = [
	'solid.lua',
	'ghost.lua',
	'wave.lua',
	'default.lua',
]
```
4. Reboot my PC (because i dont know if there is something wich has to scan folders at first or somethin
--> Roccat Vulcan basically working with default.profile active

5. $ eruptionctl switch profile /home/sebastian/.config/eruption/profiles/user.profile --> Nothing changed
6. $ eruptionctl status profile --> Still default.profile active
7. Check debug log during trying to change profile:
```

ERROR eruption::plugins::audio               > Pulse Audio error: Could not open Pulse Audio: Connection refused
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 89
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
 ERROR eruption                               > Pulse Audio error: Could not open Pulse Audio: Connection refused
 DEBUG eruption::hwdevices::roccat_vulcan_1xx >   |0300fb6b 01000000|                   ...k....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_1xx >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 2 args: 107
 INFO  eruption                               > Loading profile: /home/sebastian/.config/eruption/profiles/user.profile
 INFO  eruption                               > Switching to profile: /home/sebastian/.config/eruption/profiles/user.profile
 ERROR eruption                               > Script file or manifest inaccessible: default.lua
 ERROR eruption                               > Invalid profile: /home/sebastian/.config/eruption/profiles/user.profile, refusing to switch profiles
 DEBUG eruption::hwdevices::roccat_vulcan_1xx >   |0300fb6b 00000000|                   ...k....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_1xx >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 1 args: 107
 DEBUG eruption::scripting::script::callbacks > Macros: Key up: Index: 89
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
 ERROR eruption                               > Pulse Audio error: Could not open Pulse Audio: Connection refused

```
Despite the Pulse Audio error it says, that my Script file or manifest default.lua is inaccessible and the user.profile is invalid

The chance is high, that i did do something wrong.
Tell me if this is the case ...

Thank you again!

## eruption.conf
```

# Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices
# Main configuration file

[global]
profile_dirs = ["/var/lib/eruption/profiles/", "/home/USER/.config/eruption/profiles/"]
script_dirs = ["/usr/share/eruption/scripts/", "/home/USER/.config/eruption/scripts/"]

# select your keyboard variant
# keyboard_variant = "ANSI"
keyboard_variant = "ISO"

# Mouse handling
enable_mouse = true
grab_mouse = true

# "Away from keyboard" handling
afk_profile = "/var/lib/eruption/profiles/rainbow-wave.profile"
afk_timeout_secs = 0

# [[devices]]
# entry_type = "device"
# device_class = "serial"
# device_name = "Adalight Custom Serial LEDs"
# device_file = "/dev/ttyACM0"

# [[devices]]
# entry_type = "blacklist"
# vendor_id = 0x1e7d
# product_id = 0x311a

```
## user.profile
```

id = '5cd23fa6-e965-45cb-a0cd-e87d28713092'
name = 'USER'
description = 'User defined profile'
active_scripts = [
	'solid.lua',
	'ghost.lua',
	'wave.lua',
	'default.lua',
]
```
## destinytwo.lua

no changes to gaming

## destinytwo.lua.manifest
only changed name and author

## System
```

inxi -Fxxxrz

System:    Kernel: 5.14.18-1-MANJARO x86_64 bits: 64 compiler: gcc v: 11.1.0 Desktop: awesome 4.3-1307-g0246c44d4 vt: 1
           dm: SDDM Distro: Manjaro Linux base: Arch Linux

```

X3n0m0rph59, 2021-11-21:
@SHU-red

Yes you did understand that correctly. The AUR package `eruption-git` does ship with support for multiple script and profile directories.

It seems that the Lua script `default.lua` does not exist in your custom directory!? Eruption does have a `eruption/src/scripts/lib/themes/default.lua` but it this is a color theme, not an effect script.

You are missing `macros.lua` in your `.profile`; it is possible to run a profile without `macros.lua` but you will be lacking a lot of features:
 * `FN` + `F1` - `F4` won't switch slots anymore
 * Master volume overlay and audio muted state will cease to work
 * Macro keys will be disabled
 * GameMode/EasyShift+ macros will be disabled

So please try to replace `default.lua` with `macros.lua`!

SHU-red, 2021-11-21:
OK thanks a lot!
I think there is something i completely got wrong!

I thougt each profile has slots fro F1 to F4 which are defined in the .profile file.

But i think one profile only defines the behaviour of for example ohnly the Slot F1

Thanks ...

I will try to discover this

SHU-red, 2021-11-22:
I got it working!

One last thing, regarding the topic of this issue:
- I would like it if you could add directories in the user-home-folder already in the delivered eruption-version
- e.g. /home/$USER/.config/eruption/{scripts profiles} or /home/$USER/.local/eruption/{scripts profiles}
- This way the delivered eruption.conf could be left untouched (dont know if this is overwritten each time eruption is updated)
- Most applications work like this, thats why everyone keeps backing up their "DotFiles", which contain their personal settings
- I know that there is a very hich chance that i don't tell you something new here! 😅

Thank you very much for helping me and i will try to contribute if necessary and possible for me!
Looking forward do future releases 😉

X3n0m0rph59, 2021-11-24:
Hi @SHU-red, great that it is working now!

The `/etc/eruption/eruption.conf` file is handled as a conffile by most package managers, and should not be overwritten when it has been modified.

Mainly because of the security implications I will refrain from shipping such a change in the default installation, but I will continue to support this feature for the users who like to have that kind of setup.

closed 2021-11-24 by X3n0m0rph59

SHU-red, 2021-11-24:
If it doesn't get overwritten it's totally perfect for me!

Thanks!