eruption-project/eruption #28: Building from source fails: no associated item named `EPSILON` found for type `f64` in the current scope

DreamTexX, 2020-11-12:
While building with `cargo build --all -release` the following error appears:

```
error[E0599]: no associated item named `EPSILON` found for type `f64` in the current scope
   --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/mlua-0.4.2/src/ffi/compat53.rs:292:54
    |
292 |         if (n - i as lua_Number).abs() < lua_Number::EPSILON {
    |                                                      ^^^^^^^ associated item not found in `f64`

error[E0599]: no associated item named `EPSILON` found for type `f64` in the current scope
   --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/mlua-0.4.2/src/ffi/compat53.rs:318:65
    |
318 |     if ok != 0 && (n - n_int as lua_Number).abs() < lua_Number::EPSILON {
    |                                                                 ^^^^^^^ associated item not found in `f64`

error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0599`.
error: could not compile `mlua`.
```

How can i solve this?

X3n0m0rph59, 2020-11-12:
Hi, thanks for opening this issue!

This looks like an outdated version of Rust. Please try to upgrade at least to the latest stable release. If this is not possible for you, then you may still compile Eruption by applying a few tweaks to the build system.

Could you please provide the following information:

* What distro and which version of it are you using?
* Which version of rust are you using?
```
$ rustc --version
```

X3n0m0rph59, 2020-11-12:
In case you are using Linux from Scratch, there exists a compatibility branch of Eruption in the branch `lfs-compat`, that should compile under LFS >= 9.1 using the old stable version of Rust.

```
$ cargo +stable-2020-03-12-x86_64-unknown-linux-gnu build
```

This successfully compiles the `lfs-compat` branch on my system.

DreamTexX, 2020-11-12:
I'm using Debian 10
And `rustc --version` outputs:
```
rustc 1.41.1
```

DreamTexX, 2020-11-12:
> In case you are using Linux from Scratch, there exists a compatibility branch of Eruption in the branch `lfs-compat`, that should compile under LFS >= 9.1 using the old stable version of Rust.
>
> ``` [...]

I switched to that branch, but running `cargo +stable-2020-03-12-x86_64-unknown-linux-gnu build` just fails with
```
error: no such subcommand: `+stable-2020-03-12-x86_64-unknown-linux-gnu`
```

DreamTexX, 2020-11-12:
And running `cargo build` in the `lfs-compat` branch gives another error...

X3n0m0rph59, 2020-11-12:
The version of Rust in Debian 10 is really old. Is it possible for you to install stable Rust via `rustup`? This is the recommended way to build Eruption.

https://www.rust-lang.org/tools/install

```sh
$ cargo +stable --version
cargo 1.47.0 (f3c7e066a 2020-08-28)
```

X3n0m0rph59, 2020-11-12:
You do not have to build from source, btw. You can manually download the package from the Ubuntu bionic PPA.

https://launchpad.net/~x3n0m0rph59/+archive/ubuntu/eruption-roccat-vulcan/+files/eruption-roccat-vulcan-git_0.1.17-ubuntu4_amd64.deb

DreamTexX, 2020-11-12:
Thanks, its working now. Downloaded the deb

closed 2020-11-12 by DreamTexX