eruption-project/eruption #29: Kone AIMO displays default colors

1drp, 2020-11-12:
My Kone is still not working with the profiles.  I rebooted and pulled the journalctl log for eruption.
```-- Logs begin at Sun 2020-11-01 12:04:05 PST, end at Thu 2020-11-12 12:42:49 PST. --
Nov 12 12:40:06 Ruhig systemd[1]: Starting Driver for ROCCAT Vulcan 100/12x series keyboards...
Nov 12 12:40:06 Ruhig systemd[1]: Started Driver for ROCCAT Vulcan 100/12x series keyboards.
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption > Starting user-mode driver for ROCCAT Vulcan 100/12x series keyboards: Version 0.1.17
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption > Loading saved state...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption > Registering plugins...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Process keyboard events
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Process mouse events
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                       > Plugins loaded and initialized successfully
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                       > Enumerating connected devices...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::hwdevices            > Found Control interface: "0001:0004:01": ROCCAT Vulcan AIMO
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::hwdevices            > Found LED interface: "0001:0004:03": ROCCAT Vulcan AIMO
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::hwdevices::roccat_vulcan > Bound driver: ROCCAT Vulcan
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::hwdevices::generic_mouse > Bound driver: Generic Mouse Device
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Device enumeration completed
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Spawning keyboard input thread...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Spawning mouse input thread...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Initializing devices...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Waiting for tasks to complete...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Opening keyboard device...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::keyboard        > Now listening on: /dev/input/event21
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::keyboard        > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::keyboard        > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::keyboard        > Physical location: usb-0000:03:00.0-13/input0
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::keyboard        > Grabbing the keyboard device exclusively
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::mouse           > Now listening on: /dev/input/event16
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::mouse           > Input device name: "ROCCAT ROCCAT Kone Aimo Mouse"
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::mouse           > Input device ID: bus 0x3 vendor 0x1e7d product 0x2e27
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::mouse           > Physical location: usb-0000:03:00.0-12/input0
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption::plugins::mouse           > Grabbing the mouse device exclusively
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Initializing keyboard device...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Configuring keyboard LEDs...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Opening mouse device...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Initializing mouse device...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Configuring mouse LEDs...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Keyboard firmware revision: 136
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Mouse firmware revision: <not supported>
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Performing late initializations...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Initializing D-Bus API...
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Late initializations completed
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Startup completed
Nov 12 12:40:07 Ruhig eruption[1406]:  INFO  eruption                           > Switching to profile: /var/lib/eruption/profiles/swirl-turbulence.profile```
I have tried all included profiles, but none will change the lights on my mouse.  (I must say that how moving the mouse moves the swirls is rather impressive).
Running Arch with 5.9.6-arch1-1

1drp, 2020-11-12:
The things I see that may be problems:
It appears to be binding the Generic Mouse Driver
Mouse firmware revision: <not supported>

1drp, 2020-11-12:
out of curiosity, I ran the debug tool's (0.0.3) list function, and got nothing useful:
```Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other subcommands of this tool

Index: 00: ID: 1e7d:3098 <unknown>/<unknown> Subdev: 0
Index: 01: ID: 1e7d:3098 <unknown>/<unknown> Subdev: 1
Index: 02: ID: 1e7d:3098 <unknown>/<unknown> Subdev: 2
Index: 03: ID: 1e7d:3098 <unknown>/<unknown> Subdev: 3
Index: 04: ID: 1e7d:2e27 <unknown>/<unknown> Subdev: 0
Index: 05: ID: 1e7d:2e27 <unknown>/<unknown> Subdev: 1
```
I know that 00-03 are the keyboard and 04-05 are the mouse from our earlier work to get the mouse LEDs calibrated, but it's not very useful as is.

X3n0m0rph59, 2020-11-13:
I believe I found the bug in Eruption. The device probing code really needs to be rewritten... I will make a new release soon.

The `eruption-debug-tool` needs to be run as the `root` user. Otherwise it wont be able to access the USB devices.

1drp, 2020-11-13:
ah yes...works normally as root.  Had the red then blue on the mouse as expected with run-tests.
```
Index: 04: ID: 1e7d:2e27 ROCCAT/ROCCAT Kone Aimo Subdev: 0
Bound driver: ROCCAT Kone Aimo
Sending device init sequence...
Sending control device feature report
  |040080|                              ...              00000000
                                                         00000003
Invalid result
Waiting for control device to respond...
  |040100ce|                            ....             00000000
                                                         00000004
Sending control device feature report
  |0e060101 00ff|                       ......           00000000
                                                         00000006
Waiting for control device to respond...
  |040100ce|                            ....             00000000
                                                         00000004
Sending control device feature report
  |0d2e0000 00000000 00000000 00000000| ................ 00000000
  |00000000 00000000 00000000 00000000| ................ 00000010
  |00000000 00000000 00000000 0000|     ..............   00000020
                                                         0000002e
Waiting for control device to respond...
  |040300ce|                            ....             00000000
                                                         00000004
  |040300ce|                            ....             00000000
                                                         00000004
  |040100ce|                            ....             00000000
                                                         00000004
Setting LEDs from supplied map...
  |0d2eff00 00ffff00 00ffff00 00ffff00| ................ 00000000
  |00ffff00 00ffff00 00ffff00 00ffff00| ................ 00000010
  |00ffff00 00ffff00 00ffff00 00ff|     ..............   00000020
                                                         0000002e
Setting LEDs from supplied map...
  |0d2e0000 ffff0000 ffff0000 ffff0000| ................ 00000000
  |ffff0000 ffff0000 ffff0000 ffff0000| ................ 00000010
  |ffff0000 ffff0000 ffff0000 ffff|     ..............   00000020
                                                         0000002e
```

X3n0m0rph59, 2020-11-18:
@1drp I prepared an Arch Linux pre-release package of Eruption. It _should_ now find your ROCCAT Kone Aimo mouse. I did a complete rewrite of the device probing code. Please report back any issues you encounter. Thank you in Advance!

Please unzip the file prior to installing it:
[eruption-roccat-vulcan-git-0.1.18-0-x86_64.pkg.tar.zst.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5560525/eruption-roccat-vulcan-git-0.1.18-0-x86_64.pkg.tar.zst.zip)

See [CHANGES.md](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/blob/master/CHANGES.md) for a list of changes.

1drp, 2020-11-18:
Colors work, but mouse doesn't.  No movement, and no clicks.

X3n0m0rph59, 2020-11-18:
Thanks for taking your time to test this! 
I will build another package which binds the ROCCAT Kone AIMO driver to HID subdevice 1 instead of subdevice 0.

X3n0m0rph59, 2020-11-18:
@1drp Please test again with this build:
[eruption-roccat-vulcan-git-0.1.18-1-x86_64.pkg.tar.zst.zip](https://github.com/X3n0m0rph59/eruption-roccat-vulcan/files/5562564/eruption-roccat-vulcan-git-0.1.18-1-x86_64.pkg.tar.zst.zip)

I hope that this one will work!

1drp, 2020-11-18:
That seems to work.  Movement, Left, Right, Middle, scroll (up/down), DPI (up/down), forward, and back all check out.  colors are still functional.  Brightness control on the keyboard also changes mouse.

X3n0m0rph59, 2020-11-19:
@1drp Excellent, thanks for your patience!

>Brightness control on the keyboard also changes mouse.

That's by design, they share a unified canvas. The master brightness slider in the yet-to-be published GUI as well as in the `Eruption Profile Switcher` GNOME Shell extension all change the brightness of that canvas.

Next step would be to finish support for multiple different colors. This amounts to assigning positions in the color map array to LEDs on the mouse. You may have noticed that all LEDs on your Kone just use a single color at the moment.

I will close this issue now, please open up new issues anytime!

closed 2020-11-19 by X3n0m0rph59