eruption-project/eruption #30: audioviz3.lua bug ?

Glog78, 2020-11-16:
aur/eruption-roccat-vulcan-git 0.1.17-4 
eruptionctl switch profile profile3.profile

switches to audioviz3.lua but this visualizer only shows one bar for analyzing the sound on the most left keys (ESC downwards) ... the green line on the botton is present too. I would love to have a fancy spectrum bars all over my keyboard <3

X3n0m0rph59, 2020-11-16:
Hi! Thank you for your bug report!

Did you follow the steps listed in the `README.md`, to enable Eruption to connect to PulseAudio?

For the audio related features of Eruption to work, it is important that you perform the steps listed here:
https://github.com/X3n0m0rph59/eruption-roccat-vulcan#after_setup

If you already completed the aforementioned configuration steps, it may be the case that the Eruption audio grabber is listening on the wrong PulseAudio source. You may want to try `pavucontrol`... There, switch to the 2nd tab (called "Recording"), and then in the upper right corner of the window, please select something like `Monitor of Internal Audio Stereo` as the source.

I suspect that Eruption is currently listening on the Video Card HDMI audio or something like that.

Glog78, 2020-11-16:
I had this setup done and everyone should do it cause else there will be a lot of messages in the logs about not being able to connect to pulseaudio. (maybe you might want to check this one)

I don't know what was the issue ... now it's working ... and yes i was checking pulse input routing 
Thank you for takeing your time i will close this one for now

Glog78, 2020-11-16:
Seems like some strange bug on my side. Close this for the moment.

closed 2020-11-16 by Glog78