eruption-project/eruption #31: Roccat Kone Aimo adressing individual LEDs

7ritn, 2020-12-05:
Hi,

first of all. I wanna thank you for this project. Since I got my mouse I looked for a driver enabling RGB for it.

When I installed it, I noticed that all the effects change all the LEDs on the mouse at the same time. I was wondering, if the ability to address the individual LEDs has been implemented already and if so, how do I address them? If not, I'd would be great to implement this feature. I'd like to help, but I'm not super experienced in coding and never used RUST before.

Thanks again!

X3n0m0rph59, 2020-12-05:
Hi, thank you for your request!

As you noticed, the driver sub-module for the ROCCAT Kone Aimo in Eruption `0.1.18` just shows a single color value on all the built-in LEDs. The color value is simply taken from that of the `Enter` key on the keyboard's numpad.

The upcoming release of Eruption, version `0.1.19` will ship with initial support for addressing each LED on the mouse individually. The new version implements a shared color map for all connected devices. The color map can be imagined as a huge canvas, on which the Lua scripts paint their respective effects. Each device is assigned a distinct zone on that canvas. The keyboard zone lies in the range `0 .. 144`,  followed by the zone of the mouse from `145 .. 180`. I already adapted all Lua scripts to use the new addressing scheme.

The driver module for the ROCCAT Kone Aimo declares 11 distinct LEDs:

`eruption/src/hwdevices/roccat_kone_aimo.rs:36`:
```Rust
// canvas to LED index mapping
pub const LED_0: usize = constants::CANVAS_SIZE - 36;
pub const LED_1: usize = constants::CANVAS_SIZE - 35;
pub const LED_2: usize = constants::CANVAS_SIZE - 34;
pub const LED_3: usize = constants::CANVAS_SIZE - 33;
pub const LED_4: usize = constants::CANVAS_SIZE - 32;
pub const LED_5: usize = constants::CANVAS_SIZE - 31;
pub const LED_6: usize = constants::CANVAS_SIZE - 30;
pub const LED_7: usize = constants::CANVAS_SIZE - 29;
pub const LED_8: usize = constants::CANVAS_SIZE - 28;
pub const LED_9: usize = constants::CANVAS_SIZE - 27;
pub const LED_10: usize = constants::CANVAS_SIZE - 26;
```

The code that sets the LEDs on the device:

`eruption/src/hwdevices/roccat_kone_aimo.rs:643`:
```Rust
fn send_led_map(&mut self, led_map: &[RGBA]) -> Result<()> {
        trace!("Setting LEDs from supplied map...");

        if !self.is_bound {
            Err(HwDeviceError::DeviceNotBound {}.into())
        } else if !self.is_opened {
            Err(HwDeviceError::DeviceNotOpened {}.into())
        } else if !self.is_initialized {
            Err(HwDeviceError::DeviceNotInitialized {}.into())
        } else {
            let ctrl_dev = self.ctrl_hiddev.as_ref().lock();
            let ctrl_dev = ctrl_dev.as_ref().unwrap();

            let buf: [u8; 46] = [
                0x0d,
                0x2e,
                led_map[LED_0].r,
                led_map[LED_0].g,
                led_map[LED_0].b,
                led_map[LED_0].a,
                led_map[LED_1].r,
                led_map[LED_1].g,
                led_map[LED_1].b,
                led_map[LED_1].a,
                led_map[LED_2].r,
                led_map[LED_2].g,
                led_map[LED_2].b,
                led_map[LED_2].a,
                led_map[LED_3].r,
                led_map[LED_3].g,
                led_map[LED_3].b,
                led_map[LED_3].a,
                led_map[LED_4].r,
                led_map[LED_4].g,
                led_map[LED_4].b,
                led_map[LED_4].a,
                led_map[LED_5].r,
                led_map[LED_5].g,
                led_map[LED_5].b,
                led_map[LED_5].a,
                led_map[LED_6].r,
                led_map[LED_6].g,
[... 83 lines cut ...]

    -- calculate batique effect
    if ticks % animation_delay == 0 then
        for i = mouse_zone_start, mouse_zone_end do
            local x = i / (mouse_zone_end - mouse_zone_start)
            local y = i / (mouse_zone_end - mouse_zone_start)

            local val = super_simplex_noise((x / coord_scale),
                                            (y / coord_scale),
                                            ticks / time_scale)
            val = lerp(0, 360, val)

            color_map[i] = hsla_to_color((val / color_divisor) + color_offset,
                                            color_saturation, color_lightness,
                                            lerp(0, 255, opacity))
        end

        submit_color_map(color_map)
    end
end
```

Will push the `batique-mouse.lua` support to the `master` branch soon. Please stay tuned... It would be great if you could adapt it and make it look nice!

I am willing to give further support. Please feel free to open up issues any time!

7ritn, 2020-12-05:
Thanks for all that info. When I add the batique-mouse.lua script to my scripts and add it to a profile, the deamon just crashes. Is it supposed to work or do you still need to implement the shared canvas for it to work?

X3n0m0rph59, 2020-12-05:
It is only supposed to work with Eruption version `0.1.19` and greater. I will make a new release in a few days from now!
Eruption version `0.1.18` is unable to address the LEDs on the Kone Aimo individually.

X3n0m0rph59, 2020-12-05:
Could you please tell me which Linux distribution you are using? Maybe I can build a pre-release package, so that we can verify that the code is working correctly. Otherwise, I would have to implement the new code behind a feature gate, just to be safe.

7ritn, 2020-12-05:
Sure, I'm using Manjaro

X3n0m0rph59, 2020-12-06:
Hi, okay I have updated the PKGBUILD now, inside the `master` branch.

To build a package from the current state of the `master` branch please perform these steps:
* Save [PKGBUILD](https://raw.githubusercontent.com/X3n0m0rph59/eruption/master/support/pkg/arch/PKGBUILD) to a folder of your choice
* Open up a terminal, `cd` to that folder and then run the command `makepkg -crsi`. That should build and install the latest version of Eruption
* Run the command `sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service` to restart the daemon
* Run `sudoedit /var/lib/eruption/profiles/profile1.profile ` and remove the '#' character in front of `#    batique-mouse.lua` at line 13
* Switch forth and back to a slot that has `profile1.profile` assigned

Now you should see the new effect rendered on the mouse!

7ritn, 2020-12-06:
I followed your steps, but couldn't install it with makepkg, because eruption.install was missing. I therefor installed the newest version via AUR and just replaced the PCKBUILD with the new one. It installed without a prolblem. However, when I start the service it crashes immediatly. However, it is able to change the colour of the mouse and the effect stays after the crash. This happens on whatever profile I choose. After the crash I'm sorta able to change the profile. eruptionctl gives me an error, but when I restart the service the new profile is applied. When I looked into journalctl I found the following error message: 

```
Dec 06 10:49:10 triton-moon eruption[15217]:  ERROR eruption::scripting::script           > Lua error: runtime error: [string "?"]:95: attempt to perform arithmetic on a nil value
Dec 06 10:49:10 triton-moon eruption[15217]: stack traceback:
Dec 06 10:49:10 triton-moon eruption[15217]:         ok: in function '__add'
Dec 06 10:49:10 triton-moon eruption[15217]:         ok:95: in function <[string "?"]:89>
Dec 06 10:49:10 triton-moon eruption[15217]:         UnknownError
Dec 06 10:49:10 triton-moon eruption[15217]:  ERROR eruption::scripting::script           > Lua error: runtime error: [string "?"]:41: attempt to perform arithmetic on a nil value
Dec 06 10:49:10 triton-moon eruption[15217]: stack traceback:
Dec 06 10:49:10 triton-moon eruption[15217]:         ok: in function '__add'
Dec 06 10:49:10 triton-moon eruption[15217]:         ok:41: in function <[string "?"]:37>
Dec 06 10:49:10 triton-moon eruption[15217]:         UnknownError
```

I attached the whole error log.

X3n0m0rph59, 2020-12-06:
Ah, sorry! I forgot to provide you the `eruption.install` file.

[PKGBUILD](https://raw.githubusercontent.com/X3n0m0rph59/eruption/master/support/pkg/arch/PKGBUILD)
[eruption.install](https://raw.githubusercontent.com/X3n0m0rph59/eruption/master/support/pkg/arch/eruption.install)

It would be great if you could try again to build using `makepkg`. Let's first be sure that you are using the latest Lua scripts.

You are using a setup with an unsupported keyboard, but a supported mouse. I would like to support this kind of setup too, but I have to look into this more thoroughly. This kind of setup is currently not tested very well. If the error persists even after successful installation of Eruption `0.1.19`, could you please try to use a profile that only instantiates `batique-mouse.lua`, and no other Lua scripts at all?

I am thinking of something like this:

`/var/lib/eruption/profiles/my-batique-mouse.profile`
```toml
id = '5dc59fa6-e965-45cb-a0da-e87d28713094'
name = 'Batique (Mouse only)'
description = 'The batique profile (Mouse only)'
active_scripts = [
	'batique-mouse.lua',
]
```

7ritn, 2020-12-06:
No problem! So I reinstalled eruption. Eruption crashes in profile1 even if batique-mouse.lua is disabled. Enabling it doesn't change anything. I'm no longer able to change profiles at all, so I replaced profile1 with batique-mouse-only profile. After that eruption starts normal without any error (except ERROR eruption::hwdevices                   > Failed to query HID device class: No compatible devices found, this error is also present in the log I posted). However, all the LEDs display the same colour.

I tried switching to batique.profile to see if it crashes. It didn't produce any errors. Even more surprising batique.profile colours the LEDs in different colours! I attached a video showing the effect: https://www.youtube.com/watch?v=xdPZhaPFxHo

X3n0m0rph59, 2020-12-06:
@7ritn Great, thank you for your patience!

It _could be_ that `batique-mouse.lua` is working too, but would just require tuning of some parameters to really show differing colors on the mouse.

I just tried to reproduce the setup that you are using (an unsupported keyboard and a supported mouse), and I instantly hit multiple bugs. Will fix them now and get back to you soon!

X3n0m0rph59, 2020-12-06:
@7ritn I just pushed out the first round of bug fixes to the Lua scripts and the generic keyboard driver! It should be much less crash prone now. To rebuild the Eruption package, please download the latest `PKGBUILD`. I updated the commit ID and package release version. (The links are still valid)

7ritn, 2020-12-06:
Oh wow! All errors are gone. Profile1.profile is just solid and with batique-mouse.lua enabled sorta like batique. Profile2 and Profile3 are still just disabled LEDs.

Btw. how is Batique supposed to look like, do you maybe have a reference image of the pattern or video? I don't have a Roccat keyboard myself to see how it looks on the keyboard.

X3n0m0rph59, 2020-12-08:
Nice! Yes, most of the currently existing profiles treat the mouse as a second class citizen. The focus primarily has been on keyboard effects, until now. There are some Lua scripts that paint the full canvas, so this will include mouse support that is "by-design". More complex effects tend to be specific to a single device class though, but the different Lua scripts can be combined in a profile as needed.

Table of profiles and their respective device support:
| Profile     | Type |
| ------------- | ------------- |
| Batique: batique.profile | Paints full canvas |
| Batique (Mouse only): batique-mouse.profile | Device specific: Mouse, Keyboard: Minimal support |
| Blue FX: profile1.profile | Device specific: Keyboard, Mouse: Solid color only |
| Checkerboard: checkerboard.profile | Paints full canvas |
| Color Swirls (Perlin): swirl-perlin.profile | Device specific: Keyboard, Mouse: Special treatment* |
| Color Swirls (Perlin): Blue and Red: swirl-perlin-blue-red.profile | Device specific: Keyboard, Mouse: Special treatment* |
| Color Swirls (Perlin): Rainbow: swirl-perlin-rainbow.profile | Device specific: Keyboard, Mouse: Special treatment* |
| Color Swirls (Perlin): Red and Yellow: swirl-perlin-red-yellow.profile | Device specific: Keyboard, Mouse: Special treatment* |
| Color Swirls (Turbulence): swirl-turbulence.profile | Device specific: Keyboard, Mouse: Special treatment* |
| Color Swirls (Voronoi): swirl-voronoi.profile | Device specific: Keyboard, Mouse: Special treatment* |
| FX1: fx1.profile | Device specific: Keyboard, Mouse: Solid color only |
| FX2: fx2.profile | Device specific: Keyboard, Mouse: Solid color only |
| Fireplace: fireplace.profile | Device specific: Keyboard, Mouse: No support |
| Fireworks: fireworks.profile | Device specific: Keyboard, Mouse: Solid color only |
| Flight (Perlin): flight-perlin.profile | Device specific: Keyboard, Mouse: No Support |
| Gaming: Generic: gaming.profile | Device specific: Keyboard, Mouse: Solid color only |
| Gaming: StarCraft 2: starcraft2.profile | Device specific: Keyboard, Mouse: Solid color only |
| Gradient Noise: gradient-noise.profile | Paints full canvas |
| Heartbeat: System Monitor: heartbeat-sysmon.profile | Device specific: Keyboard, Mouse: No support |
| Heat Map: heatmap.profile | Device specific: Keyboard, Mouse: No support |
| Heat Map (Typing Errors): heatmap-errors.profile | Device specific: Keyboard, Mouse: No support |
| Linear Gradient: profile4.profile | Paints full canvas |
| Matrix: matrix.profile | Device specific: Keyboard, Mouse: Solid color only |
| Network FX: netfx.profile | Device specific: Keyboard, Mouse: No Support |
| Organic FX: default.profile | Paints full canvas |
| Preset: Blue and Red: preset-blue-red.profile | Paints full canvas |
| Preset: Red and Yellow: preset-red-yellow.profile | Paints full canvas |
| Psychedelic Smooth: psychedelic.profile | Paints full canvas |
| Psychedelic Twinkle: twinkle.profile | Paints full canvas |
| Rainbow: rainbow.profile | Paints full canvas |
| Rainbow + Wave: rainbow-wave.profile | Paints full canvas |
| Red FX: red-fx.profile | Device specific: Keyboard, Mouse: Solid color only |
| Red Wave: red-wave.profile | Device specific: Keyboard, Mouse: Solid color only |
| Snake: snake.profile | Device specific: Keyboard, Mouse: Solid color only |
| Solid Color + Wave: solid-wave.profile | Device specific: Keyboard, Mouse: Solid color only |
| Spectrum Analyzer: spectrum-analyzer.profile | Device specific: Keyboard, Mouse: No support |
| Spectrum Analyzer + Color Swirls (Perlin): spectrum-analyzer-swirl.profile | Device specific: Keyboard, Mouse: Special treatment* |
| Spectrum Analyzer + Stripes: profile3.profile | Device specific: Keyboard, Mouse: No support |
| Stripes: profile2.profile | Device specific: Keyboard, Mouse: No support |
| Turbulence: turbulence.profile | Paints full canvas |
| VU-Meter: vu-meter.profile | Device specific: Keyboard, Mouse: Solid color only |

*Special treatment: I added some kind of special treatment for mouse devices to this profile
No support: Mouse LEDs are dark

>Btw. how is Batique supposed to look like, do you maybe have a reference image of the pattern or video? I don't have a Roccat keyboard myself to see how it looks on the keyboard.

I will push out the latest changes to the `master` branch soon. You may want to check the preview release of the new `eruption-gui`, it features a realtime view of the canvas, drawn on a virtual keyboard (nearly finished) and a grid view that shows the state of the "mouse zone" of the canvas.

I am planning to add more mouse-specific Lua scripts and profiles in the near future!

7ritn, 2020-12-09:
Sounds awesome! Is there even anything left for me to do? :D
With all the cool scripts, I'm really envious, that my keyboard will never support any of these. Well guess it's time to buy a Roccat keyboard XD

X3n0m0rph59, 2021-04-07:
I will close this issue now! Please reopen it or create a new one if you encounter further problems!

closed 2021-04-07 by X3n0m0rph59