eruption-project/eruption #32: Volume control stops working

MauritsvanderZee, 2020-12-14:
Hi X3n0m0rph59,

First of all thank you for your awesome project! Finally I can use my Roccat Vulcan 120 with RGB support on Linux.
I as well as a friend of mine (with same keyboard) have the issue that our volume control wheel (top right corner) stopped working as soon as we loaded a profile.

Is there a fix for this?

Best regards
Maurits

X3n0m0rph59, 2020-12-14:
Hi @MauritsvanderZee, thanks for reporting this issue!

Could you please tell me what distro and desktop environment you are using?

MauritsvanderZee, 2020-12-14:
Hi @X3n0m0rph59 ,

I am using Manjaro Linux with KDE and my friend is using Ubuntu 18.04 with Gnome. 

Best regards
Maurits

X3n0m0rph59, 2020-12-14:
@MauritsvanderZee So far I was unable to reproduce this issue on Ubuntu 18.04 using the Gnome desktop environment. Will try Manjaro with KDE desktop later. Please note that on Ubuntu 18.04, the volume level indication on the keyboard is disabled when changing the volume using the dial in the upper right corner. On Manjaro you should see the current volume level of the primary sound device, displayed as highlighted colors.

Could you please post the output of the following command:
```sh
$ sudo journalctl -u eruption -b
```

I am especially interested in the dump of the firmware revision. It should read `Firmware revision: 136`, which is the current version of the firmware for the ROCCAT Vulcan 120.

If the firmware revision is below the supported version, Eruption will print a warning message on startup. In this case, please update the firmware using ROCCAT Swarm on Windows 10.

MauritsvanderZee, 2020-12-15:
Dear @X3n0m0rph59 ,
So the issue we have is **not** that the actual indication is not showing up. The volume itself is not adjusting at all. So if we "spin" the dial in the upper right corner nothing happens.

In the meantime I checked my firmware. It is version 136.
Please also see my output of the command you provided:

```
> sudo journalctl -u eruption -b
-- Logs begin at Mon 2020-11-09 22:23:36 CET, end at Tue 2020-12-15 12:55:38 CET. --
Dec 15 11:16:29 zulu systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Dec 15 11:16:29 zulu systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.18
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption > Loading saved state...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption > Registering plugins...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                       > Plugins loaded and initialized successfully
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                       > Enumerating connected devices...
Dec 15 11:16:29 zulu eruption[746]:  ERROR eruption::hwdevices            > Failed to query HID device class: No compatible devices found
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::hwdevices            > Found unsupported mouse device: 0x738:0x1703 (0002:0006:00) - Madcatz Mad Catz R.A.T.3 Mouse
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::hwdevices::generic_mouse > Bound driver: Generic Mouse Device
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::hwdevices                > Found supported keyboard device: 0x1e7d:0x3098 (0002:0004:00) - ROCCAT ROCCAT Vulcan AIMO
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption::hwdevices::roccat_vulcan_1xx > Bound driver: ROCCAT Vulcan 100/12x
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Opening keyboard device...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Initializing keyboard device...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Configuring keyboard LEDs...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Firmware revision: 136
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Spawning keyboard input thread...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Found mouse device, but mouse support is DISABLED by configuration
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Device enumeration completed
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Performing late initializations...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Initializing D-Bus API...
Dec 15 11:16:29 zulu eruption[746]:  INFO  eruption                               > Late initializations completed
```

My friend however, is on an older firmware version. So on Ubuntu with Gnome this could be the issue.

X3n0m0rph59, 2021-01-16:
@MauritsvanderZee sorry for taking so long to reply! I am still unable to reproduce the issue.

Do you have any updates on this? Did your friend get around to update his keyboard's firmware?

Are you maybe using third party drivers, that could lock the USB HID device in any way?
Do you mind posting the output of `ps -elf`?

To make some progress on this one, could you please run `eruption` from a terminal with `debug` logging enabled, and then post the output?

Mask and stop the daemon:
```sh
$ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
```

Run `eruption` with `debug` logging enabled:
```sh
$ sudo RUST_BACKTRACE=full RUST_LOG=debug eruption
```

Please use the volume dial now.... And then please post the output!

Re-enable the daemon afterwards:
```sh
$ sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

Neovenat0r, 2021-04-15:
My System: POP OS 20.10
Keyboard: Vulcan 122

I have finally found this thread and saw that someone else has the same Problem. Eruption works like a charm but when i use the Volume Knob it doesnt switch volume, it turns down or up the Brightness of the Keyboard. No matter if the FX Key has been pressed or not. This is my output:

INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x6a3:0x762 - Saitek Saitek X52 Pro Flight Control System
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x46d:0xa87 - Logitech G935 Gaming Headset
 DEBUG eruption::hwdevices            > Found supported HID device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x3098 (0001:0003:00) - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption::hwdevices::roccat_vulcan_1xx > Bound driver: ROCCAT Vulcan 100/12x AIMO
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found unsupported HID device: 0x951:0x16df - Kingston HyperX Quadcast
 DEBUG eruption::hwdevices                    > Found unsupported HID device: 0x46d:0xc539 - Logitech USB Receiver
 DEBUG eruption::hwdevices                    > Found unsupported HID device: 0x46d:0xc539 - Logitech USB Receiver
 DEBUG eruption::hwdevices                    > Found unsupported HID device: 0x46d:0xc539 - Logitech USB Receiver
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 WARN  eruption::hwdevices::roccat_vulcan_1xx > Outdated firmware version: 131, should be: >= 136
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: 131
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 DEBUG eruption                               > Registering filesystem watcher...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 DEBUG eruption                               > Entering the main loop now...
 INFO  eruption                               > Switching to profile: /var/lib/eruption/profiles/red-wave.profile
 DEBUG eruption::scripting::script            > Applying parameter Color { name: "color_background", description: "Background color", default: 4294905872 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "opacity", description: "Opacity value in the range 0.0 .. 1.0, where 1.0 is fully opaque", default: 1.0 }
 DEBUG eruption::scripting::script            > Applying parameter Bool { name: "mouse_events", description: "React on mouse events", default: true }
 DEBUG eruption::scripting::script            > Applying parameter Color { name: "color_mouse_click_flash", description: "Color of flash effect shown on mouse button clicks", default: 553626282 }
 DEBUG eruption::scripting::script            > Applying parameter Color { name: "color_mouse_wheel_flash", description: "Color of flash effect shown on mouse wheel scroll", default: 1084926719 }
 DEBUG eruption::scripting::script            > Applying parameter Color { name: "color_afterglow", description: "Afterglow color", default: 4294967295 }
 DEBUG eruption::scripting::script            > Applying parameter Color { name: "color_step_afterglow", description: "Color value that gets subtracted from afterglow every [n] frames", default: 134217728 }
[... cut ...]
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 6
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 6
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 28
^C DEBUG eruption                               > Left the main loop
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 DEBUG eruption                               > Saving state...
 INFO  eruption                               > Exiting now

X3n0m0rph59, 2021-04-16:
Hi @Neovenat0r!

Thanks for providing the debug logs! It seems that the state machine inside Eruption gets confused somehow. I will look into this...
I noticed that your keyboards firmware revision is not the latest one that is available! Do you mind updating using `ROCCAT Swarm` on a Windows machine, and then test again?

X3n0m0rph59, 2021-04-24:
Hi @MauritsvanderZee @Neovenat0r,
 
I just published new packages (`eruption-git`) for all supported distros that contain the commit 1432a3f. Please tell me if this fixes the issue for you!

budo76, 2021-07-07:
Really thank you for this great software, and for the support. I cannot talk for @MauritsvanderZee but I think you can close this issue, because I had the exact same problem but, after updating the firmware using Roccat Swarm (on Windows) it suddenly disappeared.. so as you suggested is a matter of upgrading the firmware...

MauritsvanderZee, 2021-07-07:
Hi everyone,

yes my volume wheel is working as well after firmware upgrade on a windows machine. So please feel free to close the issue.

Thanks again for the amazing project @X3n0m0rph59

Maurits

X3n0m0rph59, 2021-07-18:
Great, glad to hear that it works now! I will close this issue now.

closed 2021-07-18 by X3n0m0rph59