eruption-project/eruption #37: Vulcan TKL support?

Madcliff, 2020-12-29:
Does this have support for the new Roccat Vulcan TLK and TKL Pro?

X3n0m0rph59, 2021-01-06:
Hi! The ROCCAT Vulcan TKL series keyboards are not supported yet. I would like to add support for these devices but I don't have the hardware available to test on. If you have access to one of these devices, and are willing to assist, I would need a trace of the device initialization sequence as performed by ROCCAT Swarm, e.g. captured via Wireshark.

Madcliff, 2021-01-06:
Once mine ships from Germany I will get around to giving you the output. 

On a similar question about device compatibility, what was the status with the Roccat Nyth? I tried recently to use the feature set to do with mice but it didn't seem to want to recognise anything.

Madcliff, 2021-01-12:
Ok so how do I get the needed data for the TKL? MIne arrived this morning.

X3n0m0rph59, 2021-01-13:
Please refer to this wiki article on how to get the required protocol traces:
https://github.com/X3n0m0rph59/eruption/wiki

>On a similar question about device compatibility, what was the status with the Roccat Nyth? I tried recently to use the feature set to do with mice but it didn't seem to want to recognise anything.

The ROCCAT Nyth should be recognized by Eruption, but the state of the driver is *experimental* at best. Can you post logs of Eruption? (Please open up a separate issue for this)

```sh
sudo journalctl -u eruption.service -b
```

Madcliff, 2021-01-14:
Roccat TKL linear switches
Should be USB device 27, on hub 1. Hope this is what you require. 

[roccat_tkl, cap, device 27.pcapng.gz](https://github.com/X3n0m0rph59/eruption/files/5813418/roccat_tkl.cap.device.27.pcapng.gz)

X3n0m0rph59, 2021-01-15:
The capture looks good! Thank you very much!

I have implemented a driver module for the ROCCAT Vulcan TKL series keyboards, based on the driver for the ROCCAT Vulcan 1xx series keyboards. 

You may want to run `eruption-debug-tool` first, to test if the new driver is at least able to initialize your device.

```sh
 $ sudo eruption-debug-tool list
 $ sudo eruption-debug-tool run-tests <device-index of your Vulcan TKL, subdev 0>
```

Do you want to build from source, or do you prefer me to build a package for you?

Madcliff, 2021-01-15:
I would appreciate an arch PKG build if possible. I haven't really worked out how to install the drivers from source.

X3n0m0rph59, 2021-01-16:
I have updated the PKGBUILD now on the `master` branch.

To build a package from the current state of the `master` branch please perform these steps:

* Unzip the attached files from [eruption-PKGBUILD.zip](https://github.com/X3n0m0rph59/eruption/files/5823862/eruption-PKGBUILD.zip)
* Run `makepkg -crsi` inside the unzipped directory
* Run the command `sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service` to restart the daemon

Madcliff, 2021-01-16:
So it would appear the daemon is panicking

```[mcliff@malleus eruption]$ sudo eruption-debug-tool run-tests 00

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service

Index: 00: ID: 1e7d:2fee ROCCAT/ROCCAT Vulcan TKL Subdev: 0
The application panicked (crashed).
Message:  Could not open LED sub-device: OpenHidDeviceError
Location: eruption-debug-tool/src/hwdevices/mod.rs:113

Backtrace omitted.
Run with RUST_BACKTRACE=1 environment variable to display it.
Run with RUST_BACKTRACE=full to include source snippets.
Aborted
```

X3n0m0rph59, 2021-01-16:
Thanks for testing!

Did you run `sudo systemctl mask eruption.service && sudo systemctl stop eruption.service` prior to running the `eruption-debug-tool`? Otherwise the control device and maybe the LED device will be locked by `eruption`.

Could you please post the output of:
```sh
 $ sudo eruption-debug-tool list
```

Madcliff, 2021-01-16:
[output.log](https://github.com/X3n0m0rph59/eruption/files/5824291/output.log)

Madcliff, 2021-01-16:
I chedked the dmsg logs, found this 

```Jan 16 11:28:41 malleus eruption[1144194]:  WARN  eruption::hwdevices::roccat_vulcan_tkl > Outdated firmware version: 115, should be: >136```

X3n0m0rph59, 2021-01-16:
Thanks for reporting this, I just corrected the current firmware revision of the ROCCAT Vulcan TKL series keyboard to be `115`.
The revision of `136` does only apply to the  ROCCAT Vulcan 1xx series keyboards.

Do you notice a change in behavior when you select a different sub device when running `eruption-debug-tool run-tests`?
Is the driver able to set the LED colors when you specify a different sub-device of your ROCCAT Vulcan TKL?

Madcliff, 2021-01-16:
Only sub device 3 crashes. the others just seem to set the colour of the keyboard to the random rainbow pattern that is the default.

X3n0m0rph59, 2021-01-17:
> Only sub device 3 crashes. the others just seem to set the colour of the keyboard to the random rainbow pattern that is the default.

It should initialize the device to AIMO mode, and show a red and blue test pattern.

Did the Wireshark capture file that you made capture the initialization sequence to AIMO mode, or did it capture initialization to the rainbow pattern? 

Could you please do a second capture of the device being initialized by Swarm to AIMO mode? I am especially looking for captures of the packets starting with "0xa1, 0x01, 0x01, 0xb4", which should be fairly large and set the color of each of the LEDs on the keyboard device. These packets were missing from the first capture file, so it could very well be that it just contained initialization to the rainbow pattern.

X3n0m0rph59, 2021-01-20:
@Madcliff I just pushed out the latest changes to the driver code. The Wireshark capture that you created seems to be equivalent to the initialization sequence of the ROCCAT Vulcan 100/12x series devices (into rainbow mode). 

I now changed the ROCCAT Vulcan TKL driver to use the init sequence of the ROCCAT Vulcan 100/12x keyboards (into AIMO mode). There is at least a slight chance that this will work.

You may compile the latest release using this PKGBUILD:
[eruption-PKGBUILD.zip](https://github.com/X3n0m0rph59/eruption/files/5841531/eruption-PKGBUILD.zip)

(The `eruption` daemon as well as the `eruption-debug-tool` are using the latest code)

Madcliff, 2021-01-21:
That package just seems to break everything. Had to revert to package 18.4 to get the drivers working again

X3n0m0rph59, 2021-01-21:
@Madcliff Then could you please try again to capture the init sequence into AIMO mode, as done by ROCCAT Swarm?

Madcliff, 2021-01-21:
I will do! Just taking some time to do it because I need to remember how to enable the USB listener before I run Qemu.

Sent from Mailspring (https://getmailspring.com/), the best free email app for work

Madcliff, 2021-01-21:
Device 1.13 is the TKL. I set it into snake mode first, and then into AIMO 
[wireshark-tkl.zip](https://github.com/X3n0m0rph59/eruption/files/5851321/wireshark-tkl.zip)

X3n0m0rph59, 2021-01-22:
@Madcliff Thank you very much! I have updated `eruption` as well as `eruption-debug-tool`, according to the new Wireshark capture file.

You may use the `eruption-debug-tool`, as described above, to try to initialize the ROCCAT Vulcan TKL to a red/blue test pattern.

[eruption-PKGBUILD.zip](https://github.com/X3n0m0rph59/eruption/files/5854032/eruption-PKGBUILD.zip)

Madcliff, 2021-01-22:
Still a no go. Will give it another go after the weekend is over.

Madcliff, 2021-01-28:
INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
 INFO  eruption > Loading saved state...
 WARN  eruption::state > Invalid saved state: slot_names
 INFO  eruption        > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x2fee (0001:000a:00) - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices::roccat_vulcan_tkl > Bound driver: ROCCAT Vulcan TKL
 INFO  eruption::hwdevices                    > Found supported mouse device: 0x1e7d:0x2e7c (0001:0007:00) - ROCCAT ROCCAT Nyth
 INFO  eruption::hwdevices::roccat_nyth       > Bound driver: ROCCAT Nyth
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: 115
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Opening mouse device...
 INFO  eruption                               > Now listening on keyboard: /dev/input/event30
 INFO  eruption                               > Input device name: "ROCCAT ROCCAT Vulcan TKL"
 INFO  eruption                               > Input device ID: bus 0x3 vendor 0x1e7d product 0x2fee
 INFO  eruption                               > Physical location: usb-0000:00:14.0-1.1.2/input0
 INFO  eruption                               > Grabbing the keyboard device exclusively
 INFO  eruption                               > Initializing mouse device...
 INFO  eruption                               > Configuring mouse LEDs...
 INFO  eruption                               > Firmware revision: 138
 INFO  eruption                               > Spawning mouse input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 INFO  eruption                               > Switching to profile: mcliff-matrix.profile
 INFO  eruption                               > Now listening on mouse: /dev/input/event21
 INFO  eruption                               > Input device name: "ROCCAT ROCCAT Nyth Mouse"
 INFO  eruption                               > Input device ID: bus 0x3 vendor 0x1e7d product 0x2e7c
 INFO  eruption                               > Physical location: usb-0000:00:14.0-1.1.1/input0
 INFO  eruption                               > Grabbing the mouse device exclusively
^C INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 INFO  eruption                               > Exiting now

Madcliff, 2021-01-28:
So I will be honest, I suspect the TKL and the Regular keyboard config's are clashing and this is meaning I cannot control the keyboard.

Madcliff, 2021-01-28:
When I force restart the demon the keys get to a strange flickering pattern rather than whatever profile is set.

RoggeHaj, 2021-01-30:
Hi, I ordered a Vulcan Pro TKL that should arrive tuesday. In the meantime I set up the development environment on my Ubuntu-20.10 system and is able to build the master branch using

```shell
cargo build --all --release
```
so I can aid with debugging once I get the keyboard if you want.

X3n0m0rph59, 2021-01-31:
@RoggeHaj Much appreciated, thanks!

I am currently working on full support for the ROCCAT Vulcan Pro TKL keyboard. Will push the latest changes to the `master` branch in a few days.

I hope that this work will aid in getting the ROCCAT Vulcan Pro and the ROCCAT Vulcan TKL supported too.

X3n0m0rph59, 2021-02-02:
@RoggeHaj I have just pushed out the latest changes to the `master` branch. We now have working (experimental) support for ROCCAT Vulcan Pro TKL series keyboards!

Madcliff, 2021-02-02:
I will give this a test once I get my system working again! Ended up without a computer due to reasons!

Sent from Mailspring (https://link.getmailspring.com/link/1A514F3A-514E-437E-B8BA-C58D62AC7AD6@getmailspring.com/0?redirect=https%3A%2F%2Fgetmailspring.com%2F&recipient=cmVwbHkrQUNWQVhKU1NENUJNR1ZFT0c1RkdMQUY2RVQ1VFRFVkJOSEhDNFFGNFFNQHJlcGx5LmdpdGh1Yi5jb20%3D), the best free email app for work

X3n0m0rph59, 2021-02-02:
@Madcliff Ok, thank you! I will update the code to support the "non Pro" variant of the ROCCAT Vulcan TKL with mostly the same code as the Pro variant. But please don't expect too much! I can't promise that it will work, since I can only test on the Pro variant of the keyboard.

RoggeHaj, 2021-02-02:
@X3n0m0rph59 Sweet. I just got a message from the delivery guys that the keyboard will be delivered in roughly 8 hrs so I can give it a go tonite.

I rebuilt the latest master without any issues, but I guess I need to properly install it to get the service working properly? In the README, the last step in the `From Source` section is the build step so I'm kind of dead in the water here.

X3n0m0rph59, 2021-02-02:
I have built a pre-release package suitable for `Ubuntu 20.04 LTS` and `Ubuntu 20.10`.

[eruption-git_0.1.19-ubuntu9_amd64.deb.zip](https://github.com/X3n0m0rph59/eruption/files/5910526/eruption-git_0.1.19-ubuntu9_amd64.deb.zip)

Please install using the following commands:
```sh
 $ unzip eruption-git_0.1.19-ubuntu9_amd64.deb.zip
 $ sudo apt install ./eruption-git_0.1.19-ubuntu9_amd64.deb
```

X3n0m0rph59, 2021-02-02:
@Madcliff I have adapted Eruption to mostly use the code from the Pro variant for the driver module for ROCCAT Vulcan TKL series keyboards.

I can confirm that this works with the ROCCAT Vulcan TKL Pro. As said before I don't have a  ROCCAT Vulcan TKL (non-pro) to test on.

[eruption-PKGBUILD.zip](https://github.com/X3n0m0rph59/eruption/files/5911672/eruption-PKGBUILD.zip)

RoggeHaj, 2021-02-02:
So, I tried the pre-release package and I must say that I'm really impressed with it. The only thing I've found so far is that after reboot the Win-key was not working until i fired up the eruption-gui.

X3n0m0rph59, 2021-02-03:
@RoggeHaj Thanks for your feedback! If you find more bugs, then please feel free to open up issues any time!

Madcliff, 2021-02-03:
So with the new package it does the initialization then it seems to freeze for me. Sent on the go with Vodafone

X3n0m0rph59, 2021-02-03:
@Madcliff Could you please provide the output of the following commands?

```sh
 $ sudo journalctl -u eruption.service -b
```

```sh
 $ sudo systemctl stop eruption.service
 $ sudo RUST_BACKTRACE=full RUST_LOG=debug eruption
```

I am working on another package, please stay tuned...

Madcliff, 2021-02-03:
Output of both commands

-- Journal begins at Tue 2021-02-02 10:49:16 GMT, ends at Wed 2021-02-03 09:26:08 GMT. --
Feb 02 19:20:42 archlinux systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Feb 02 19:20:42 archlinux systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption > Loading saved state...
Feb 02 19:20:42 archlinux eruption[16729]: WARN eruption::state > Invalid saved state: slot_names
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption > Registering plugins...
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: System - Basic system information and status
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::plugins::persistence > Loading persistent state data from disk...
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption > Plugins loaded and initialized successfully
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption > Enumerating connected devices...
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::hwdevices > Found supported mouse device: 0x1e7d:0x2e7c (0005:0009:00) - ROCCAT ROCCAT Nyth
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::hwdevices::roccat_nyth > Bound driver: ROCCAT Nyth
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::hwdevices > Found supported keyboard device: 0x1e7d:0x2fee (0005:0008:00) - ROCCAT ROCCAT Vulcan TKL
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption::hwdevices::roccat_vulcan_tkl > Bound driver: ROCCAT Vulcan TKL
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption > Opening keyboard device...
Feb 02 19:20:42 archlinux eruption[16729]: INFO eruption > Initializing keyboard device...
Feb 02 19:20:43 archlinux eruption[16729]: ERROR eruption::hwdevices::roccat_vulcan_tkl > Invalid result

INFO eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
INFO eruption > Loading saved state...
WARN eruption::state > Invalid saved state: slot_names
INFO eruption > Registering plugins...
INFO eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
INFO eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
INFO eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
INFO eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
INFO eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
INFO eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
INFO eruption::plugin_manager > Registering plugin: System - Basic system information and status
INFO eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
INFO eruption::plugin_manager > Registering plugin: Audio - Audio related functions
INFO eruption::plugins::persistence > Loading persistent state data from disk...
INFO eruption > Plugins loaded and initialized successfully
INFO eruption > Enumerating connected devices...
DEBUG eruption::hwdevices > Found unsupported HID device: 0x48d:0xc100 - ITE Tech. Inc. ITE Device(8910)
DEBUG eruption::hwdevices > Found unsupported HID device: 0xfd9:0x6c - Elgato Stream Deck XL
DEBUG eruption::hwdevices > Found supported HID device: 0x1e7d:0x2e7c - ROCCAT ROCCAT Nyth
INFO eruption::hwdevices > Found supported mouse device: 0x1e7d:0x2e7c (0005:000f:00) - ROCCAT ROCCAT Nyth
INFO eruption::hwdevices::roccat_nyth > Bound driver: ROCCAT Nyth
DEBUG eruption::hwdevices > Found supported HID device: 0x1e7d:0x2e7c - ROCCAT ROCCAT Nyth
DEBUG eruption::hwdevices > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
INFO eruption::hwdevices > Found supported keyboard device: 0x1e7d:0x2fee (0005:0011:00) - ROCCAT ROCCAT Vulcan TKL
INFO eruption::hwdevices::roccat_vulcan_tkl > Bound driver: ROCCAT Vulcan TKL
DEBUG eruption::hwdevices > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
[... cut ...]
ERROR eruption > Could not send the LED map to the device: Device not opened
ERROR eruption > Could not send the LED map to the device: Device not opened
ERROR eruption > Could not send the LED map to the device: Device not opened

Sent from Mailspring (https://link.getmailspring.com/link/ABF829EA-63A9-4920-8613-C496B7A58E54@getmailspring.com/0?redirect=https%3A%2F%2Fgetmailspring.com%2F&recipient=cmVwbHkrQUNWQVhKVVRZS0dQVVlVTDUyQlRXRUY2RVpFM1pFVkJOSEhDNFFGNFFNQHJlcGx5LmdpdGh1Yi5jb20%3D), the best free email app for work

X3n0m0rph59, 2021-02-03:
I changed the way that Eruption sends LED state data to the keyboard. But judging from the log output, I bet that it still won't work, because Eruption fails to initialize the device earlier. Could you please redo the Wireshark packet capture? I mostly need the initialization sequence that SWARM sends to the device (to AIMO mode). This initialization sequence is missing from the previous packet captures.

You can get the data like this:

* Prepare for capture of USB packets (modprobe usbmon)
* Run Wireshark and start the capture
* Then boot your Windows 10 VM and quickly connect the Keyboard to the VM as it boots
* Wait until SWARM has initialized the keyboard and AIMO effect is running for a few seconds
* Now you can disconnect the device from the VM and save the packet capture

I have updated the PKGBUILD on the `master` branch.
[eruption-PKGBUILD.zip](https://github.com/X3n0m0rph59/eruption/files/5917444/eruption-PKGBUILD.zip)

To build a package from the current state of the master branch please perform these steps:

* Unzip the attached files from eruption-PKGBUILD.zip
* Run `makepkg -crsi` inside the unzipped directory
* Run the command `sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service` to restart the daemon

RoggeHaj, 2021-02-04:
I noticed a discrepancy between the key assignment in the eruption-gui and the keys on the Vulcan TKL keyboard. It was very obvious for the `Gaming: Generic` profile as can be seen in the screenshot
[image]

On the keyboard, the `W` `A` `S` `D` are lit as intended.

X3n0m0rph59, 2021-02-04:
@RoggeHaj thanks for reporting this issue! The GUI is not ready for prime time yet, it is still missing many even the most basic features. The next tech preview `0.0.3` will include support for TKL keyboard variants. I am currently working on that. Version `0.0.2` of the GUI just supports the standard form factor keyboards (ROCCAT Vulcan 100/12x). 

I will build a new pre-release package in a few days, please stay tuned!

Madcliff, 2021-02-04:
I have tried to do another USB dump. Let's see if it has what you need. it defaulted to AIMO, I moved the settings from AIMO, but always back in the direction of AIMO. It should be device 30 on hub 5, and device 31 after I reconnected it.

Sent from Mailspring (https://link.getmailspring.com/link/B6A4C0A2-D472-4D6E-B75F-4DA36E9AAAA6@getmailspring.com/0?redirect=https%3A%2F%2Fgetmailspring.com%2F&recipient=cmVwbHkrQUNWQVhKVFhQNFZMQjc3QjQ1QzNUQUY2RTdFV1RFVkJOSEhDNFFGNFFNQHJlcGx5LmdpdGh1Yi5jb20%3D), the best free email app for work

X3n0m0rph59, 2021-02-04:
@Madcliff The attached file seems to be missing!

Madcliff, 2021-02-04:
[usbdump.pcapng.gz](https://github.com/X3n0m0rph59/eruption/files/5927018/usbdump.pcapng.gz)

Madcliff, 2021-02-04:
That worked!

X3n0m0rph59, 2021-02-04:
@Madcliff Thank you for the capture file! The trace contained much more usable information this time. I have adapted the respective driver sub-module. There is at least a slight chance that it might work!

[eruption-PKGBUILD.zip](https://github.com/X3n0m0rph59/eruption/files/5927870/eruption-PKGBUILD.zip)

Madcliff, 2021-02-04:
[mcliff@archlinux eruption]$ sudo systemctl stop eruption.service
[mcliff@archlinux eruption]$ sudo RUST_BACKTRACE=full RUST_LOG=debug eruption

 Eruption is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 Eruption is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
 INFO  eruption > Loading saved state...
 WARN  eruption::state > Invalid saved state: slot_names
 INFO  eruption        > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x48d:0xc100 - ITE Tech. Inc. ITE Device(8910)
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0xfd9:0x6c - Elgato Stream Deck XL
 DEBUG eruption::hwdevices            > Found supported HID device: 0x1e7d:0x2e7c - ROCCAT ROCCAT Nyth
 INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2e7c (0005:0009:00) - ROCCAT ROCCAT Nyth
 INFO  eruption::hwdevices::roccat_nyth > Bound driver: ROCCAT Nyth
 DEBUG eruption::hwdevices              > Found supported HID device: 0x1e7d:0x2e7c - ROCCAT ROCCAT Nyth
 DEBUG eruption::hwdevices              > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices              > Found supported keyboard device: 0x1e7d:0x2fee (0005:000c:00) - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices::roccat_vulcan_tkl > Bound driver: ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: 115
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Opening mouse device...
 ERROR eruption                               > Error opening the mouse device: Could not open the device file
 ERROR eruption                               > This could be a permission problem, or maybe the device is locked by another process?
 INFO  eruption                               > Initializing mouse device...
 ERROR eruption                               > Could not initialize the device: Device not opened
 INFO  eruption                               > Configuring mouse LEDs...
 ERROR eruption                               > Could not initialize LEDs: Device not opened
 INFO  eruption                               > Firmware revision: <unknown>
 INFO  eruption                               > Spawning mouse input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 DEBUG eruption                               > Registering filesystem watcher...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 DEBUG eruption                               > Entering the main loop now...
 INFO  eruption                               > Switching to profile: netfx.profile
 DEBUG eruption::dbus_interface               > Message: Message { Type: Signal, Path: "/org/freedesktop/DBus", Interface: "org.freedesktop.DBus", Member: "NameAcquired", Sender: "org.freedesktop.DBus", Destination:  [... cut ...]
 INFO  eruption::scripting::script::callbacks > Network FX: Server shutting down
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 ERROR eruption                               > Could not finalize LEDs configuration: Device not opened
 WARN  eruption                               > Could not close the device: Device not opened
 DEBUG eruption                               > Saving state...
 INFO  eruption                               > Exiting now

Though I feel like the actual config is working, just not the LED's

Madcliff, 2021-02-04:
Feb 04 21:54:59 archlinux systemd[1]: Stopped Linux user-mode input and LED driver for keyboards, mice and other devices.
Feb 04 21:55:08 archlinux systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Feb 04 21:55:08 archlinux systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption > Loading saved state...
Feb 04 21:55:08 archlinux eruption[4596]:  WARN  eruption::state > Invalid saved state: slot_names
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption        > Registering plugins...
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption                       > Plugins loaded and initialized successfully
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption                       > Enumerating connected devices...
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2e7c (0005:0009:00) - ROCCAT ROCCAT Nyth
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::hwdevices::roccat_nyth > Bound driver: ROCCAT Nyth
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::hwdevices              > Found supported keyboard device: 0x1e7d:0x2fee (0005:000b:00) - ROCCAT ROCCAT Vulcan TKL
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption::hwdevices::roccat_vulcan_tkl > Bound driver: ROCCAT Vulcan TKL
Feb 04 21:55:08 archlinux eruption[4596]:  INFO  eruption                               > Opening keyboard device...
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Initializing keyboard device...
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Configuring keyboard LEDs...
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Firmware revision: 115
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Spawning keyboard input thread...
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Opening mouse device...
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Initializing mouse device...
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Now listening on keyboard: /dev/input/event25
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Input device name: "ROCCAT ROCCAT Vulcan TKL"
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Input device ID: bus 0x3 vendor 0x1e7d product 0x2fee
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Physical location: usb-0000:06:00.4-1.2.1.1/input0
Feb 04 21:55:09 archlinux eruption[4596]:  INFO  eruption                               > Grabbing the keyboard device exclusively
Feb 04 21:55:14 archlinux eruption[4596]:  ERROR eruption                               > Fatal: Keyboard device went away: No such device (os error 19)

Madcliff, 2021-02-05:
More findings, running the backtrace sorta works, but it is very buggy. When I run sudo backtrace it applies the LED profiles even while telling me that the LED device cannot be accessed, But running the driver normally doesn't work at all. 

Also macros appear to not work yet :-D

X3n0m0rph59, 2021-02-05:
@Madcliff So did you see the LEDs light up? The logs from your latest post seem to indicate that it successfully initialized the keyboard device.

Do you see a red and blue test pattern when you run the commands:

```sh
$ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
$ sudo eruption-debug-tool list

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 

Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other sub-commands of this tool

Index: 00: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 0
Index: 01: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 1
Index: 02: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 2
Index: 03: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 3
Index: 04: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 0
Index: 05: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 1
Index: 06: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 2
Index: 07: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 3

Enumeration completed
```

On my system, subdev 1 of the ROCCAT Vulcan Pro TKL has index 5. So I continue with that one...

```sh
$ sudo RUST_LOG=debug eruption-debug-tool run-tests 5 

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 
Index: 05: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 1
Bound driver: ROCCAT Vulcan Pro TKL
```

The keyboard should now switch from showing red LEDs to showing blue LEDs.
Please tell me if this works for you!

To re-enable the systemd unit please run this command:

```sh
$ sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

Madcliff, 2021-02-05:
Yes doing that worked perfectly fine!  lights switched from red to blue.

Madcliff, 2021-02-09:
The switching between though only works when I start the service in the debug mode. Trying to start the service normally just bugs out.

X3n0m0rph59, 2021-02-12:
Hi @RoggeHaj, @Madcliff 
I have built a new pre-release package suitable for `Ubuntu 20.04 LTS` and `Ubuntu 20.10`, and a new PKGBUILD as well.

It contains a few new features, general bug fixes and fixes to the initialization sequences of most supported devices.

[eruption-git_0.1.19-18_amd64.deb.zip](https://github.com/X3n0m0rph59/eruption/files/5970472/eruption-git_0.1.19-18_amd64.deb.zip)

@Madcliff please tell me if that fixes the problem with Eruption not starting/initializing on your system.
[eruption-PKGBUILD.zip](https://github.com/X3n0m0rph59/eruption/files/5970476/eruption-PKGBUILD.zip)

RoggeHaj, 2021-02-15:
> Hi @RoggeHaj, @Madcliff
> I have built a new pre-release package suitable for `Ubuntu 20.04 LTS` and `Ubuntu 20.10`, and a new PKGBUILD as well.
>
> It contains a few new features, general bug fixes and [...]

Great. I will look into it during the day.

Madcliff, 2021-02-15:
@X3n0m0rph59 Still no joy with it starting.

Madcliff, 2021-02-15:
It appears that the last subdev on the TKL craches. The first two work fine and the keyboard will initialize to blue. But the last will do the following. 

Index: 07: ID: 1e7d:2fee ROCCAT/ROCCAT Vulcan TKL Subdev: 3
The application panicked (crashed).
Message:  Could not open LED sub-device: OpenHidDeviceError
Location: eruption-debug-tool/src/hwdevices/mod.rs:132

Backtrace omitted.
Run with RUST_BACKTRACE=1 environment variable to display it.
Run with RUST_BACKTRACE=full to include source snippets.
Aborted

X3n0m0rph59, 2021-02-15:
@Madcliff Thanks for testing this again!

Could you please show me the output of the following commands:
```sh
$ pacman -Q eruption-git
```
This should print `eruption-git 0.1.19-23`. If you don't have this installed then please update to the latest version.

Then please post again the full output of the following command:
```sh
$ sudo RUST_LOG=debug eruption
```

With this information I should be able to see at which step of the initialization sequence it bugs out!

X3n0m0rph59, 2021-02-15:
@Madcliff Please post the output of this command too:

```sh
$ sudo lsof /dev/bus/usb/*/*
```

This will allow us to rule out any errors caused by other software grabbing the device file.

Madcliff, 2021-02-15:
[mcliff@archlinux ~]$ pacman -Q eruption-git
eruption-git 0.1.19-23
[mcliff@archlinux ~]$ sudo RUST_LOG=debug eruption
[sudo] password for mcliff: 

 Eruption is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 Eruption is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x48d:0xc100 - ITE Tech. Inc. ITE Device(8910)
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0xfd9:0x6c - Elgato Stream Deck XL
 DEBUG eruption::hwdevices            > Found supported HID device: 0x1e7d:0x2e7c - ROCCAT ROCCAT Nyth
 INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2e7c (0005:000c:00) - ROCCAT ROCCAT Nyth
 INFO  eruption::hwdevices::roccat_nyth > Bound driver: ROCCAT Nyth
 DEBUG eruption::hwdevices              > Found supported HID device: 0x1e7d:0x2e7c - ROCCAT ROCCAT Nyth
 DEBUG eruption::hwdevices              > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices              > Found supported keyboard device: 0x1e7d:0x2fee (0005:000e:00) - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices::roccat_vulcan_tkl > Bound driver: ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: 115
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Opening mouse device...
 INFO  eruption                               > Initializing mouse device...
 DEBUG eruption::hwdevices                    > Picking evdev device: /dev/input/event22
 INFO  eruption                               > Now listening on keyboard: /dev/input/event22
 INFO  eruption                               > Input device name: "ROCCAT ROCCAT Vulcan TKL"
 INFO  eruption                               > Input device ID: bus 0x3 vendor 0x1e7d product 0x2fee
 INFO  eruption                               > Physical location: usb-0000:06:00.4-1.1.1.2/input0
 INFO  eruption                               > Grabbing the keyboard device exclusively

X3n0m0rph59, 2021-02-15:
@Madcliff Strange, it does not log any errors at all...

Could you please post the full output of this command:
```sh
$ sudo RUST_LOG=trace eruption
```

[12 more replies]