eruption-project/eruption #40: Nyth buttons not binding

Madcliff, 2021-01-15:
Buttons for the Nynth does not bind properly 
[log.txt](https://github.com/X3n0m0rph59/eruption/files/5821399/log.txt)

Madcliff, 2021-01-16:
Ok, so using the PKG build for the TKL, the driver partially functions, but the hid mapping seems to be more than a little hit and miss. I do not think it is correctly mapping the device keyboard properly.

X3n0m0rph59, 2021-01-16:
Could you please run `eruption` with `debug` log-level enabled, and post a log of clicking the mouse buttons and the generated output?

To enable the `debug` log-level please perform these actions:

Stop and mask the daemon:
```sh
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
```

Run the `eruption` daemon from a terminal
```sh
sudo RUST_BACKTRACE=full RUST_LOG=debug eruption
```

Afterwards re-enable the `eruption` daemon
```sh
 $ sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

Madcliff, 2021-01-16:
This all looks pretty strange to me, I started with the top row of 4 buttons, then proceeded to the middle row, and finally the bottom row. It does seem to regularly suffer from the missed HID event

 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(13)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 14
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(13)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 14
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 9
 DEBUG eruption                               > FPS: 18
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(9)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 10
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(9)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 10
 DEBUG eruption                               > FPS: 17
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(9)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 10
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(9)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 10
 DEBUG eruption                               > FPS: 17
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(9)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 10
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(9)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 10
 DEBUG eruption                               > FPS: 18
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(10)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 11
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 9
 DEBUG eruption                               > FPS: 18
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(9)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 10
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(9)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 10
 DEBUG eruption                               > FPS: 18
 ERROR eruption::hwdevices::roccat_nyth       > We missed a HID event, mouse button states will be inconsistent
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 9
 DEBUG eruption                               > FPS: 18
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 9
 DEBUG eruption                               > FPS: 18
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 9
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(8)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 9
 DEBUG eruption                               > FPS: 18
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(10)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 11
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(13)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 14
[... 7 lines cut ...]
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(10)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 2 args: 11
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonUp(10)
 DEBUG eruption::scripting::script::callbacks > Macros: HID event (mouse): 3 args: 11
 DEBUG eruption::hwdevices::roccat_nyth       > ButtonDown(8)
[... cut ...]
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 2 args: 23
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 6
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 28
^C DEBUG eruption                               > Left the main loop
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 DEBUG eruption                               > Saving state...
 INFO  eruption                               > Exiting now

Madcliff, 2021-01-16:
If I were to report my suspicions, I think the Nyth with its configurations, needs a setting set somewhere for it to recignise that it has twelve buttons rather than other configurations.

X3n0m0rph59, 2021-01-17:
> If I were to report my suspicions, I think the Nyth with its configurations, needs a setting set somewhere for it to recignise that it has twelve buttons rather than other configurations.

I think this may be the case for the evdev event reporting, but on the USB HID layer it should report each of the buttons regardless of their mapping state to a specific function or key.

I suspect that the code that handles the HID event is still wrong. 
Could you please re-build eruption using the attached PKGBUILD, run it from a terminal with `debug` logging enabled, and then please post the logs? I am especially interested in the hexdump of the HID events.

[eruption-PKGBUILD.zip](https://github.com/X3n0m0rph59/eruption/files/5825664/eruption-PKGBUILD.zip)

Madcliff, 2021-01-17:
what is the command to run this in debug mode from the terminal?

X3n0m0rph59, 2021-01-17:
Run the eruption daemon from a terminal with debug logging enabled:

```sh
 $ sudo RUST_BACKTRACE=full RUST_LOG=debug eruption
```

Madcliff, 2021-01-17:
Is this what you are after? 

[nyth.log](https://github.com/X3n0m0rph59/eruption/files/5825875/nyth.log)