eruption-project/eruption #41: Not working with Roccat Vulcan Pro on Ubuntu 20.10

DirkSchueler, 2021-01-22:
I know the software supports this keyboard in an experimental way.
But after an successful installation, there absolutely no change.
I installed also eruption-profile-switcher - extension. 
Seems there is no connection to the keyboard?

Please let me know if I need to provide any logfile, setting, etc.

Thank you for your support.

X3n0m0rph59, 2021-01-22:
Hi, thanks for submitting your bug report!

The support for `ROCCAT Vulcan Pro` series keyboards is only available in the `master` branch of Eruption. That is, what will become version `0.1.19`. It is highly experimental, and as of now, even **completely untested**. It will probably cause serious problems.

If you want to assist in the development of Eruption, you can unzip and install the attached package and help in debugging errors.

[eruption-git_0.1.19-0-ubuntu1_amd64.deb.zip](https://github.com/X3n0m0rph59/eruption/files/5857425/eruption-git_0.1.19-0-ubuntu1_amd64.deb.zip)

DirkSchueler, 2021-01-25:
Hi!

Thank you for providing the software.

I installed it properly and was able to run the eruption service.
I ran the "eruption-process-monitor daemon" command line.

But regardless of what I changed using either the eruption-gui program or
the Eruption Profile Switcher the LEDs on the keyboard never change.

I am running Ubuntu 20.10 with Gnome Desktop 3.38.1.
What kind of information do you need to solve this?

X3n0m0rph59, 2021-01-25:
@DirkSchueler Great, thanks for testing the pre-release package!

To debug the problem, I would need the output of the following commands:

Kernel log:
```sh
$ sudo dmesg -T
```

Eruption daemon log and process monitor log:
```sh
$ sudo journalctl -u eruption.service -b
$ journalctl --user -u eruption-process-monitor.service -b 
```

You may want to run the Eruption daemon from  a terminal to set the log level to `debug`:
```sh
$ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
$ sudo RUST_BACKTRACE=full RUST_LOG=debug eruption
```

> But regardless of what I changed using either the eruption-gui program or
the Eruption Profile Switcher the LEDs on the keyboard never change.

Do you see any kind of activity after the daemon is started, e.g. are the LED effects working? Or does the daemon crash after initializing the keyboard?

* Does it switch profiles when you press the `FN + F<num>` keys? For example: `FN + F2` or `FN + F3`?
* Does it switch profiles when you execute: `eruptionctl switch profile swirl-perlin-rainbow.profile`?

To reset the systemd services state of the Eruption daemons please run the following commands:
```sh
$ sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
$ systemctl --user enable --now eruption-process-monitor.service
```

DirkSchueler, 2021-01-26:
Thank you for your fast reply.
Unfortunately none of the commands you provided led to any reaction on the keyboard.
I can change the profiles using FN + F<1-4>.

"eruptionctl switch profile swirl-perlin-rainbow.profile" => "Switching to profile: swirl-perlin-rainbow.profile", but nothing else happens. 

Maybe this one is interesting:

"sudo systemctl status eruption.service" => 
"● eruption.service - Linux user-mode input and LED driver for keyboards, mice and other devices
     Loaded: loaded (/lib/systemd/system/eruption.service; disabled; vendor preset: disabled)
     Active: active (running) since Tue 2021-01-26 18:44:58 CET; 6min ago
       Docs: man:eruption(8)
             man:eruption.conf(5)
             man:eruptionctl(1)
             man:eruption-netfx(1)
   Main PID: 171847 (eruption)
      Tasks: 20 (limit: 19026)
     Memory: 7.6M
     CGroup: /system.slice/eruption.service
             └─171847 /usr/bin/eruption -c /etc/eruption/eruption.conf

Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized
Jan 26 18:51:39 delli-Ubuntu eruption[171847]:  ERROR eruption                               > Could not send the LED map to the device: Device not initialized"

X3n0m0rph59, 2021-01-26:
@DirkSchueler Looks like the Eruption daemon somehow failed to initialize the keyboard.

Could you please mask the `eruption.service` and then run the daemon from a terminal:
```sh
$ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
$ sudo RUST_BACKTRACE=full RUST_LOG=debug eruption
```

With the output generated by the second command, I should be able to narrow down the problem!

DirkSchueler, 2021-01-27:
$ sudo RUST_BACKTRACE=full RUST_LOG=debug eruption

 Eruption is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 Eruption is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
 INFO  eruption > Loading saved state...
 WARN  eruption::state > Invalid saved state: slot_names
 INFO  eruption        > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x4f3:0x201 - ELAN Touchscreen
 DEBUG eruption::hwdevices            > Found supported HID device: 0x1e7d:0x30f7 - ROCCAT ROCCAT Vulcan Pro
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x30f7 (0002:0003:00) - ROCCAT ROCCAT Vulcan Pro
 INFO  eruption::hwdevices::roccat_vulcan_pro > Bound driver: ROCCAT Vulcan Pro
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x30f7 - ROCCAT ROCCAT Vulcan Pro
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x30f7 - ROCCAT ROCCAT Vulcan Pro
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x30f7 - ROCCAT ROCCAT Vulcan Pro
 DEBUG eruption::hwdevices                    > Found unsupported HID device: 0x46d:0xc092 - Logitech G203 LIGHTSYNC Gaming Mouse
 DEBUG eruption::hwdevices                    > Found unsupported HID device: 0x46d:0xc092 - Logitech G203 LIGHTSYNC Gaming Mouse
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 ERROR eruption                               > Could not initialize the device: Invalid result
 INFO  eruption                               > Configuring keyboard LEDs...
 ERROR eruption                               > Could not initialize LEDs: Device not initialized
 INFO  eruption                               > Firmware revision: <unknown>
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 DEBUG eruption                               > Registering filesystem watcher...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 DEBUG eruption                               > Entering the main loop now...
 INFO  eruption                               > Switching to profile: turbulence.profile
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "animation_delay", description: "Step animation every [n] frames", default: 1.0 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "time_scale", description: "Scale of time axis (3rd coordinate of the noise function)", default: 3.5 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "coord_scale", description: "Scale of coordinates (1st and 2nd coordinates of the noise function)", default: 0.25 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "color_saturation", description: "Saturation in the range 0.0 .. 1.0", default: 1.0 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "color_lightness", description: "Lightness in the range 0.0 .. 0.5; above 0.5 == whiten color", default: 0.5 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "color_divisor", description: "Divide resulting hue angle by color_divisor. Final value will be mapped to the interval [0..360)", default: 1.0 }
[... cut ...]
 ERROR eruption                               > Could not send the LED map to the device: Device not initialized
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 28
^C DEBUG eruption                               > Left the main loop
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 ERROR eruption                               > Could not finalize LEDs configuration: Device not initialized
 DEBUG eruption                               > Saving state...
 INFO  eruption                               > Exiting now

DirkSchueler, 2021-02-15:
Any progress on this topic? Do I need to provide more information?

X3n0m0rph59, 2021-02-15:
@DirkSchueler Sorry for taking so long to reply! Had been busy preparing a new release.

Since I don't currently have access to a `ROCCAT Vulcan Pro` (non-TKL variant), and therefore am unable to test, could you please provide me with Wireshark traces of the USB bus traffic generated by ROCCAT Swarm?

There is at least a chance to get this working, even without me having access to the keyboard model.

I have written a short [HowTo](https://github.com/X3n0m0rph59/eruption/wiki) on how to capture traffic on the USB bus using Wireshark.

Please get back to me, if you should need help!

DirkSchueler, 2021-02-18:
Sorry I do not have the time to investigate on this  topic.
Thank you for your support.
I will return my keyboard back to Roccat and switch to a brand that supports Linux

DirkSchueler, 2021-03-01:
Back in action. Kept my Roccat Vulcan Pro.
Tried to install software from source.
Last line "cargo build --all --release" 
delivered :

`   Compiling eruption-gui v0.0.6 (/home/dirk/data/eruption/eruption-gui)
   Compiling mlua v0.5.1
   Compiling typenum v1.12.0
   Compiling radium v0.6.2
error: failed to run custom build command for `mlua v0.5.1`

Caused by:
  process didn't exit successfully: `/home/dirk/data/eruption/target/release/build/mlua-6c14542291e66848/build-script-main` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=LUA_INC
  cargo:rerun-if-env-changed=LUA_LIB
  cargo:rerun-if-env-changed=LUA_LIB_NAME
  cargo:rerun-if-env-changed=LUA_LINK
  cargo:rerun-if-env-changed=LUAJIT_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LUAJIT_STATIC
  cargo:rerun-if-env-changed=LUAJIT_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Failure { command: "\"pkg-config\" \"--libs\" \"--cflags\" \"luajit\" \"luajit >= 2.0.5\"", output: Output { status: ExitStatus(ExitStatus(256)), stdout: "", stderr: "Package luajit was not found in the pkg-config search path.\nPerhaps you should add the directory containing `luajit.pc\'\nto the PKG_CONFIG_PATH environment variable\nNo package \'luajit\' found\nPackage luajit was not found in the pkg-config search path.\nPerhaps you should add the directory containing `luajit.pc\'\nto the PKG_CONFIG_PATH environment variable\nNo package \'luajit\' found\n" } }', /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/mlua-0.5.1/build/find_normal.rs:100:13
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
`

X3n0m0rph59, 2021-03-01:
@DirkSchueler to build from source on Ubuntu you need to install the following dev-dependencies:

```sh
$ sudo apt install libusb-1.0-0-dev libhidapi-dev libevdev-dev libudev-dev libdbus-1-dev \
 libpulse-dev luajit libluajit-5.1-dev libx11-dev libxrandr-dev libgtk-3-dev libgdk-pixbuf2.0-dev \
 libatk1.0-dev libpango1.0-dev libcairo2-dev libgtksourceview-3.0-dev 
```

DirkSchueler, 2021-03-01:
Much better! 
But not successful, due to 

`   Compiling dbus-tokio v0.7.3
error[E0277]: the trait bound `i32: std::os::unix::io::AsRawFd` is not satisfied
   --> /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/dbus-tokio-0.7.3/src/connection.rs:59:16
    |
59  |     Registered(AsyncFd<RawFd>),
    |                ^^^^^^^^^^^^^^ the trait `std::os::unix::io::AsRawFd` is not implemented for `i32`
    | 
   ::: /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.2.0/src/io/async_fd.rs:137:23
    |
137 | pub struct AsyncFd<T: AsRawFd> {
    |                       ------- required by this bound in `tokio::io::unix::AsyncFd`

error[E0277]: the trait bound `i32: std::os::unix::io::AsRawFd` is not satisfied
   --> /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/dbus-tokio-0.7.3/src/connection.rs:189:34
    |
189 |     poll_ready: impl FnOnce() -> task::Poll<std::io::Result<AsyncFdReadyGuard<'a, RawFd>>>,
    |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `std::os::unix::io::AsRawFd` is not implemented for `i32`
    | 
   ::: /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.2.0/src/io/async_fd.rs:148:37
    |
148 | pub struct AsyncFdReadyGuard<'a, T: AsRawFd> {
    |                                     ------- required by this bound in `tokio::io::unix::AsyncFdReadyGuard`

error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0277`.
error: could not compile `dbus-tokio`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
`

X3n0m0rph59, 2021-03-02:
What version of rustc are you using? Looks like the packaged one from Ubuntu. That would be the old stable release, version `1.47`.

```sh
$ rustc -V
```

To build Eruption from source, you need to install the current stable rust branch preferably using [rustup.rs](https://rustup.rs/).

```sh
$ rustc -V
rustc 1.50.0 (cb75ad5db 2021-02-10)
```

DirkSchueler, 2021-03-02:
Thanks for your help.
Now compilation went fine.
` Compiling eruptionctl v0.0.2 (/home/dirk/data/eruption/eruptionctl)
   Compiling eruption-util v0.0.1 (/home/dirk/data/eruption/eruption-util)
   Compiling eruption-debug-tool v0.0.8 (/home/dirk/data/eruption/eruption-debug-tool)
   Compiling eruption-process-monitor v0.0.4 (/home/dirk/data/eruption/eruption-process-monitor)
   Compiling noise v0.7.0
   Compiling eruption-netfx v0.0.3 (/home/dirk/data/eruption/eruption-netfx)
   Compiling eruption v0.1.20 (/home/dirk/data/eruption/eruption)
    Finished release [optimized] target(s) in 12m 58s
`

But how do I install it?

X3n0m0rph59, 2021-03-02:
@DirkSchueler you can run Eruption from the build directory like this:

to run the debug build:
```sh
$ sudo target/debug/eruption -c support/config/eruption.conf
```
or to run the optimized build:
```sh
$ sudo target/release/eruption -c support/config/eruption.conf
```
The optimized build makes different assumptions about search paths of scripts and profiles.

A list of all required files for an Ubuntu based installation can be found in `support/pkg/debian/install`.

DirkSchueler, 2021-03-02:
`$ sudo target/debug/eruption -c support/config/eruption.conf`
did not work, because target/debug/eruption does not exist.

But

`$ sudo target/release/eruption -c support/config/eruption.conf` led to:

 Eruption is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 Eruption is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.20
 INFO  eruption > Loading saved state...
 WARN  eruption > Could not parse state file: Could not load global runtime state: configuration file "/var/lib/eruption/eruption.state" not found
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 WARN  eruption                       > Could not load persisted state: No such file or directory (os error 2)
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x30f7 (0002:0003:00) - ROCCAT ROCCAT Vulcan Pro
 INFO  eruption::hwdevices::roccat_vulcan_pro > Bound driver: ROCCAT Vulcan Pro
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: <unknown>
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 INFO  eruption                               > Switching to profile: profile1.profile
 ERROR eruption                               > Could not register directory watch: No such file or directory (os error 2)
 ERROR eruption                               > Could not register directory watch: No such file or directory (os error 2)
 ERROR eruption                               > Could not open profile file for reading
 INFO  eruption                               > Now listening on keyboard: /dev/input/event10
 INFO  eruption                               > Input device name: "ROCCAT ROCCAT Vulcan Pro"
 INFO  eruption                               > Input device ID: bus 0x3 vendor 0x1e7d product 0x30f7
 INFO  eruption                               > Physical location: usb-0000:00:14.0-4/input0
 INFO  eruption                               > Grabbing the keyboard device exclusively
 WARN  eruption                               > Timed out while waiting for a Lua VM to shut down, terminating now
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 ERROR eruption                               > Could not write persisted state: No such file or directory (os error 2)
 ERROR eruption                               > Could not save runtime state: No such file or directory (os error 2)
 INFO  eruption                               > Exiting now

Afterwards all key led went out.

DirkSchueler, 2021-03-02:
Calling `support/pkg/debian/install`
=>
Failed to copy 'support/systemd/eruption-suspend.sh': No such file or directory at /usr/share/dh-exec/dh-exec-install-rename line 51, <> line 8.
target/release/eruption usr/bin
target/release/eruptionctl usr/bin
target/release/eruption-netfx usr/bin
target/release/eruption-util usr/bin
target/release/eruption-debug-tool usr/bin
target/release/eruption-process-monitor usr/bin
target/release/eruption-gui usr/bin

X3n0m0rph59, 2021-03-07:
@DirkSchueler I have written up some more documentation on how to build and install Eruption from source. Please find it here: [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/master/INSTALL.md)

DirkSchueler, 2021-03-07:
Hi @X3n0m0rph59 ! 
Thanks for providing [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/master/INSTALL.md) 
That worked like a charm!
But...I cannot use the "Eruption Profile Switcher" anymore as it does not react to any mouse or keyboard events.

X3n0m0rph59, 2021-03-07:
@DirkSchueler, great and thanks for testing!

> I cannot use the "Eruption Profile Switcher" anymore as it does not react to any mouse or keyboard events.

I have just committed a fix for this issue to the `master` branch.

> Not working: Use the FN key to access special function keys (F5-F8)

You can enable this functionality by setting `ENABLE_FUNCTION_KEYS = true` in the file `eruption/src/scripts/lib/hwdevices/keyboards/roccat_vulcan_pro.lua`

X3n0m0rph59, 2021-03-07:
@DirkSchueler 

> Not working: Use the FN key to access special function keys (F5-F8)

I have committed a fix for this to the `master` branch as well!

DirkSchueler, 2021-03-07:
@X3n0m0rph59 : Pull latest verison & Rebuild all from source: 
1. Working: Use the FN key to access special function keys (F5-F8)
2. Working: "Eruption Profile Switcher"
3. Not Working: Mute key
4. Not working: Keyboard Shortcut to launch Terminal was set to CTRL-ALT-T using Ubuntu settings dialog previously. It does not work anmore. If I try to assign the shortcut again it is displayed as CTRL-ALT-LAUNCH7 whenver press CRTL-ALT-T

DirkSchueler, 2021-03-07:
@

DirkSchueler, 2021-03-07:
Sorry closed by mistake