eruption-project/eruption #42: Add per-user config directory

RoggeHaj, 2021-02-04:
I think it would be beneficial if eruption had support for a local user config directory, `${HOME}/.config/eruption`, or similar that would allow overriding the central scripts and config files as well as adding new scripts without having to modify the central files.

RoggeHaj, 2021-02-04:
*sigh* I really need to improve my reading skills.

closed 2021-02-04 by RoggeHaj

X3n0m0rph59, 2021-02-04:
@RoggeHaj, thanks for your feature request! This has been requested multiple times, and I too think that it would be really nice to have per-user config directories. This feature mainly does not exist at the moment, because of the security implications this has.

Please see issue #27