eruption-project/eruption #43: Support for Roccat Khan Aimo

Slypth, 2021-02-13:
Hi there, I have always wanted to be able to use my Khan Aimo headset on Ubuntu, but there has never been support for it. I would love to be able to get it working. It would be really useful for me, and would save me a few. If I can help in any way, please let me know. 
- Thanks.

X3n0m0rph59, 2021-02-13:
Hi @Slypth! Thank you for your feature request!

I am interested in adding support for this device to Eruption. What features do you have in mind specifically? Do you want support just for the AIMO LED lighting, or does the headset have any other features that require special support?
(I currently don't have access to this device, so I can't tell what already works on Linux and what does not)

To get the AIMO LEDs supported by Eruption, I would need a Wireshark capture of the USB bus traffic during initialization of the LEDs as done by ROCCAT Swarm.

There is a short HowTo available [here](https://github.com/X3n0m0rph59/eruption/wiki) on how to do this.
It would be great to get this supported! Please get back to me if you need further assistance.

Slypth, 2021-02-16:
Hi there, sorry for taking so long to get back to you.

I'm really glad to hear that you're interested in adding support for the device. 
Having support for the AIMO LED lighting would be great, but the device does have some more features. I realize now that this would probably be a lot harder than I expected. One of the features emulates 5.1 surround sound. While the emulation isn't amazing, it can be a fun feature to have. It honestly wouldn't be a game-changer to me if support for this isn't possible because I find myself barely using this feature. 

I also realized that a lot of the configuration for the device is in ROCCAT's Swarm application. Features like equalizers, mic sensitivity, voice changers, and some other features too. To be fair, most of them I don't care about. I would be really excited if it was somehow possible to be able to have different profiles with equalizer settings. What was great about the Swarm application was that you could have a specific profile set with custom equalizer settings for certain apps. That was definitely a great feature.

I realize now that most of this is basically just recreating ROCCAT's Swarm. That is incredibly hard to do. So, if I could even just have a cli version of it where I'm able to just control even some basic elements of the device, that would be amazing.

I will look into a Wireshark capture of the device. Thank you for your interest.