eruption-project/eruption #45: Macros not working on Vulcan TKL

Madcliff, 2021-02-18:
The very well featured and flexible macro system is not working for the Vulcan TKL. I suspect there is a clash between certain shortcuts for setting up the macro layers, as this uses the home keys in the Vulcan and on the Vulcan TKL several of these keys are used to enable different functionality in the keyboard. 

Other possibility is that the onboard functionality is being enabled when game mode is turned on. In fact that appears to be likely as I cannot enable the super key while game mode is on.

X3n0m0rph59, 2021-02-18:
@Madcliff Thanks for reporting this issue!

I have verified that the macro system is **working** on ROCCAT Vulcan 1xx and ROCCAT Vulcan Pro TKL. So there has to be a bug with key indices in the following files:

`/usr/share/eruption/scripts/lib/hwdevices/keyboards/roccat_vulcan_tkl.lua`
`/usr/share/eruption/scripts/lib/macros/modifiers.lua`

The TKL is lacking macro keys `M1-M6`, that the standard form factor keyboards have. So for the TKL variants we set:
```Lua
ENABLE_MACRO_KEYS 	 = false	-- ROCCAT Vulcan TKL variant does not have any macro keys
```
_This does not disable the macro system._

Could you please determine whether these key-codes are correct? I just used the ones from the ROCCAT Vulcan **Pro** TKL
```Lua
-- HID key codes
GAME_MODE_KEY = 106
EASY_SHIFT_KEY = 21
```

You could try these first  (taken from ROCCAT Vulcan 1xx):
```Lua
-- HID key codes
GAME_MODE_KEY = 96
EASY_SHIFT_KEY = 255

-- HID key codes
GAME_MODE_KEY = 96
EASY_SHIFT_KEY = 255

-- character to key index mapping
key_to_index = {}

key_to_index['GAME_MODE'] = 104

key_to_index['FN'] = 77
key_to_index['CAPS_LOCK'] = 4

-- ......
```

You may want to run Eruption in debug mode:

```sh
 $ sudo RUST_LOG=debug eruption
```

On my keyboard this output gets printed when I press `CAPS LOCK`:
```log
 DEBUG eruption::scripting::script::callbacks      > Macros: HID event: 2 args: 21
 DEBUG eruption::scripting::script::callbacks      > Macros: Key down: Index: 6
 DEBUG eruption::scripting::script::callbacks      > Macros: HID event: 1 args: 21
 DEBUG eruption::scripting::script::callbacks      > Macros: Key up: Index: 6
```

So the HID code of `CAPS LOCK` == 21
And the KEY_INDEX of `CAPS LOCK` == 6

```Lua
-- HID key codes
EASY_SHIFT_KEY = 21
-- ....
key_to_index['CAPS_LOCK'] = 6
```

The `FN` key on my keyboard produces:
```Log
 DEBUG eruption::scripting::script::callbacks      > Macros: HID event: 2 args: 119
 DEBUG eruption::scripting::script::callbacks      > Macros: Key down: Index: 65
 DEBUG eruption::scripting::script::callbacks      > Macros: HID event: 1 args: 119
 DEBUG eruption::scripting::script::callbacks      > Macros: Key up: Index: 65
```

And in the file `/usr/share/eruption/scripts/lib/macros/modifiers.lua` should look like this:

```Lua
-- This file is part of Eruption.

-- Eruption is free software: you can redistribute it and/or modify
-- it under the terms of the GNU General Public License as published by
-- the Free Software Foundation, either version 3 of the License, or
-- (at your option) any later version.

-- Eruption is distributed in the hope that it will be useful,
-- but WITHOUT ANY WARRANTY without even the implied warranty of
-- MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
-- GNU General Public License for more details.

-- You should have received a copy of the GNU General Public License
-- along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

ENABLE_EASY_SHIFT = true   -- set this to false if you don't want to
						   -- use the Easy Shift+ functionality

ENABLE_SUPER_KEY_IN_GAME_MODE = false -- set this to true to enable the Windows key even when in game mode

-- comment out the declarations below to change the modifier key you want to use; default is the "FN" key:

MODIFIER_KEY = FN		       				   --
MODIFIER_KEY_INDEX   = key_to_index['FN'] 	   -- the KEY_INDEX of the modifier key; has to match the key defined above
MODIFIER_KEY_EV_CODE = 464     				   -- the EV_KEY code of the modifier key; has to match the key defined above

-- or use this if you prefer "Right Menu" as the modifier key:

-- MODIFIER_KEY = RIGHT_MENU   				   --
-- MODIFIER_KEY_INDEX   = key_to_index['RIGHT_MENU']  -- the KEY_INDEX of the modifier key; has to match the key defined above
-- MODIFIER_KEY_EV_CODE = 127  				   -- the EV_KEY code of the modifier key; has to match the key defined above

-- or use this if you prefer "Right Alt" as the modifier key:

-- MODIFIER_KEY = RIGHT_ALT	   				   --
-- MODIFIER_KEY_INDEX   = key_to_index['RIGHT_ALT']  -- the KEY_INDEX of the modifier key; has to match the key defined above
-- MODIFIER_KEY_EV_CODE = 100  				   -- the EV_KEY code of the modifier key; has to match the key defined above

-- or use this if you prefer "Right Shift" as the modifier key:

-- MODIFIER_KEY = RIGHT_SHIFT  				   --
-- MODIFIER_KEY_INDEX   = key_to_index['RIGHT_SHIFT']  -- the KEY_INDEX of the modifier key; has to match the key defined above
-- MODIFIER_KEY_EV_CODE = 54   				   -- the EV_KEY code of the modifier key; has to match the key defined above

-- or use this if you prefer "Right Ctrl" as the modifier key:

-- MODIFIER_KEY = RIGHT_CTRL   				   --
-- MODIFIER_KEY_INDEX   = key_to_index['RIGHT_CTRL']  -- the KEY_INDEX of the modifier key; has to match the key defined above
-- MODIFIER_KEY_EV_CODE = 97   				   -- the EV_KEY code of the modifier key; has to match the key defined above

```

Madcliff, 2021-03-14:
My FN key has the following outputs on the TKL 

 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 1 args: 119
 DEBUG eruption::scripting::script::callbacks > Macros: Key up: Index: 77

It feels like that when I press the FN key + PGDOWN the hardware functionality takes over rather than registering a key input. As all that registers is a HID event.

X3n0m0rph59, 2021-03-14:
@Madcliff Could you please show the outputs (HID event and index) when pressing the following keys:

* `PGDOWN` 
* `CAPS LOCK`

Madcliff, 2021-03-14:
pgdown  
DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 111
DEBUG eruption::scripting::script::callbacks > Macros: HID event: 2 args: 106
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 1 args: 106
 DEBUG eruption::scripting::script::callbacks > Macros: Key up: Index: 111

Capslock

Madcliff, 2021-03-14:
Just a question, how do we change shift layers on the TKL variants as we do not have the macro keys

X3n0m0rph59, 2021-03-14:
@Madcliff Thanks for posting the logs!

> Just a question, how do we change shift layers on the TKL variants as we do not have the macro keys

The EasyShift layers can be selected just like on the non-TKL variants, by entering GameMode and then pressing `CAPS LOCK` and `INSERT` - `PGDOWN`.

Madcliff, 2021-03-14:
Ah then yes, the game mode is not being entered on the TKL. It is
defaulting to the Hardware level Gamemode..

X3n0m0rph59, 2021-03-14:
@Madcliff I have updated the key definitions of the ROCCAT Vulcan TKL.

Could you please verify, that you can now switch to GameMode and select EasyShift layers?

Madcliff, 2021-03-14:
Yeah this is still not working at all. I am wondering if the hardware is just redirecting the input, or if in the alternative, the macro layer is just not enabling itself.

X3n0m0rph59, 2021-03-15:
@Madcliff I have just pushed out another update to the `master` branch. I hope that it will work now...

> Yeah this is still not working at all. I am wondering if the hardware is just redirecting the input, or if in the alternative, the macro layer is just not enabling itself.

I suspect the latter. It is expected for the hardware to switch states when enabling GameMode. Eruption just has to keep track of this state change as well. When in GameMode, the hardware should behave somewhat differently compared to the default mode.

I have updated the driver for the ROCCAT Vulcan TKL, so that it closely resembles the driver of the ROCCAT Vulcan.

Could you please build a new package using the latest `PKGBUILD`, then install it, and then please verify that everything is working?

Madcliff, 2021-03-15:
No joy. Still not working. Maybe time to put some debugging messages into
the game-mode function.

X3n0m0rph59, 2021-03-15:
@Madcliff Sounds reasonable! I have added, some more debug messages to `macros.lua`.

The following messages should get printed:

```lua
debug("Macros: FN key event registered")
debug("Macros: Easy Shift+ key event registered")
```

```lua
debug("Macros: Game mode toggled")
```

Please tell me, if you can find one of these messages in the log output.

Madcliff, 2021-03-15:
Which logs exactly?

X3n0m0rph59, 2021-03-19:
@Madcliff I have updated the key definitions in the `master` branch! Please try again to build from source!

> Which logs exactly?

Please post the full log of `sudo journalctl -u eruption.service -b`

Madcliff, 2021-03-21:
Mar 21 09:11:41 archlinux eruption[1067]:  ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Mar 21 09:11:41 archlinux eruption[1067]:  ERROR eruption::plugins::audio               > Could not start the audio grabber: Pulse Audio error: Could not open Pulse Audio: Connection refused
Mar 21 09:11:44 archlinux eruption[1067]:  INFO  eruption                               > Switching to profile: zz-matrix.profile
Mar 21 09:11:44 archlinux eruption[1067]:  INFO  eruption                               > Loading Lua script: 'solid.lua'
Mar 21 09:11:44 archlinux eruption[1067]:  INFO  eruption                               > Loading Lua script: 'ghost.lua'
Mar 21 09:11:44 archlinux eruption[1067]:  INFO  eruption                               > Loading Lua script: 'wave.lua'
Mar 21 09:11:44 archlinux eruption[1067]:  INFO  eruption                               > Loading Lua script: 'afterhue.lua'
Mar 21 09:11:44 archlinux eruption[1067]:  INFO  eruption                               > Loading Lua script: 'macros.lua'

Madcliff, 2021-03-21:
This is what occurs after updating, switching to a profile with macros, than pressing the FN key, enabling game mode and trying to use the easy-shift key. It appears that none of those events are being properly registered.

X3n0m0rph59, 2021-03-21:
@Madcliff I don't see any debug level messages in the log, is the `RUST_LOG=debug` environment variable set?

These are debug level messages, they won't be shown on non-debug logs:
```lua
debug("Macros: FN key event registered")
debug("Macros: Easy Shift+ key event registered")
debug("Macros: Game mode toggled")
```

To enable the debug messages, you can start the Eruption daemon like this:
```sh
 $ sudo RUST_LOG=debug eruption
```

and if you want to run from the compiled sources, then run it like this:
```sh
 $ sudo RUST_LOG=debug target/debug/eruption -c support/config/eruption-dev.conf
```

Then please try to enter game mode...

If possible, please post the full output of:
```sh
 $ sudo journalctl -u eruption.service -b
```

With the full log output, I will be able to see the USB HID event codes and key indices of the keys pressed, while you try to enter game mode and execute macros.

If it still does not work please check your `macros.lua` file:

```sh
 $ sha1sum /usr/share/eruption/scripts/macros.lua
 342444767433978973ccc5c861f1b20e226b1198  /usr/share/eruption/scripts/macros.lua
```

Madcliff, 2021-03-21:
INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.20 (0.1.19.r135.g4f64061) (release build)
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x48d:0xc100 - ITE Tech. Inc. ITE Device(8910)
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x1e7d:0x39d0 - ROCCAT ROCCAT Khan AIMO
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0xfd9:0x6c - Љ Љ
 DEBUG eruption::hwdevices            > Found supported HID device: 0x1e7d:0x2e7c - ROCCAT ROCCAT Nyth
 INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2e7c (0005:000c:00) - ROCCAT ROCCAT Nyth
 ERROR eruption::hwdevices            > Failed to bind the device driver
 DEBUG eruption::hwdevices            > Found supported HID device: 0x1e7d:0x2e7c - ROCCAT ROCCAT Nyth
 INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2e7c (0005:000c:01) - ROCCAT ROCCAT Nyth
 ERROR eruption::hwdevices            > Failed to bind the device driver
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x1532:0x511 - R Razer 
 DEBUG eruption::hwdevices            > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x2fee (0005:0008:00) - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices::roccat_vulcan_tkl > Bound driver: ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: 115
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 DEBUG eruption                               > Registering filesystem watcher...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 DEBUG eruption                               > Entering the main loop now...
 INFO  eruption                               > Switching to profile: zz-matrix.profile
 INFO  eruption                               > Loading Lua script: 'solid.lua'
 INFO  eruption                               > Loading Lua script: 'ghost.lua'
 INFO  eruption                               > Loading Lua script: 'wave.lua'
 INFO  eruption                               > Loading Lua script: 'afterhue.lua'
 INFO  eruption                               > Loading Lua script: 'macros.lua'
 DEBUG eruption::scripting::script            > Applying parameter Color { name: "color_background", description: "Background color", min: None, max: None, default: 4294905872 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "opacity", description: "Opacity value in the range 0.0 .. 1.0, where 1.0 is fully opaque", min: Some(0.0), max: Some(1.0), default: 1.0 }
 DEBUG eruption                               > Switch successful
 DEBUG eruption::scripting::script::callbacks > Loading device specific Lua script: 'keyboards/roccat_vulcan_1xx.lua'
 DEBUG eruption::scripting::script            > Applying parameter Int { name: "ghost_backoff_secs", description: "Time (in seconds) that has to pass, before ghost typing commences", min: Some(0), max: Some(120), default: 10 }
[... cut ...]
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 28
^C DEBUG eruption                               > Left the main loop
 INFO  eruption                               > Shutting down all Lua VMs now...
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 INFO  eruption                               > Closing all devices now...
 DEBUG eruption                               > Saving state...
 INFO  eruption                               > Exiting now

The debug errors are not appearing in the journal at all.

Madcliff, 2021-03-21:
So if I run eruption in the debug mode the journal will not record any of the events.