eruption-project/eruption #47: runtime error in modifiers.lua: attemtp to index global 'key_to_index' (a nil value) on Pop 20.04

greenseeker, 2021-02-19:
On first launch of the service, it fails with:
```
Feb 19 18:31:53 hostname systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Feb 19 18:31:53 hostname systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption > Loading saved state...
Feb 19 18:31:53 hostname eruption[384410]:  WARN  eruption > Could not parse state file: Could not load global runtime state: configuration file "/var/lib/eruption/eruption.state" not found
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption > Registering plugins...
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Feb 19 18:31:53 hostname eruption[384410]:  WARN  eruption                       > Could not load persisted state: No such file or directory (os error 2)
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption                       > Plugins loaded and initialized successfully
Feb 19 18:31:53 hostname eruption[384410]:  INFO  eruption                       > Enumerating connected devices...
Feb 19 18:31:54 hostname eruption[384410]:  INFO  eruption                       > Device enumeration completed
Feb 19 18:31:54 hostname eruption[384410]:  INFO  eruption                       > Performing late initializations...
Feb 19 18:31:54 hostname eruption[384410]:  INFO  eruption                       > Initializing D-Bus API...
Feb 19 18:31:54 hostname eruption[384410]:  INFO  eruption                       > Late initializations completed
Feb 19 18:31:54 hostname eruption[384410]:  INFO  eruption                       > Startup completed
Feb 19 18:31:54 hostname eruption[384410]:  INFO  eruption                       > Switching to profile: profile1.profile
Feb 19 18:31:54 hostname eruption[384410]:  ERROR eruption::scripting::script    > Lua error in file /usr/share/eruption/scripts/macros.lua: runtime error: /usr/share/eruption/scripts/lib/macros/modifiers.lua:25: attempt to index global 'key_to_index' (a nil value)
Feb 19 18:31:54 hostname eruption[384410]: stack traceback:
Feb 19 18:31:54 hostname eruption[384410]:         ok: in function '__index'
Feb 19 18:31:54 hostname eruption[384410]:         ok:25: in main chunk
Feb 19 18:31:54 hostname eruption[384410]:         ok: in function 'require'
Feb 19 18:31:54 hostname eruption[384410]:         ok:32: in main chunk
Feb 19 18:31:54 hostname eruption[384410]:         UnknownError
Feb 19 18:31:54 hostname eruption[384410]:  ERROR eruption                       > Script execution failed
Feb 19 18:31:54 hostname systemd[1]: eruption.service: Main process exited, code=killed, status=6/ABRT
Feb 19 18:31:54 hostname systemd[1]: eruption.service: Failed with result 'signal'.
```

X3n0m0rph59, 2021-02-20:
Hi @greenseeker, thank you for submitting this bug report!

It seems that Eruption is unable to find any supported devices. Starting up with no devices at all seems to trigger this bug.
Could you please tell me what devices you have connected to your system?

```sh
 $ sudo lsusb
```

greenseeker, 2021-02-20:
Hi @X3n0m0rph59,

It's the Kone AIMO Remastered:
`Bus 003 Device 002: ID 1e7d:2e2c ROCCAT ROCCAT Kone Aimo 16K`

If not supported, please let me know if I can provide anything helpful. I am not a developer but I am a Linux engineer.

X3n0m0rph59, 2021-02-22:
@greenseeker The ROCCAT Kone Aimo **16K** is currently not supported by Eruption. To add support for it, I would need a trace of the packets sent over the USB bus to the device, during the device initialization sequence done by ROCCAT Swarm.

This can be achieved using the `usbmon` Linux kernel module as well as Wireshark, capturing the USB traffic originating on a Windows 10 VM that is running ROCCAT Swarm.

I wrote a short article in the [Eruption Wiki](https://github.com/X3n0m0rph59/eruption/wiki) on how to do this.

It would be really nice if you could find the time!

TheMysteriousOH, 2021-02-27:
Hi @X3n0m0rph59, 
I have the same Issue with the **Roccat Vulcan Pro**.
My System is Arch Linux.
Attached you find the journal log.
[eruption.log](https://github.com/X3n0m0rph59/eruption/files/6053976/eruption.log)

TheMysteriousOH, 2021-02-27:
Here is a ROCCAT Swarm Capture for the ROCCAT VULCAN Pro
[ROCCAT_SWARM_Capture.pcapng.zip](https://github.com/X3n0m0rph59/eruption/files/6054357/ROCCAT_SWARM_Capture.pcapng.zip)

greenseeker, 2021-02-28:
[kone_aimo_remastered.pcapng.gz](https://github.com/X3n0m0rph59/eruption/files/6057876/kone_aimo_remastered.pcapng.gz)
Let me know if I've done this incorrectly or if you need anything else.

cadenhenrich, 2021-03-07:
I'm having the same issue, but I'm using the ROCCAT Kone Pure Ultra, which is listed in the supported devices for `eruption`. This *might* be because I'm on Artix, and therefore no `systemd`, but usually I can just add the executable to the `default` run level using `rc-upate` if I copy it to `/etc/init.d/`, so I actually don't think a lack of `systemd` is the issue here.

X3n0m0rph59, 2021-03-08:
@cadenhenrich I set up an Artix virtual machine using the `OpenRC` init system and investigated the issue, but was so far unable to reproduce. It seems that Eruption is able to build and run on non-`systemd` distros.

> It seems that Eruption is unable to find any supported devices. Starting up with no devices at all seems to trigger this bug.

Maybe the Eruption daemon is somehow failing to enumerate the devices on your system. To further investigate this issue, could you please provide some logs of Eruption starting up?
```sh
 $ sudo RUST_LOG=debug eruption 
```

or, if you built from source

```sh
 $ sudo RUST_LOG=debug target/debug/eruption -c support/config/eruption-dev.conf
```

X3n0m0rph59, 2021-03-08:
@cadenhenrich I can reproduce it now! Starting Eruption with only a supported mouse device connected, and no keyboard, will trigger this bug. I am working on a fix for this now...

X3n0m0rph59, 2021-03-08:
@cadenhenrich Thanks for bringing this to my attention. I believe that this problem is fixed now in the `master` branch!

The Eruption GUI version `0.0.7` (tech preview) will not work, as it currently depends on a connected keyboard.

cadenhenrich, 2021-03-09:
Seems to be fixed, although the LED keeps going from solid blue to breathing rainbow and back again. Now to figure out how to turn on the zero-debounce option!

X3n0m0rph59, 2021-03-09:
@cadenhenrich, great!

I have looked into the zero-debounce configuration and done some Wireshark captures. I got it working on a `ROCCAT Kone Pure Ultra`!

On a `ROCCAT Kone Pure Ultra` mouse, this is how to enable zero-debounce:

- Find mouse device index like this:
```sh
 $ sudo eruption-debug-tool list

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 

Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other sub-commands of this tool

Index: 00: ID: 46d:c051 Logitech/USB-PS/2 Optical Mouse Subdev: 0
Index: 01: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 0
Index: 02: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 1
Index: 03: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 2
Index: 04: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 3
Index: 05: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 0
Index: 06: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 1
Index: 07: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 2
Index: 08: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 3

Enumeration completed
```

So the device index on my system would be `1`.

- We need to write to the device with index `1`, as determined previously
- Byte 3 of the data structure represents the zero-debounce configuration option (on a `ROCCAT Kone Pure Ultra`)

So, to turn on zero-debounce, I would need to run this command:
```sh
 $ sudo eruption-debug-tool write 1 '[0x11, 0x0e, 0x01, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x4c, 0x00, 0x05, 0x05, 0x00, 0x08, 0x00, 0x10, 0x00, 0x18, 0x00, 0x20, 0x00, 0x40, 0x00, 0x00, 0x00, 0x03, 0x09, 0x06, 0xff, 0x0f, 0x00, 0x00, 0x14, 0xff, 0xff, 0x00, 0x00, 0x14, 0xff, 0x00, 0x48, 0xff, 0x14, 0xff, 0x00, 0x48, 0xff, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xf7, 0x09, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00]'
```

And to turn off the zero-debounce option, I run this command:
```sh
 $ sudo eruption-debug-tool write 1 '[0x11, 0x0e, 0x00, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x4c, 0x00, 0x05, 0x05, 0x00, 0x08, 0x00, 0x10, 0x00, 0x18, 0x00, 0x20, 0x00, 0x40, 0x00, 0x00, 0x00, 0x03, 0x09, 0x06, 0xff, 0x0f, 0x00, 0x00, 0x14, 0xff, 0xff, 0x00, 0x00, 0x14, 0xff, 0x00, 0x48, 0xff, 0x14, 0xff, 0x00, 0x48, 0xff, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xf7, 0x09, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00]'
```

To read out the state from the device:

```sh
 $ sudo eruption-debug-tool read 1 0x11 78
```

## In case your mouse device needs a different byte sequence

* Save a state snapshot of your mouse using:
```sh
 $ sudo eruption-debug-tool state-diff 1
```
* Fire up ROCCAT Swarm (e.g. in a VM), change a single setting (zero-debounce)
* Again, save a state snapshot of your mouse using:
```sh
 $ sudo eruption-debug-tool state-diff 1
```

On the second invocation of `state-diff` you will get a diff view of the changed bytes.

```sh
 $ sudo eruption-debug-tool state-diff 1

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 
Index: 01: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 0
Reading data from device...
The following USB HID report IDs have changed bytes:

Changed bytes: [116]
0x05: [0x05, 0x80, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00, 0x03, 0x00, 0xef, 0xe7, 0x01, 0x00, 0x00, 0x00, 0x90, 0x93, 0x00, 0x00, 0x94, 0x93, 0x00, 0x00, 0xa4, 0x93, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x09, 0x01, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x20, 0x00, 0x00, 0x00, 0x23, 0x00, 0x00, 0x00, 0x22, 0x40, 0x78, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0xff, 0x00, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00=>0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x38, 0x00, ]

Changed bytes: [116]
0x04: [0x04, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0x00, 0xef, 0xe7, 0x01, 0x00, 0x00, 0x00, 0x90, 0x93, 0x00, 0x00, 0x94, 0x93, 0x00, 0x00, 0xa4, 0x93, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x09, 0x01, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x20, 0x00, 0x00, 0x00, 0x23, 0x00, 0x00, 0x00, 0x22, 0x40, 0x78, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0xff, 0x00, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00=>0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x38, 0x00, ]

Changed bytes: [2, 12]
[... cut ...] 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, ]

Saving state data...
Done
```

cadenhenrich, 2021-03-09:
Oh wow, thanks for all the help! I was just goingto figure it out on my own, so thanks for saving me a lot of work!