eruption-project/eruption #48: Add Support for ROCCAT Vulcan Pro

X3n0m0rph59, 2021-02-28:
This issue will track the progress on the support for the ROCCAT Vulcan Pro series keyboards.

_Originally posted by @TheMysteriousOH in https://github.com/X3n0m0rph59/eruption/issues/47#issuecomment-787037147_

X3n0m0rph59, 2021-02-28:
@TheMysteriousOH, thank you very much for providing the USB capture file! I have updated the codebase respectively. Since I currently do not have access to a ROCCAT Vulcan Pro, this is entirely untested.

I pushed out the latest changes to the `master` branch. You may want to build it using the `PKGBUILD` from `support/pkg/arch/PKGBUILD`.

Running `eruption-debug-tool` on your device should give you a red/blue test pattern.

```sh
$ sudo eruption-debug-tool list 

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 

Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other sub-commands of this tool

Index: 00: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 0
Index: 01: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 1
Index: 02: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 2
Index: 03: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 3
Index: 04: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 0
Index: 05: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 1
Index: 06: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 2
Index: 07: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 3

Enumeration completed
```

Please try the subdev 1 first. On my system that would be the device with index 5:
```sh
$ sudo eruption-debug-tool run-tests 5
```

TheMysteriousOH, 2021-02-28:
@X3n0m0rph59 thank you for your work.

Herte is the result of my test:
```
$ sudo eruption-debug-tool list

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service

Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other sub-commands of this tool

Index: 00: ID: 46d:c52b Logitech/USB Receiver Subdev: 0
Index: 01: ID: 46d:c52b Logitech/USB Receiver Subdev: 1
Index: 02: ID: 46d:c52b Logitech/USB Receiver Subdev: 2
Index: 03: ID: 1e7d:30f7 ROCCAT/ROCCAT Vulcan Pro Subdev: 0
Index: 04: ID: 1e7d:30f7 ROCCAT/ROCCAT Vulcan Pro Subdev: 1
Index: 05: ID: 1e7d:30f7 ROCCAT/ROCCAT Vulcan Pro Subdev: 2
Index: 06: ID: 1e7d:30f7 ROCCAT/ROCCAT Vulcan Pro Subdev: 3
Index: 07: ID: 46d:a29 Logitech/Logitech Wireless Headset Subdev: 3

Enumeration completed
$ sudo eruption-debug-tool run-tests 4

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service

Index: 04: ID: 1e7d:30f7 ROCCAT/ROCCAT Vulcan Pro Subdev: 1
Bound driver: ROCCAT Vulcan Pro
$
```
But I did not get the red/blue test Pattern

X3n0m0rph59, 2021-02-28:
@TheMysteriousOH, thanks for taking your time to try this out! I just pushed out the latest batch of changes to the `master` branch. Again, this code is entirely untested, since I have no access to the hardware.

I believe that I fixed multiple bugs within the code of the ROCCAT Vulcan Pro driver module. Please try again with `eruption-debug-tool` revision `0.0.7`.

```sh
$ eruption-debug-tool -V
...
eruption-debug-tool 0.0.7
```

TheMysteriousOH, 2021-03-01:
@X3n0m0rph59, thank you for the reply. 
The test with the eruption-debug-tool have the same result.
No change of the keyboard lights.

```
sudo eruption-debug-tool run-tests 4

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service

Index: 04: ID: 1e7d:30f7 ROCCAT/ROCCAT Vulcan Pro Subdev: 1
Bound driver: ROCCAT Vulcan Pro
Step 1
Step 2 skipped
Step 3
Step 4
Step 5
```
In the next step I have unmask the eruption.service and unplug and replug the keyboard.
Attached you find the systemlog, maybe this will help.
[eruption.log](https://github.com/X3n0m0rph59/eruption/files/6059472/eruption.log)

Please, let me know if you need more tests.

X3n0m0rph59, 2021-03-01:
@TheMysteriousOH, thanks again for testing and for providing the log file! I just fixed two bugs that crept in. I hope that it works now...

Please be sure that you are using version `0.0.8` of the `eruption-debug-tool`.

TheMysteriousOH, 2021-03-01:
@X3n0m0rph59, good Job.
Thank you very much. It is working now.
The eruption.service is also running.

If you need further tests for release, please let me now.

X3n0m0rph59, 2021-03-01:
@TheMysteriousOH great! I am glad that the initialization of the device is working now!

What probably will not work currently:

* The `FN` key?
* Highlight of the macro keys `M1-M6` when `FN` is pressed?
* Neighbor topology maps still need to be updated, some effects will show wrong results when keys on the numpad are pressed
* Some of the parameters in `eruption/src/scripts/lib/hwdevices/keyboards/roccat_vulcan_pro.lua` still need to be updated

I am working on `eruption-util`, a companion tool to Eruption that allows a user to determine most of the required parameters and tables, like e.g. the neighbor topology maps, etc just by pressing keys that are highlighted by said utility. It will need to be refined somewhat though, to work on the ROCCAT Vulcan Pro. I am working on it...

TheMysteriousOH, 2021-03-01:
FYI:

When I press the `FN` key the following keys are highlighted:
`M1` - `M6`
`F1` - `F4`
`F9` - `F12`

X3n0m0rph59, 2021-03-01:
@TheMysteriousOH, okay at least this part seems to be working as intended!

TheMysteriousOH, 2021-03-07:
@X3n0m0rph59, Please let me know, if I should test the `eruption-tool`.
If I record the rows:
`sudo eruption-util record-topology rows 4`
At row 1 it jumps to row 2 after pressing the `page up` key, before I can press the keys on the `num pad`

X3n0m0rph59, 2021-03-07:
@TheMysteriousOH It would be really cool if you could generate new, or update the existing topology maps for the ROCCAT Vulcan Pro!

The code for `eruption-util` still contains some hard-coded values and fixed assumptions. This needs to be fixed first. I will update the `eruption-util` soon, and let you know!

TheMysteriousOH, 2021-03-07:
@X3n0m0rph59 Here is the layout of the ROCCAT Vulcan Pro (QWERTZ) german layout.
I hope this will help: The `FN` Key generate no record.

**row 0** 0x00, 0x0b, 0x11, 0x17, 0x1c, 0x30, 0x35, 0x3b, 0x41, 0x4e, 0x54, 0x55, 0x56, 0x63, 0x67, 0x6c
**row 1** 0x01, 0x06, 0x0c, 0x12, 0x18, 0x1d, 0x21, 0x31, 0x36, 0x3c, 0x42, 0x48, 0x4f, 0x57, 0x64, 0x68, 0x6d, 0x71, 0x77, 0x7c, x81
**row 2** 0x02, 0x07, 0x0d, 0x13, 0x19, 0x1e, 0x22, 0x32, 0x37, 0x3d, 0x43, 0x49, 0x50, 0x58, 0x65, 0x69, 0x6e, 0x72, 0x78, 0x7d, 0x82
**row 3** 0x03, 0x08, 0x0e, 0x14, 0x1a, 0x1f, 0x23, 0x33, 0x38, 0x3e, 0x44, 0x4a, 0x60, 0x73, 0x79, 0x7e
**row 4** 0x04, 0x09, 0x0f, 0x15, 0x1b, 0x20, 0x24, 0x34, 0x39, 0x3f, 0x45, 0x4b, 0x52, 0x6a, 0x74, 0x7a, 0x7f, 0x83
**row 5**   0x05, 0x0a, 0x10, 0x25, 0x46, 0x53, 0x59, 0x66, 0x6b, 0x6f, 0x75

TheMysteriousOH, 2021-03-08:
@X3n0m0rph59 I have seen that you updated the eruption-util

I have tested `sudo eruption-util record-topology rows 4`

And got an error on the 4th Row:
```
Please press all keys in row 4, press ESC to skip
 INFO  eruption_util > Recorded key with index 4
 INFO  eruption_util > Recorded key with index 9
 INFO  eruption_util > Recorded key with index 15
 INFO  eruption_util > Recorded key with index 21
 INFO  eruption_util > Recorded key with index 27
 INFO  eruption_util > Recorded key with index 32
 INFO  eruption_util > Recorded key with index 36
 INFO  eruption_util > Recorded key with index 52
 INFO  eruption_util > Recorded key with index 57
 INFO  eruption_util > Recorded key with index 63
 INFO  eruption_util > Recorded key with index 69
 INFO  eruption_util > Recorded key with index 75
 INFO  eruption_util > Recorded key with index 82
 INFO  eruption_util > Recorded key with index 106
 INFO  eruption_util > Recorded key with index 116
The application panicked (crashed).
Message:  index out of bounds: the len is 102 but the index is 102
Location: eruption-util/src/main.rs:800

Backtrace omitted.
Run with RUST_BACKTRACE=1 environment variable to display it.
Run with RUST_BACKTRACE=full to include source snippets.
zsh: abort      sudo eruption-util record-topology rows 4

```

X3n0m0rph59, 2021-03-08:
@TheMysteriousOH Thanks for submitting the row topology of the ROCCAT Vulcan Pro! This seems to be pretty much consistent with the configuration of the ROCCAT Vulcan 1xx:

I have updated the source code accordingly:
* `eruption/src/hwdevices/roccat_vulcan_pro.rs`
* `eruption/src/scripts/lib/hwdevices/keyboards/roccat_vulcan_pro.lua`

Row 5 is differing slightly, it seems that it is missing a key around the space bar?

TheMysteriousOH, 2021-03-08:
@X3n0m0rph59 Yes you are right, sorry.
The last Key was missing `0x80` the `,` key on the Num Block
here is the correct row 5:
`0x05, 0x0a, 0x10, 0x25, 0x46, 0x53, 0x59, 0x66, 0x6b, 0x6f, 0x75, 0x80`

X3n0m0rph59, 2021-03-08:
@TheMysteriousOH Thank you again! I have incorporated the changes into the latest push to the `master` branch.

TheMysteriousOH, 2021-03-12:
There is a Bug with the `left Control `key.
Some Key combination are no longer working, when the **eruption.service** is started.

For example:
`Control` + `d`
`Control` + `w`
`Control` + `r`
`Control` + `t`

The right Control Key ist working

If I stoped the eruption.service, the left Control key is working as expected

X3n0m0rph59, 2021-03-13:
@TheMysteriousOH I looked into this, but as I don't have the hardware available I can not try out any changes.

Could you please provide some more context? Eruption debug logs would be nice to have! Please rebuild from source and run:
```sh
 $ sudo RUST_LOG=debug target/debug/eruption -c support/config/eruption-dev.conf
```

Then press the `LEFT CTRL` key and please post the full log output!

TheMysteriousOH, 2021-03-17:
@X3n0m0rph59 thank you for take attention on this topic.
Sorry for my late reply, I have this keyboard only at work.

```
$ sudo RUST_LOG=debug target/release/eruption -c support/config/eruption-dev.conf

 Eruption is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 Eruption is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.20 (release build)
 WARN  eruption > ** EXPERIMENTAL FEATURES are ENABLED, this may expose serious bugs! **
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x46d:0xc52b - Logitech USB Receiver
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x46d:0xc52b - Logitech USB Receiver
 DEBUG eruption::hwdevices            > Found unsupported HID device: 0x46d:0xc52b - Logitech USB Receiver
 DEBUG eruption::hwdevices            > Found supported HID device: 0x1e7d:0x30f7 - ROCCAT ROCCAT Vulcan Pro
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x30f7 (0003:0009:00) - ROCCAT ROCCAT Vulcan Pro
 INFO  eruption::hwdevices::roccat_vulcan_pro > Bound driver: ROCCAT Vulcan Pro
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x30f7 - ROCCAT ROCCAT Vulcan Pro
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x30f7 - ROCCAT ROCCAT Vulcan Pro
 DEBUG eruption::hwdevices                    > Found supported HID device: 0x1e7d:0x30f7 - ROCCAT ROCCAT Vulcan Pro
 DEBUG eruption::hwdevices                    > Found unsupported HID device: 0x46d:0xa29 - Logitech Logitech Wireless Headset
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: <unknown>
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 DEBUG eruption                               > Registering filesystem watcher...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 DEBUG eruption                               > Entering the main loop now...
 INFO  eruption                               > Switching to profile: /var/lib/eruption/profiles/matrix.profile
 DEBUG eruption::scripting::script            > Applying parameter Bool { name: "horizontal", description: "Horizontal or vertical", default: true }
 DEBUG eruption::scripting::script            > Applying parameter Int { name: "direction", description: "-1 == left/down, +1 == right/up", min: None, max: None, default: -1 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "scale_factor", description: "Scale factor of the alpha value", min: Some(0.0), max: Some(255.0), default: 127.0 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "speed_divisor", description: "Speed of the wave", min: Some(0.0), max: Some(1000.0), default: 25.0 }
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "wave_length", description: "Wave length divisor", min: Some(0.0), max: Some(1000.0), default: 4.0 }
[... cut ...]
Event: time 1615961281.068622, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e0
Event: time 1615961281.068622, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1615961281.068622, -------------- SYN_REPORT ------------
Event: time 1615961281.301472, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70006
Event: time 1615961281.301472, type 1 (EV_KEY), code 46 (KEY_C), value 1
Event: time 1615961281.301472, -------------- SYN_REPORT ------------
^C%
```
Thr keys that I have pressed:

`LEFT CTRL`
`LEFT CTRL` + `w`
`LEFT CTRL`
`LEFT CTRL` + `d`

X3n0m0rph59, 2021-03-19:
@TheMysteriousOH Thank you for posting the logs! I believe this to be fixed now by the latest changes, pushed to the `master` branch.

Due to a wrong set of key indices in `eruption/src/scripts/lib/hwdevices/keyboards/roccat_vulcan_pro.lua`, the EasyShift+ feature got activated by pressing the `LEFT CTRL` key. This caused Eruption to alter the stream of generated input events, and inject the events `KEY_F13` instead of `LEFT CTRL` + `W` and `KEY_F14` instead of `LEFT CTRL` + `D`.

Could you please verify that with this set of changes to following is true:

* If you press and hold down FN, do you see the keys `F1`-`F4` and `M1`-`M6` and the media keys being highlighted?
* If you enter GameMode by pressing `FN` + `GameMode`, and then press and hold down `CAPS LOCK`, do you see the number keys `1`-`3` beeing highlighted in green?

Since I don't have access to a ROCCAT Vulcan Pro (non-TKL), I can not verify this myself.

TheMysteriousOH, 2021-03-20:
@X3n0m0rph59 Thank you for fixing the 'Left Control' bug

> Could you please verify that with this set of changes to following is true:
> 
> * If you press and hold down FN, do you see the keys `F1`-`F4` and `M1`-`M6` and the media keys being highlighted?

The following keys are highlighted in the these colors:
**ice blue**
`d` `f` `y` `x`

**light blue**
`8`
`insert`
`delete`
`left arrow`
`F7` `F8` `F9`

**red**
`Page down`
 
**green**
Num Pad: `8` `9` `5` `6` `1`

> * If you enter GameMode by pressing `FN` + `GameMode`, and then press and hold down `CAPS LOCK`, do you see the number keys `1`-`3` beeing highlighted in green?

The following keys where highlighted in red:
`ESC` `F1`
`1` `2`
`Tab` `q` `w`
`CAPS LOCK`
`left Shift` `<`
`left Control` `Windows Key` `Alt`

The `^` Key is highlighted in white

Please keep in mind, that I have a German key layout.

X3n0m0rph59, 2021-03-20:
@TheMysteriousOH Thanks for the detailed response!

Judging from the description, it seems that the indices for the macro and function keys are still incorrect. I will look into this soon...

X3n0m0rph59, 2021-03-21:
@TheMysteriousOH I have pushed the latest round of updates to the `master` branch!

I tried my best to guess the key indices but since I don't have access to a ROCCAT Vulcan Pro (non-TKL) I could not test this code.

Key indices are listed in the file `eruption/src/scripts/lib/hwdevices/keyboards/roccat_vulcan_pro.lua` starting at line `29`

```lua
key_to_index['GAME_MODE'] = 104

key_to_index['FN'] = 65
key_to_index['CAPS_LOCK'] = 4
key_to_index['LEFT_SHIFT'] = 5
key_to_index['RIGHT_SHIFT'] = 83
key_to_index['LEFT_CTRL'] = 6
key_to_index['RIGHT_CTRL'] = 90
key_to_index['LEFT_ALT'] = 17
key_to_index['RIGHT_ALT'] = 71
key_to_index['RIGHT_MENU'] = 84

key_to_index['ESC'] = 1

key_to_index['F1'] = 14
key_to_index['F2'] = 21
key_to_index['F3'] = 26
key_to_index['F4'] = 31

key_to_index['F5'] = 41
key_to_index['F6'] = 48
key_to_index['F7'] = 54
key_to_index['F8'] = 60

key_to_index['F9'] = 66
key_to_index['F10'] = 72
key_to_index['F11'] = 78
key_to_index['F12'] = 80

key_to_index['INSERT'] = 85
key_to_index['POS1'] = 89
key_to_index['PGUP'] = 94
key_to_index['DEL'] = 86
key_to_index['END'] = 90
key_to_index['PGDWN'] = 95

key_to_index['UP'] = 91
key_to_index['DOWN'] = 92
key_to_index['LEFT'] = 87
key_to_index['RIGHT'] = 96

key_to_index['1'] = 9
key_to_index['2'] = 15
key_to_index['3'] = 22

key_to_index['W'] = 16
key_to_index['A'] = 11
key_to_index['S'] = 17
key_to_index['D'] = 24
```

It would be really cool if you could go through this list and check the debug output of the Eruption daemon!
So e.g. if you press key `LEFT_CTRL` Eruption will print `Macros: Key down: Index: 6`. Therefore we need to assign the value `6`.

The desired final result would be that when you hold down the `FN` key you would see:

* Highlighting of `F1` - `F4` in ice blue (these are the profile selectors)
* Highlighting of `F5` - `F8` and  `F9` - `F12` in light blue (special function keys)
* Highlighting of the macro keys M1-M6 in green (macros)
* Highlighting of the `GAME MODE`/`SCRL LCK` key (toggle game mode)

And in game mode (after pressing `FN` + `GAME MODE`/`SCRL LCK` key, the game mode should be enabled)

* The `CAPS LOCK`/`EASY SHIFT` key should turn red when `FN` is held down to signal that the EasyShift feature is enabled.

If you hold down `CAPS LOCK`/`EASY SHIFT` while __game mode is enabled__, you should see:

* Flashing `CAPS LOCK`/`EASY SHIFT` key in red and white
* The keys `1`, `2` and `3` should be highlighted in green, since the default `macros.lua` binds 3 example user macros to these keys
* The keys `ESC` and `F1` - `F12` should be highlighted in blue (They are assigned a remapping to  `MUTE` and to `F13` - `F24`)
* The keys `M1` - `M6` should be highlighted in pink. They are used to switch between different EasyShift layers.
* `FN` should be highlighted in blue

TheMysteriousOH, 2021-03-21:
> @TheMysteriousOH I have pushed the latest round of updates to the `master` branch!
>
> I tried my best to guess the key indices but since I don't have access to a ROCCAT Vulcan Pro (non-TKL) I could [...]

I note the different values in the Bold Lines.

> key_to_index['GAME_MODE'] = 104
> 
**key_to_index['FN'] = 77**
> key_to_index['CAPS_LOCK'] = 4
> key_to_index['LEFT_SHIFT'] = 5
> key_to_index['RIGHT_SHIFT'] = 83
> key_to_index['LEFT_CTRL'] = 6
> key_to_index['RIGHT_CTRL'] = 90
> key_to_index['LEFT_ALT'] = 17
> key_to_index['RIGHT_ALT'] = 71
> key_to_index['RIGHT_MENU'] = 84
> 
> key_to_index['ESC'] = 1
> 
**key_to_index['F1'] = 12**
**key_to_index['F2'] = 18**
**key_to_index['F3'] = 24**
**key_to_index['F4'] = 29**
> 
**key_to_index['F5'] = 49**
**key_to_index['F6'] = 54**
**key_to_index['F7'] = 60**
**key_to_index['F8'] = 66**
> 
**key_to_index['F9'] = 79**
**key_to_index['F10'] = 85**
**key_to_index['F11'] = 86**
**key_to_index['F12'] = 87**
> 
**key_to_index['INSERT'] = 101**
**key_to_index['POS1'] = 105**
**key_to_index['PGUP'] = 110**
**key_to_index['DEL'] = 102**
**key_to_index['END'] = 106**
**key_to_index['PGDWN'] = 111**
> 
**key_to_index['UP'] = 107**
**key_to_index['DOWN'] = 108**
**key_to_index['LEFT'] = 103**
**key_to_index['RIGHT'] = 112**
> 
**key_to_index['1'] = 7**
**key_to_index['2'] = 13**
**key_to_index['3'] = 19**
> 
**key_to_index['W'] = 14**
**key_to_index['A'] = 9**
**key_to_index['S'] = 15**
**key_to_index['D'] = 21**
> ```

> 
> The desired final result would be that when you hold down the `FN` key you would see:
> 
* Highlighting of `F1` - `F4` in ice blue (these are the profile selectors) **ok**
* Highlighting of `F5` - `F8` and  `F9` - `F12` in light blue (special function keys) **ok**
* Highlighting of the macro keys M1-M6 in green (macros) **ok**
* Highlighting of the `GAME MODE`/`SCRL LCK` key (toggle game mode) **not ok**
The `/` on the NumPad is highlighted in red
> 
> And in game mode (after pressing `FN` + `GAME MODE`/`SCRL LCK` key, the game mode should be enabled)

It seems that the daemon will not switch to game mode:

Here is the debug, when I press `FN` + `GAME MODE`
```
 DEBUG eruption                               > FPS: 18
 DEBUG eruption::hwdevices::roccat_vulcan_pro >   |0300fb77 01000000|                   ...w....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 2 args: 119
 DEBUG eruption::scripting::script::callbacks > Macros: FN key event registered
 DEBUG eruption::scripting::script::callbacks > Macros: Key down: Index: 77
 DEBUG eruption::hwdevices::roccat_vulcan_pro >   |0300fb60 01000000|                   ...`....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 2 args: 96
 DEBUG eruption::hwdevices::roccat_vulcan_pro >   |03002100 00000000|                   ..!.....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_pro >   |0300fb60 00000000|                   ...`....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 1 args: 96
 DEBUG eruption::hwdevices::roccat_vulcan_pro >   |0300fb77 00000000|                   ...w....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 1 args: 119
 DEBUG eruption::scripting::script::callbacks > Macros: FN key event registered
 DEBUG eruption::scripting::script::callbacks > Macros: Key up: Index: 77
 DEBUG eruption                               > FPS: 18
```

X3n0m0rph59, 2021-03-22:
@TheMysteriousOH Thank you very much, this really __helped a lot__!

I have incorporated your changes to the code and pushed out the latest round of updates to the `master` branch!

>Highlighting of the GAME MODE/SCRL LCK key (toggle game mode) not ok
The / on the NumPad is highlighted in red

I took a wild guess and assigned `key_to_index['GAME_MODE'] = 95`. If this is not the right index for the `SCRL LCK` key, you should be able to obtain the correct key index by pressing `SCRL LCK` without the `FN` key being held down.

I also changed the USB HID code of `GAME_MODE_KEY = 96` (taken from the log `Macros: HID event: 1 args: 96`). I hope that with these changes, the daemon is able to switch to game mode now!

Please tell me if the daemon now correctly does the following things __when in game mode__:

* The CAPS LOCK/EASY SHIFT key should turn red when FN is held down to signal that the EasyShift feature is enabled.

If you hold down CAPS LOCK/EASY SHIFT while game mode is enabled, you should see:

* Flashing CAPS LOCK/EASY SHIFT key in red and white
* The keys 1, 2 and 3 should be highlighted in green, since the default macros.lua binds 3 example user macros to these keys
* The keys ESC and F1 - F12 should be highlighted in blue (They are assigned a remapping to MUTE and to F13 - F24)
* The keys M1 - M6 should be highlighted in pink. They are used to switch between different EasyShift layers.
* FN should be highlighted in blue

TheMysteriousOH, 2021-03-24:
I have now edit the file: eruption/src/scripts/lib/hwdevices/keyboards/roccat_vulcan_pro.lua
`key_to_index['GAME_MODE'] = 104`
The toggle of the Game_Mode is working, but in the opposite order:

When I'm **not** in Game Mode and I press CAPS LOCK I got the following Colors:

**red:**
`^`

**green:**
`<`
`ALT`

**blue:**
`LEFT Shift`
`F7`
`F8`
`F9`
`d` `f` `y` `x` `8`
`INSERT (M1)`
`DEL (M4)`
`RIGHT Control`
`LEFT Arrow`

**pink:**
Num Pad: `8` `9` `5` `6` `0`

**white:**
`1`

When I press CAPS Lock After switching into Game Mode, no keys where highlighted.

X3n0m0rph59, 2021-03-24:
@TheMysteriousOH Thanks again for providing this information!

I believe you maybe can work around the issue by stopping the Eruption daemon, turning off game mode and then restart the Eruption daemon again.

If you hold down CAPS LOCK/EASY SHIFT while game mode is enabled, you should see:

* Flashing CAPS LOCK/EASY SHIFT key in red and white
* The keys 1, 2 and 3 should be highlighted in green, since the default macros.lua binds 3 example user macros to these keys
* The keys ESC and F1 - F12 should be highlighted in blue (They are assigned a remapping to MUTE and to F13 - F24)
* The keys M1 - M6 should be highlighted in pink. They are used to switch between different EasyShift layers.
* FN should be highlighted in blue

Will have a look at this soon...

TheMysteriousOH, 2021-03-25:
@X3n0m0rph59 at the moment I can not reproduce the error.
At the moment there are no keys highlighted when I press the `CAPS LOCK` key.
Independently if i'm in GAME MODE or not.
Several restart of the eruption daemon did not help.

I get only keys highlighted when I press the `FN` key.

DirkSchueler, 2021-03-30:
I am using a Roccat Vulcan Pro keyboard on Ubuntu 20.10.
The last few changes you made to roccat_vulcan_pro.lua may have disturbed my keyboard settings.
( Game Mode is off ) 
1. Volume control does'nt have that greate gauge/progress bar light effect
2. Pressing Fn now only lights up d,f,y,x in cyan and / red and 0,1,5,8,9,6 in green
Maybe it depends on the keyboard's firmware version?

X3n0m0rph59, 2021-03-31:
@TheMysteriousOH, @DirkSchueler I have just reverted the last changes to the "hardware definition" Lua script file of the ROCCAT Vulcan Pro.

This is entirely untested, since I lack the hardware. Please let me know if this fixes the problems.

DirkSchueler, 2021-04-01:
That's by far better! 
Holding Fn key now lights F-keys, M-keys as before.
But...
1. DivideBy (/) in red light while pressing Fn 
2. Still no light effect on volume change. Turning the wheel removes current Profile for approx. 1 sec.

JonLit, 2022-07-10:
Volume change works fine for me, I only had one issue: It's stuck on the Matrix profile, but after renaming the matrix.profile file to matrix1.profile, I could change profiles.
Also, in the eruption-GUI, the FN key color is always displayed as black, but the actual FN key lights up as it should
For me the DivideBy (÷ on the numpad) key also lights up in red when holding the FN key