eruption-project/eruption #50: Add support for ROCCAT Kone Aimo Remastered

X3n0m0rph59, 2021-03-04:
This issue will track the progress on the support for the ROCCAT Kone Aimo Remastered

_Originally posted by @greenseeker in https://github.com/X3n0m0rph59/eruption/issues/47#issuecomment-787514702_

X3n0m0rph59, 2021-03-04:
@greenseeker, thanks for providing the capture file! Since I currently do not have access to a ROCCAT Kone Aimo Remastered, this is entirely untested.

Shinigami92, 2021-07-05:
If I can help in any way, feel free to contact me
I bought a Roccat Kone Aimo Remastered just now and it will (hopefully) arrive tomorrow :slightly_smiling_face: 

My system:
```yml
Operating System: Arch Linux
KDE Plasma Version: 5.22.2
KDE Frameworks Version: 5.83.0
Qt Version: 5.15.2
Kernel Version: 5.10.47-1-lts (64-bit)
Graphics Platform: X11
Processors: 28 × Intel® Core™ i9-10940X CPU @ 3.30GHz
Memory: 62.5 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2
```

Shinigami92, 2021-07-06:
I run eruption from the terminal first time

without sudo

```node
~  eruption                                                                                                                                                                                                                                         ✔  8s  16:20:24 

Eruption is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

Eruption is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.21 (0.1.20.r26.ged40c60) (release build)
INFO  eruption > Loading saved state...
INFO  eruption > Registering plugins...
INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
INFO  eruption::plugins::persistence > Loading persistent state data from disk...
INFO  eruption                       > Plugins loaded and initialized successfully
INFO  eruption                       > Enumerating connected devices...
INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2e2c (0001:000a:00) - <unknown> <unknown>
INFO  eruption::hwdevices::roccat_kone_aimo_remastered > Bound driver: ROCCAT Kone Aimo Remastered
INFO  eruption                                         > Opening mouse device...
ERROR eruption                                         > Error opening the mouse device: Could not open the device file
ERROR eruption                                         > This could be a permission problem, or maybe the device is locked by another process?
INFO  eruption                                         > Initializing mouse device...
ERROR eruption                                         > Could not initialize the device: Device not opened
INFO  eruption                                         > Configuring mouse LEDs...
ERROR eruption                                         > Could not initialize LEDs: Device not opened
INFO  eruption                                         > Firmware revision: <unknown>
INFO  eruption                                         > Spawning mouse input thread...
INFO  eruption                                         > Device enumeration completed
INFO  eruption                                         > Performing late initializations...
INFO  eruption                                         > Initializing D-Bus API...
INFO  eruption                                         > Late initializations completed
INFO  eruption                                         > Startup completed
INFO  eruption                                         > Switching to profile: /var/lib/eruption/profiles/blue-fx-swirl-perlin.profile
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/swirl-perlin.lua
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/solid.lua
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/halo.lua
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/impact.lua
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/macros.lua
ERROR eruption                                         > Could not process a D-Bus event: receiving on an empty and disconnected channel
INFO  eruption                                         > Now listening on mouse: /dev/input/event19
INFO  eruption                                         > Input device name: "ROCCAT ROCCAT Kone Aimo 16K Mouse"
INFO  eruption                                         > Input device ID: bus 0x3 vendor 0x1e7d product 0x2e2c
INFO  eruption                                         > Physical location: usb-0000:00:14.0-6.4/input0
INFO  eruption                                         > Grabbing the mouse device exclusively
The application panicked (crashed).
Message:  Could not grab the device, terminating now.: Os { code: 16, kind: Other, message: "Device or resource busy" }
[... cut ...]
INFO  eruption                                         > Grabbing the mouse device exclusively
The application panicked (crashed).
Message:  Could not grab the device, terminating now.: Os { code: 16, kind: Other, message: "Device or resource busy" }
Location: eruption/src/main.rs:499

Backtrace omitted.
Run with RUST_BACKTRACE=1 environment variable to display it.
Run with RUST_BACKTRACE=full to include source snippets.
[1]    11244 abort      sudo eruption
```

So in both cases it doesn't start :(
I would like to configure the LEDs

Shinigami92, 2021-07-06:
After a whole system restart, the behavior is different :eyes: 

without sudo
```node
~  eruption                                                                                                                                                                                                                                              ✔  17:10:40 

Eruption is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

Eruption is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.21 (0.1.20.r26.ged40c60) (release build)
INFO  eruption > Loading saved state...
INFO  eruption > Registering plugins...
INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
INFO  eruption::plugins::persistence > Loading persistent state data from disk...
INFO  eruption                       > Plugins loaded and initialized successfully
INFO  eruption                       > Enumerating connected devices...
INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2e2c (0001:000a:00) - <unknown> <unknown>
INFO  eruption::hwdevices::roccat_kone_aimo_remastered > Bound driver: ROCCAT Kone Aimo Remastered
INFO  eruption                                         > Opening mouse device...
ERROR eruption                                         > Error opening the mouse device: Could not open the device file
ERROR eruption                                         > This could be a permission problem, or maybe the device is locked by another process?
INFO  eruption                                         > Initializing mouse device...
ERROR eruption                                         > Could not initialize the device: Device not opened
INFO  eruption                                         > Configuring mouse LEDs...
ERROR eruption                                         > Could not initialize LEDs: Device not opened
INFO  eruption                                         > Firmware revision: <unknown>
INFO  eruption                                         > Spawning mouse input thread...
INFO  eruption                                         > Device enumeration completed
INFO  eruption                                         > Performing late initializations...
INFO  eruption                                         > Initializing D-Bus API...
INFO  eruption                                         > Late initializations completed
INFO  eruption                                         > Startup completed
INFO  eruption                                         > Switching to profile: /var/lib/eruption/profiles/blue-fx-swirl-perlin.profile
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/swirl-perlin.lua
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/solid.lua
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/halo.lua
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/impact.lua
INFO  eruption                                         > Loading Lua script: /usr/share/eruption/scripts/macros.lua
ERROR eruption                                         > Could not send a pending dbus API event: sending on a disconnected channel
ERROR eruption                                         > Could not process a D-Bus event: receiving on an empty and disconnected channel
INFO  eruption                                         > Now listening on mouse: /dev/input/event19
INFO  eruption                                         > Input device name: "ROCCAT ROCCAT Kone Aimo 16K Mouse"
INFO  eruption                                         > Input device ID: bus 0x3 vendor 0x1e7d product 0x2e2c
INFO  eruption                                         > Physical location: usb-0000:00:14.0-6.4/input0
INFO  eruption                                         > Grabbing the mouse device exclusively
[... cut ...]
ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused
^C INFO  eruption                                         > Shutting down all Lua VMs now...
INFO  eruption::plugins::persistence                   > Storing persistent state data to disk...
INFO  eruption                                         > Closing all devices now...
INFO  eruption                                         > Exiting now
```