eruption-project/eruption #51: Build is failing

cadenhenrich, 2021-03-07:
**Describe the bug**
Both using `paru` to build and cloning the repo directly and following the instructions results in the build failing.

**To Reproduce**
Steps to reproduce the behavior:

1. Clone the git repository
2. `cd` into the directory
3. Run `cargo build --all --release` 

**Expected behavior**
The build completes and an executable is the result.

**Logs**
```
   Compiling dbus-tokio v0.7.3
error[E0277]: the trait bound `i32: std::os::unix::io::AsRawFd` is not satisfied
   --> /home/caden/.cargo/registry/src/github.com-1ecc6299db9ec823/dbus-tokio-0.7.3/src/connection.rs:59:16
    |
59  |     Registered(AsyncFd<RawFd>),
    |                ^^^^^^^^^^^^^^ the trait `std::os::unix::io::AsRawFd` is not implemented for `i32`
    |
   ::: /home/caden/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.2.0/src/io/async_fd.rs:137:23
    |
137 | pub struct AsyncFd<T: AsRawFd> {
    |                       ------- required by this bound in `tokio::io::unix::AsyncFd`

error[E0277]: the trait bound `i32: std::os::unix::io::AsRawFd` is not satisfied
   --> /home/caden/.cargo/registry/src/github.com-1ecc6299db9ec823/dbus-tokio-0.7.3/src/connection.rs:189:34
    |
189 |     poll_ready: impl FnOnce() -> task::Poll<std::io::Result<AsyncFdReadyGuard<'a, RawFd>>>,
    |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `std::os::unix::io::AsRawFd` is not implemented for `i32`
    |
   ::: /home/caden/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.2.0/src/io/async_fd.rs:148:37
    |
148 | pub struct AsyncFdReadyGuard<'a, T: AsRawFd> {
    |                                     ------- required by this bound in `tokio::io::unix::AsyncFdReadyGuard`

error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0277`.
error: could not compile `dbus-tokio`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

**Please complete the following information:**
 - Eruption version 0.1.19-25
 - Roccat KONE Pure ultra

**Additional context**
This is a problem with installing the product. I cannot get an executable, so this is not about the actual program itself. This is about the install process.

cadenhenrich, 2021-03-07:
This looks to be an error in the `dbus-tokio` lib, although I'm not quite sure

X3n0m0rph59, 2021-03-07:
Hi @cadenhenrich, thanks for reporting this issue!

This seems to be related to an outdated version of rust. Eruption needs at least the current stable version `1.50.0` to compile. I suspect that you are using the old stable release of rust.

```sh
 $ rustc --version
```

If you are using rustup from [https://rustup.rs/](https://rustup.rs/), you can switch to the stable branch of rust by running this command:
```sh
 $ rustup default stable
```

cadenhenrich, 2021-03-07:
Yep, pretty sure this was the issue, must have been on another version of rust from another project I'm working on. Closing the issue.

cadenhenrich, 2021-03-07:
Also, if there is a way to include `rustup default stable` in the AUR install, that would probably be a good idea since I was also unable to install using that method.

closed 2021-03-07 by cadenhenrich