eruption-project/eruption #52: Changing mouse options

cadenhenrich, 2021-03-09:
> Moving this to a different thread because I didn't want to clog up the other one. See issue #47 for other information.

Given the instructions,

```
On a ROCCAT Kone Pure Ultra mouse, this is how to enable zero-debounce:

Find mouse device index like this:
 $ sudo eruption-debug-tool list

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 

Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other sub-commands of this tool

Index: 00: ID: 46d:c051 Logitech/USB-PS/2 Optical Mouse Subdev: 0
Index: 01: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 0
Index: 02: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 1
Index: 03: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 2
Index: 04: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 3
Index: 05: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 0
Index: 06: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 1
Index: 07: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 2
Index: 08: ID: 1e7d:311a ROCCAT/ROCCAT Vulcan Pro TKL Subdev: 3

Enumeration completed
So the device index on my system would be 1.

We need to write to the device with index 1, as determined previously
Byte 3 of the data structure represents the zero-debounce configuration option (on a ROCCAT Kone Pure Ultra)
So, to turn on zero-debounce, I would need to run this command:

 $ sudo eruption-debug-tool write 1 '[0x11, 0x0e, 0x01, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x4c, 0x00, 0x05, 0x05, 0x00, 0x08, 0x00, 0x10, 0x00, 0x18, 0x00, 0x20, 0x00, 0x40, 0x00, 0x00, 0x00, 0x03, 0x09, 0x06, 0xff, 0x0f, 0x00, 0x00, 0x14, 0xff, 0xff, 0x00, 0x00, 0x14, 0xff, 0x00, 0x48, 0xff, 0x14, 0xff, 0x00, 0x48, 0xff, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xf7, 0x09, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00]'
And to turn off the zero-debounce option, I run this command:

 $ sudo eruption-debug-tool write 1 '[0x11, 0x0e, 0x00, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x4c, 0x00, 0x05, 0x05, 0x00, 0x08, 0x00, 0x10, 0x00, 0x18, 0x00, 0x20, 0x00, 0x40, 0x00, 0x00, 0x00, 0x03, 0x09, 0x06, 0xff, 0x0f, 0x00, 0x00, 0x14, 0xff, 0xff, 0x00, 0x00, 0x14, 0xff, 0x00, 0x48, 0xff, 0x14, 0xff, 0x00, 0x48, 0xff, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xf7, 0x09, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00]'
To read out the state from the device:

 $ sudo eruption-debug-tool read 1 0x11 78
In case your mouse device needs a different byte sequence
Save a state snapshot of your mouse using:
 $ sudo eruption-debug-tool state-diff 1
Fire up ROCCAT Swarm (e.g. in a VM), change a single setting (zero-debounce)
Again, save a state snapshot of your mouse using:
 $ sudo eruption-debug-tool state-diff 1
On the second invocation of state-diff you will get a diff view of the changed bytes.

 $ sudo eruption-debug-tool state-diff 1

 Please stop the Eruption daemon prior to running this tool:
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 You can re-enable Eruption with this command afterwards:
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
 
Index: 01: ID: 1e7d:2dd2 ROCCAT/ROCCAT Kone Pure Ultra Subdev: 0
Reading data from device...
The following USB HID report IDs have changed bytes:

Changed bytes: [116]
0x05: [0x05, 0x80, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00, 0x03, 0x00, 0xef, 0xe7, 0x01, 0x00, 0x00, 0x00, 0x90, 0x93, 0x00, 0x00, 0x94, 0x93, 0x00, 0x00, 0xa4, 0x93, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x09, 0x01, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x20, 0x00, 0x00, 0x00, 0x23, 0x00, 0x00, 0x00, 0x22, 0x40, 0x78, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0xff, 0x00, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00=>0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x38, 0x00, ]

Changed bytes: [116]
0x04: [0x04, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0x00, 0xef, 0xe7, 0x01, 0x00, 0x00, 0x00, 0x90, 0x93, 0x00, 0x00, 0x94, 0x93, 0x00, 0x00, 0xa4, 0x93, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x09, 0x01, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x20, 0x00, 0x00, 0x00, 0x23, 0x00, 0x00, 0x00, 0x22, 0x40, 0x78, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0xff, 0x00, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00=>0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x38, 0x00, ]

Changed bytes: [2, 12]
0x11: [0x11, 0x0e, 0x00=>0x01, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x05, 0x4c=>0x4d, 0x00, 0x00, 0x00, 0x00, 0x09, 0x01, 0x00, 0x00, 0x0a, 0x01, 0x00, 0x00, 0x05, 0x01, 0x00, 0x00, 0x06, 0x01, 0x00, 0x00, 0x02, 0x02, 0x00, 0x00, 0x03, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, ]

Saving state data...
Done
```
I have to do the second way since the first option did not enable zero debounce for me (still unable to drag click). I've installed a windows VM, but the ROCCAT Swarm software does not recognize my mouse device, and whenever I try to apply the updates it prompts me for on boot the program crashes. I would ask for help elsewhere since I know this is a bit besides the purpose of this software, but I can't really find any other place to ask.

Is there anything you would recommend that would allow my mouse to be recognized in the windows VM? If not, is there any other way to see if I can enable the zero debounce option?

Also I tried drag clicking with another mouse, and it still didn't work, although I wasn't able to turn down the debounce time on that one so it might just be that reason that it didn't work.

X3n0m0rph59, 2021-03-09:
>I have to do the second way since the first option did not enable zero debounce for me (still unable to drag click). 

Could you please post the output of
```sh
 $ sudo eruption-debug-tool read <your device index> 0x11 78
```

Please replace `<your device index>` with subdevice 0 of your mouse. You can find it with 
```sh
 $ sudo eruption-debug-tool list
```

>Is there anything you would recommend that would allow my mouse to be recognized in the windows VM? If not, is there any other way to see if I can enable the zero debounce option?

What virtualization software are you using? Did you try to forward the USB device to the Windows VM?

X3n0m0rph59, 2021-03-09:
@cadenhenrich It could be the case that `libinput` is de-bouncing your clicks.

You may want to try to create the file `/etc/libinput/local-overrides.quirks`:
```sh
[ROCCAT Kone Pure Ultra]
MatchUdevType=mouse
MatchName=*ROCCAT Kone Pure Ultra
ModelBouncingKeys=1
```

closed 2021-03-09 by cadenhenrich