eruption-project/eruption #53: eruption and pipewire

Glog78, 2021-03-13:
**Describe the bug**
Not a bug / feature request

**To Reproduce**
none

**Expected behavior**
I would love to use audio viz on distributions with pipewire

**Logs**
none

As far as i see currently pulseaudio and the root "workaround" is requiered to get the audio setup done. Could you check if it could also work with a local pulse server (localhost:port) which can be provided with pipewire too. Additional config option maybe ? Needs maybe to check periodically if the server is started or something like this since there pipewire daemon in the user context.

X3n0m0rph59, 2021-03-14:
@Glog78 Thanks for posting this feature request!

I have looked into adding support for PipeWire and added some preliminary documentation on how to set up audio for PipeWire enabled Linux distributions.

I got it working using the PulseAudio interface of PipeWire on `ArcoLinux` (Arch based) using the packages `pipewire-git` and  `pipewire-pulse-git`. I only tried the unix domain socket transport, not the IP socket based transport, but I suppose that would work too.

Glog78, 2021-03-16:
Thanks alot for the additional information. I think i will update to pipewire-git and pipewire-pulse-git in this case and check over. For me this one can be closed and i will report back once i can check (might take me a little).

closed 2021-03-16 by X3n0m0rph59