eruption-project/eruption #54: Corrupt profile crashes driver

Madcliff, 2021-03-15:
If the profile slots is ever set invalidly, it causes an irrecoverable crash in the driver. This should ideally reset to a default configuration. As stands it typically requires a purge and reinstall of the driver to get to the default profile slot settings.

X3n0m0rph59, 2021-03-17:
@Madcliff I have just pushed out the first round of patches to the `master` branch, that should make the Eruption daemon somewhat less prone to crashes.

It will survive now in most cases, when a .profile file or a Lua script has invalid syntax, or encounters a runtime error.

Madcliff, 2021-03-17:
Getting a build error when I update the PKG

==> Entering fakeroot environment...
==> Starting package()...
install: cannot stat 'eruption/src/scripts/failsafe.lua': No such file or directory
==> ERROR: A failure occurred in package().
    Aborting...

X3n0m0rph59, 2021-03-17:
@Madcliff I fixed the PKGBUILD. It should work now.

X3n0m0rph59, 2021-04-07:
I will close this issue now, please reopen it if the problem persists!

closed 2021-04-07 by X3n0m0rph59