eruption-project/eruption #55: Eruption fails to start

LuxLucian, 2021-03-18:
The program fails to start with the following

**Logs**
https://pastebin.com/DwzgTM1d

`sudo systemctl -u eruption.service -b` does not seem to work and returns " systemctl: invalid option -- 'u' "

**Screenshots (optional)**
[image]

**Please complete the following information:**
 - Eruption version [e.g. 0.1.19]
 - ROCCAT ROCCAT Kone Aimo 16K

X3n0m0rph59, 2021-03-18:
@LuxLucian Thanks for reporting this issue!

Eruption does not recognize your mouse device (it does not bind the driver to the device). It seems that the USB ID is missing from the list of supported hardware.

Could you please show the output of:

```sh
 $ sudo lsusb
```

The command `sudo systemctl -u eruption.service -b` should read `sudo journalctl -u eruption.service -b`. I've corrected the GitHub issue template. Thanks for bringing this to my attention!

X3n0m0rph59, 2021-03-18:
The driver for your device was added in the yet to be released version `0.1.20`. Eruption version `0.1.19` does not support this device. Please refer to issue #50 to track the support for your device.

X3n0m0rph59, 2021-03-18:
I will close this issue now, please post follow-ups via issue #50.

closed 2021-03-18 by X3n0m0rph59