eruption-project/eruption #57: Deamon not found

GalaxyGamer5, 2021-04-10:
So I am the dude from your Reddit post, and I tried the steps you told me to do, but I am sorry to say, the problem is still there.

I don't know why the "bug" occurred. I installed the most recent version of 20.04 Ubuntu about a week ago.
I am pretty sure I made no mistakes in the installation process.

Like mentioned I get this error message: 
[image]

Then I tried your lines in the Terminal and the result were:
sudo systemctl status eruption.service:
[image]
and sudo journalctl -u eruption.service -b:
[image]
[image]

At the first line the console wouldn't do anything unless I pressed Ctrl+C again.
The second line was in a weird state were, if I pressed enter a new line would write, which would after a while recreate the text that's already written above. 

You asked if my deviced would show LED activity, and they didn't. I mean, yes they did, but only the same way they did before I installed Eruption. (if it does help, I use a Roccat Kone Aimo (the standard Version) and I have no idea if other USB deviced can lead to trubble shooting, but my other two devices are the Hyperx Alloy Elite RGB and the Razer Electra V2 USB, which have both also LEDs in it.)

 * Eruption version:  [0.1.19]
 * Your device(s) :Sudo version 1.8.31
Sudoers policy plugin version 1.8.31
Sudoers file grammar version 46
Sudoers I/O plugin version 1.8.31

X3n0m0rph59, 2021-04-11:
Hi, thanks for posting this bug report!

It seems like you have hit a bug in version `0.1.19` of Eruption. This build does not fully support running in mouse only mode, without a supported keyboard being attached to the system.

I have attached a pre-release package of Eruption version `0.1.20`. If you want to try it out, please install using the following commands. (Assuming the file has been placed in the `Downloads` directory):
```sh
$ cd Downloads/
$ unzip eruption-git_0.1.20-ubuntu0_amd64.deb.zip
$ sudo apt install ./eruption-git_0.1.20-ubuntu0_amd64.deb
$ sudo systemctl enable --now eruption.service
$ eruptionctl switch profile batique.profile
```

[eruption-git_0.1.20-ubuntu0_amd64.deb.zip](https://github.com/X3n0m0rph59/eruption/files/6291188/eruption-git_0.1.20-ubuntu0_amd64.deb.zip)

X3n0m0rph59, 2021-04-16:
@GalaxyGamer5

I have released Eruption `0.1.20`, which should fix the bug.
Please re-open this issue or create another one if you encounter further problems!

closed 2021-04-16 by X3n0m0rph59