eruption-project/eruption #58: eruption-process-monitor not found in Ubuntu Launchpad release

the-unknown, 2021-04-15:
**Introduction**
Hey, first of all: Thank you soooo much for coding this tool. It's awesome and I think it is even better than the Roccat Swarm software on Windows.
I am currently using the release available on the provided Ubuntu Launchpad.

**Please describe the bug**
The current release on the Ubuntu Launchpad seems not to include the eruption-process-monitor.
`systemctl --user enable --now eruption-process-monitor.service` --> `Failed to enable unit: Unit file eruption-process-monitor.service does not exist.`

The support for the Kone AIMO Remastered mouse seems to be missing as well. 

**To Reproduce**
Install Launchpad release and enter `systemctl --user enable --now eruption-process-monitor.service` in the terminal.

**Expected behavior**
eruption-process-monitor launches

**Logs**
`Failed to enable unit: Unit file eruption-process-monitor.service does not exist.`

**Please complete the following information:**

 * Eruption version: `Ubuntu Launchpad 0.1.19`
 * Your device(s) (`sudo lsusb`)
```
Bus 003 Device 006: ID 1e7d:2e2c ROCCAT ROCCAT Kone Aimo 16K
Bus 003 Device 007: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO
```

X3n0m0rph59, 2021-04-16:
Hi @the-unknown, thank you for opening this issue!

I am currently in the process of releasing new packages of Eruption `0.1.20` (for Ubuntu), ETA approximately a few hours.

the-unknown, 2021-04-16:
You're welcome :-)
However I have seen in your launchpad, that you are building packages for hirsute and groovy.
I am on focal (LTS). Could you provide a package for focal as well? 

Thanks a lot.

X3n0m0rph59, 2021-04-16:
>However I have seen in your launchpad, that you are building packages for hirsute and groovy.
I am on focal (LTS). Could you provide a package for focal as well?

Yes I will provide packages for Ubuntu focal as well! It might take a while though, the current build failed...

the-unknown, 2021-04-16:
Great! 
The package was released a view hours ago and I was able to update to `0.1.20`.
The mouse works now, but the main bug/error still persists: `Failed to enable unit: Unit file eruption-process-monitor.service does not exist.`

X3n0m0rph59, 2021-04-16:
Hi @the-unknown!

I tried to make it work in a Ubuntu focal VM:
* `systemctl --user enable --now eruption-process-monitor.service` worked successfully after performing a `systemctl daemon-reload`; A reboot should suffice too. (Please be sure to not accidentally prepend `sudo` to the command)
* Please be sure to use an X11 based session. The `eruption-process-monitor` does currently not support Wayland sessions

Another bug crept in the Ubuntu packages: Please edit the file `/etc/eruption/process-monitor.conf`:

```toml
# Eruption - process-monitor
# Main configuration file

[global]

[X11]
display = ":1"
```

and change the last line to read `display = ":0"`.

I am only able to reproduce the error message that you got, by leaving out the `--user` parameter from the call to `systemctl`.

```sh
$ systemctl status eruption-process-monitor.service
Unit eruption-process-monitor.service could not be found.
```

```sh
$ systemctl --user status eruption-process-monitor.service 
● eruption-process-monitor.service - Process Monitoring and Introspection for Eruption
     Loaded: loaded (/usr/lib/systemd/user/eruption-process-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2021-04-16 21:08:25 CEST; 55s ago
       Docs: man:eruption-process-monitor(1)
             man:process-monitor.conf(5)
             man:eruptionctl(1)
   Main PID: 3035 (eruption-proces)
     CGroup: /user.slice/user-1000.slice/user@1000.service/eruption-process-monitor.service
             └─3035 /usr/bin/eruption-process-monitor -c /etc/eruption/process-monitor.conf daemon

Apr 16 21:08:25 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor::sensors > X11 - Watches the state of windows on the X window system server
Apr 16 21:08:25 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          > Loading rules...
Apr 16 21:08:25 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          >   0: On window focused: Name: '.*YouTube.*' => Switch to profile: /var/lib/eruption/profiles/profile2>
Apr 16 21:08:25 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          >   1: On window focused: Instance: '.*' => Switch to profile: /var/lib/eruption/profiles/profile1.prof>
Apr 16 21:08:25 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          > Startup completed
Apr 16 21:08:25 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          > Triggered action: Switch to profile: /var/lib/eruption/profiles/profile1.profile
Apr 16 21:08:42 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          > Triggered action: Switch to profile: /var/lib/eruption/profiles/profile2.profile
Apr 16 21:08:44 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          > Triggered action: Switch to profile: /var/lib/eruption/profiles/profile1.profile
Apr 16 21:08:45 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          > Triggered action: Switch to profile: /var/lib/eruption/profiles/profile2.profile
Apr 16 21:08:46 ubuntu-vm eruption-process-monitor[3035]:  INFO  eruption_process_monitor          > Triggered action: Switch to profile: /var/lib/eruption/profiles/profile1.profile
```

So automatic switching of profiles *should* work with Eruption `0.1.20` on Ubuntu focal. If all the above is still not sufficient to make it work, please post the output of 

```sh
$ RUST_LOG=debug eruption-process-monitor daemon
```

the-unknown, 2021-04-18:
Hey there, 

the error persists. 
`systemctl --user enable --now eruption-process-monitor.service`
`Failed to enable unit: Unit file eruption-process-monitor.service does not exist.`

`systemctl --user status eruption-process-monitor.service`
`Unit eruption-process-monitor.service could not be found.`

Output of RUST_LOG=debug eruption-process-monitor daemon
```INFO  eruption_process_monitor > Starting eruption-process-monitor: Version 0.0.5
 INFO  eruption_process_monitor > Registering plugins...
 INFO  eruption_process_monitor::sensors > Registering sensor plugins:
 INFO  eruption_process_monitor::sensors > Process - Watches the system for process events
 INFO  eruption_process_monitor::sensors > X11 - Watches the state of windows on the X window system server
 INFO  eruption_process_monitor          > Loading rules...
 INFO  eruption_process_monitor          >   0: On window focused: Instance: '.*' => Switch to profile: /var/lib/eruption/profiles/swirl-perlin-rainbow.profile (enabled: true, internal: true)
 DEBUG eruption_process_monitor          > Registering filesystem watcher...
 INFO  eruption_process_monitor          > Startup completed
 DEBUG eruption_process_monitor          > Entering the main loop now...
 DEBUG eruption_process_monitor::dbus_interface > Message: Message { Type: Signal, Path: "/org/freedesktop/DBus", Interface: "org.freedesktop.DBus", Member: "NameAcquired", Sender: "org.freedesktop.DBus", Destination: ":1.161", Serial: 2, Args: [":1.161"] }
 DEBUG eruption_process_monitor::dbus_interface > Message: Message { Type: Signal, Path: "/org/freedesktop/DBus", Interface: "org.freedesktop.DBus", Member: "NameAcquired", Sender: "org.freedesktop.DBus", Destination: ":1.161", Serial: 3, Args: ["org.eruption.process_monitor"] }
 INFO  eruption_process_monitor                 > Triggered action: Switch to profile: /var/lib/eruption/profiles/swirl-perlin-rainbow.profile
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
[... cut ...] 0, bad_value: 0, minor_opcode: 0, major_opcode: 20 }
 DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }```

This line: ` DEBUG eruption_process_monitor                 > Could not poll a sensor: X11 error X11Error { error_kind: Window, error_code: 3, sequence: 10, bad_value: 0, minor_opcode: 0, major_opcode: 20 }` appears as I launch the Firefox browser.

X3n0m0rph59, 2021-04-19:
@the-unknown, 
seems that something is wrong with the `.service` file of  the process monitor daemon then.
The log messages `Could not poll a sensor: X11 error X11Error` that you get, are about the properties of the Xorg root window. They are not related to the problem with the service file.

Could you please post the output of the following commands:

```sh
$ dpkg-query -s eruption-git
$ dpkg-query -L eruption-git | grep -i .service
```

The file `/usr/lib/systemd/user/eruption-process-monitor.service` has to exist in the file system. If it is missing, then could you please reinstall `eruption-git`?

```sh
$ sudo apt install --reinstall eruption-git
```

the-unknown, 2021-04-19:
Ok, I have it up an running.

Here is what went wrong:
The output of `systemctl --user enable --now eruption-process-monitor.service` is says:
```
/lib/systemd/system/eruption-hotplug-helper.service
/lib/systemd/system/eruption.service
/lib/systemd/user/eruption-process-monitor.service
```
The file is there, so why does it not work??
A Google search got me here: https://unix.stackexchange.com/questions/224992/where-do-i-put-my-systemd-unit-file

There it says, that systemd files for the user need to be located at: `/etc/systemd/user` or `$HOME/.config/systemd/user`
I copied the file `eruption-process-monitor.service` to `$HOME/.config/systemd/user` and was then able to enable the service.

So you might want to adjust your package so that `eruption-process-monitor.service` is deployed to `$HOME/.config/systemd/user` upon installation.

here is the status output:
```
● eruption-process-monitor.service - Process Monitoring and Introspection for Eruption
     Loaded: loaded (/home/pix3l/.config/systemd/user/eruption-process-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2021-04-19 09:14:26 CEST; 30s ago
       Docs: man:eruption-process-monitor(1)
             man:process-monitor.conf(5)
             man:eruptionctl(1)
   Main PID: 12956 (eruption-proces)
     CGroup: /user.slice/user-1000.slice/user@1000.service/eruption-process-monitor.service
             └─12956 /usr/bin/eruption-process-monitor -c /etc/eruption/process-monitor.conf daemon

Apr 19 09:14:26 pix3lbox systemd[3015]: Started Process Monitoring and Introspection for Eruption.
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor > Starting eruption-process-monitor: Version 0.0.5
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor > Registering plugins...
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor::sensors > Registering sensor plugins:
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor::sensors > Process - Watches the system for process events
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor::sensors > X11 - Watches the state of windows on the X window system server
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor          > Loading rules...
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor          >   0: On window focused: Instance: '.*' => Switch to profile: /var/lib/eruption/profiles/swirl-perlin-rainbow.profile (enabled: true, in>
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor          > Startup completed
Apr 19 09:14:26 pix3lbox eruption-process-monitor[12956]:  INFO  eruption_process_monitor          > Triggered action: Switch to profile: /var/lib/eruption/profiles/swirl-perlin-rainbow.profile
```

X3n0m0rph59, 2021-04-19:
@the-unknown excellent, thank you very much for debugging this issue! I will update the (Ubuntu) packages respectively.

the-unknown, 2021-04-19:
You're welcome. I'm glad that I was able to help.

I think this can be closed then.

closed 2021-04-20 by X3n0m0rph59