eruption-project/eruption #59: Roccat Kone Pure Ultra not detecting on ubuntu 20.04

man-exists, 2021-04-16:
I recently switched to ubuntu from windows and I needed Drivers for my roccat kone pure ultra.

When I execute Eruption GUI I get an error message that says:
The name org.eruption was not provided by any .service files

The issue repeats itself everytime i try to execute.

**Logs**
```
-- Logs begin at Wed 2021-04-14 12:47:19 IDT, end at Fri 2021-04-16 09:29:21 IDT. --
Apr 16 09:13:52 topsilly systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Apr 16 09:13:52 topsilly systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption > Loading saved state...
Apr 16 09:13:52 topsilly eruption[28022]:  WARN  eruption > Could not parse state file: Could not load global runtime state: configuration file "/var/lib/eruption/eruption.state" >
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption > Registering plugins...
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Apr 16 09:13:52 topsilly eruption[28022]:  WARN  eruption                       > Could not load persisted state: No such file or directory (os error 2)
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption                       > Plugins loaded and initialized successfully
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption                       > Enumerating connected devices...
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2dd2 (0001:0004:00) - ROCCAT ROCCAT Kone Pure Ultra
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption::hwdevices::roccat_kone_pure_ultra > Bound driver: ROCCAT Kone Pure Ultra
Apr 16 09:13:52 topsilly eruption[28022]:  INFO  eruption                                    > Opening mouse device...
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Initializing mouse device...
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Configuring mouse LEDs...
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Firmware revision: 106
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Spawning mouse input thread...
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Device enumeration completed
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Performing late initializations...
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Initializing D-Bus API...
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Late initializations completed
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Startup completed
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Switching to profile: profile1.profile
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Now listening on mouse: /dev/input/event8
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Input device name: "ROCCAT ROCCAT Kone Pure Ultra"
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Input device ID: bus 0x3 vendor 0x1e7d product 0x2dd2
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Physical location: usb-0000:00:14.0-8/input0
Apr 16 09:13:53 topsilly eruption[28022]:  INFO  eruption                                    > Grabbing the mouse device exclusively
Apr 16 09:13:53 topsilly eruption[28022]:  ERROR eruption::scripting::script                 > Lua error in file /usr/share/eruption/scripts/macros.lua: runtime error: /usr/share/>
Apr 16 09:13:53 topsilly eruption[28022]: stack traceback:
Apr 16 09:13:53 topsilly eruption[28022]:         ok: in function '__index'
Apr 16 09:13:53 topsilly eruption[28022]:         ok:25: in main chunk
Apr 16 09:13:53 topsilly eruption[28022]:         ok: in function 'require'
Apr 16 09:13:53 topsilly eruption[28022]:         ok:32: in main chunk
Apr 16 09:13:53 topsilly eruption[28022]:         UnknownError
Apr 16 09:13:53 topsilly eruption[28022]:  ERROR eruption                                    > Script execution failed
Apr 16 09:13:55 topsilly systemd[1]: eruption.service: Main process exited, code=killed, status=6/ABRT
Apr 16 09:13:55 topsilly systemd[1]: eruption.service: Failed with result 'signal'.
Apr 16 09:14:07 topsilly systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Apr 16 09:14:07 topsilly systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.19
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption > Loading saved state...
Apr 16 09:14:07 topsilly eruption[28898]:  WARN  eruption > Could not parse state file: Could not load global runtime state: configuration file "/var/lib/eruption/eruption.state" >
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption > Registering plugins...
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Apr 16 09:14:07 topsilly eruption[28898]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
[... 13 lines cut ...]
Apr 16 09:14:08 topsilly eruption[28898]:  INFO  eruption                                    > Device enumeration completed
Apr 16 09:14:08 topsilly eruption[28898]:  INFO  eruption                                    > Performing late initializations...
Apr 16 09:14:08 topsilly eruption[28898]:  INFO  eruption                                    > Initializing D-Bus API...
[... cut ...]
Apr 16 09:14:08 topsilly eruption[28898]:  ERROR eruption                                    > Script execution failed
Apr 16 09:14:08 topsilly systemd[1]: eruption.service: Main process exited, code=killed, status=6/ABRT
Apr 16 09:14:08 topsilly systemd[1]: eruption.service: Failed with result 'signal'.
```

**Please complete the following information:**

 * Eruption 0.1.19
 * Mouse Device: ROCCAT ROCCAT Kone Pure Ultra (lsusb)

**Additional context**
The driver should support Roccat Kone Pure Ultra

X3n0m0rph59, 2021-04-16:
Hi @man-exists, thank you for reporting this issue!

You have encountered a known bug in Eruption `0.1.19`. It does not work without a supported keyboard being attached to the system. I am currently in the process of rolling out new packages of Eruption version `0.1.20`, ETA a few hours. The bug should be fixed there.

man-exists, 2021-04-16:
thanks. btw, will there be a mouse macro feature (while holding do x)

X3n0m0rph59, 2021-04-16:
>thanks. btw, will there be a mouse macro feature (while holding do x)?

Eruption is still very much centered around the keyboard atm. I will look into advancing `macros.lua` in a direction where it will be useful for users with the "supported mouse, but unsupported keyboard" configuration. Currently we have no way of entering  `Game Mode` on unsupported keyboards. That would be required for mouse macros to work.

Did you consider using libratbag/Piper as well? I don't know if they support the `ROCCAT Kone Pure Ultra`, but I suspect that if this device is supported, they would offer a broader set of features than what Eruption currently delivers.

I have access to a `ROCCAT Kone Pure Ultra` too, so I can implement new features without a round-trip. Maybe, I finally get around implementing device specific configuration (DPI, zero-debounce,...) soon.

P.s.:
Should you need to enable zero-debounce under Linux please refer to https://github.com/X3n0m0rph59/eruption/issues/47#issuecomment-793542865.

X3n0m0rph59, 2021-09-03:
I will close this issue now, please re-open it or create a new one if the problem persists.

closed 2021-09-03 by X3n0m0rph59