eruption-project/eruption #61: Please add support for Roccat Leadr.

dmnmsc, 2021-04-20:
Hi, 

I have a Roccat Leadr mouse. Feel free to ping me and request anything you need. 

Thank you very much.

X3n0m0rph59, 2021-04-21:
Hi @dmnmsc, thanks for opening this feature request! I am interested in adding support for this device to Eruption.

Since I don't have access to a `ROCCAT Leadr` at this time, it would be great if you could create a `*.pcapng` file of the packets generated by `ROCCAT Swarm` as it initializes the mouse to AIMO (RGB) mode.

You might want to have a look at the [Eruption Wiki](https://github.com/X3n0m0rph59/eruption/wiki) on how to create such a USB packet capture file.

Feel free to get back to me any time if you should require assistance!

dmnmsc, 2021-05-06:
Hi!

Can we livechat? I think I need a step-by-step for dummies. I have wireshark installed, a windows 10 VM machine running and usbmon module loaded.