eruption-project/eruption #62: Eruption don't detect device after Ubuntu sleep

poldolot, 2021-04-23:
Hi, I have a bug after waking Ubuntu 20.04 from sleep and connecting keyboard. It started after updating to v0.1.20

Corresponding log:
```systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
eruption[117814]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.21 () (release build)
eruption[117814]:  INFO  eruption > Loading saved state...
/usr/lib/gdm3/gdm-x-session[3808]: (II) systemd-logind: got resume for 13:80
eruption[117814]:  INFO  eruption > Registering plugins...
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
kernel: [75675.185157] input: Eruption Virtual Keyboard as /devices/virtual/input/input53
kernel: [75675.185399] input: Eruption Virtual Mouse as /devices/virtual/input/input54
/usr/lib/gdm3/gdm-x-session[2493]: (II) config/udev: Adding input device Eruption Virtual Mouse (/dev/input/mouse3)
/usr/lib/gdm3/gdm-x-session[2493]: (II) No input driver specified, ignoring this device.
/usr/lib/gdm3/gdm-x-session[2493]: (II) This device may have been added with another device file.
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
eruption[117814]:  INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
eruption[117814]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
eruption[117814]:  INFO  eruption                       > Plugins loaded and initialized successfully
eruption[117814]:  INFO  eruption                       > Enumerating connected devices...
/usr/lib/gdm3/gdm-x-session[3808]: (II) config/udev: Adding input device Eruption Virtual Mouse (/dev/input/mouse3)
/usr/lib/gdm3/gdm-x-session[3808]: (II) No input driver specified, ignoring this device.
/usr/lib/gdm3/gdm-x-session[3808]: (II) This device may have been added with another device file.
/usr/lib/gdm3/gdm-x-session[3808]: (II) systemd-logind: got resume for 13:68
/usr/lib/gdm3/gdm-x-session[2493]: (II) config/udev: Adding input device Eruption Virtual Keyboard (/dev/input/event10)
/usr/lib/gdm3/gdm-x-session[2493]: (**) Eruption Virtual Keyboard: Applying InputClass "libinput keyboard catchall"
/usr/lib/gdm3/gdm-x-session[2493]: (II) Using input driver 'libinput' for 'Eruption Virtual Keyboard'
/usr/lib/gdm3/gdm-x-session[3808]: (II) config/udev: Adding input device Eruption Virtual Keyboard (/dev/input/event10)
/usr/lib/gdm3/gdm-x-session[3808]: (**) Eruption Virtual Keyboard: Applying InputClass "libinput keyboard catchall"
/usr/lib/gdm3/gdm-x-session[3808]: (II) Using input driver 'libinput' for 'Eruption Virtual Keyboard'
eruption[117814]:  INFO  eruption                       > Device enumeration completed
eruption[117814]:  ERROR eruption                       > No supported devices found, exiting now
/usr/lib/gdm3/gdm-x-session[2493]: (II) systemd-logind: got fd for /dev/input/event10 13:74 fd 27 paused 1
/usr/lib/gdm3/gdm-x-session[2493]: (II) systemd-logind: releasing fd for 13:74
/usr/lib/gdm3/gdm-x-session[3808]: (II) systemd-logind: got fd for /dev/input/event10 13:74 fd 112 paused 0
/usr/lib/gdm3/gdm-x-session[3808]: (**) Eruption Virtual Keyboard: always reports core events
/usr/lib/gdm3/gdm-x-session[3808]: (**) Option "Device" "/dev/input/event10"
/usr/lib/gdm3/gdm-x-session[3808]: (**) Option "_source" "server/udev"
/usr/lib/gdm3/gdm-x-session[3808]: (II) event10 - Eruption Virtual Keyboard: is tagged by udev as: Keyboard
/usr/lib/gdm3/gdm-x-session[3808]: (II) event10 - Eruption Virtual Keyboard: device is a keyboard
/usr/lib/gdm3/gdm-x-session[3808]: (II) event10 - Eruption Virtual Keyboard: device removed
/usr/lib/gdm3/gdm-x-session[3808]: (**) Option "config_info" "udev:/sys/devices/virtual/input/input53/event10"
/usr/lib/gdm3/gdm-x-session[3808]: (II) XINPUT: Adding extended input device "Eruption Virtual Keyboard" (type: KEYBOARD, id 16)
/usr/lib/gdm3/gdm-x-session[3808]: (**) Option "xkb_model" "pc105"
/usr/lib/gdm3/gdm-x-session[3808]: (**) Option "xkb_layout" "pl"
/usr/lib/gdm3/gdm-x-session[3808]: (WW) Option "xkb_variant" requires a string value
/usr/lib/gdm3/gdm-x-session[3808]: (WW) Option "xkb_options" requires a string value
/usr/lib/gdm3/gdm-x-session[3808]: (II) config/udev: Adding input device Eruption Virtual Mouse (/dev/input/event11)
/usr/lib/gdm3/gdm-x-session[3808]: (**) Eruption Virtual Mouse: Applying InputClass "libinput pointer catchall"
/usr/lib/gdm3/gdm-x-session[3808]: (II) Using input driver 'libinput' for 'Eruption Virtual Mouse'
/usr/lib/gdm3/gdm-x-session[3808]: (EE) systemd-logind: failed to take device /dev/input/event11: No such device
/usr/lib/gdm3/gdm-x-session[3808]: (**) Eruption Virtual Mouse: always reports core events
/usr/lib/gdm3/gdm-x-session[3808]: (**) Option "Device" "/dev/input/event11"
/usr/lib/gdm3/gdm-x-session[3808]: (**) Option "_source" "server/udev"
/usr/lib/gdm3/gdm-x-session[3808]: (EE) client bug: Invalid path /dev/input/event11
/usr/lib/gdm3/gdm-x-session[3808]: (EE) libinput: Eruption Virtual Mouse: Failed to create a device for /dev/input/event11
/usr/lib/gdm3/gdm-x-session[3808]: (EE) PreInit returned 2 for "Eruption Virtual Mouse"
/usr/lib/gdm3/gdm-x-session[3808]: (II) UnloadModule: "libinput"
/usr/lib/gdm3/gdm-x-session[2493]: (II) config/udev: Adding input device (unnamed) (/dev/input/event11)
/usr/lib/gdm3/gdm-x-session[2493]: (**) (unnamed): Applying InputClass "libinput pointer catchall"
/usr/lib/gdm3/gdm-x-session[2493]: (II) Using input driver 'libinput' for '(unnamed)'
/usr/lib/gdm3/gdm-x-session[2493]: (EE) systemd-logind: failed to take device /dev/input/event11: No such device
/usr/lib/gdm3/gdm-x-session[2493]: (**) (unnamed): always reports core events
/usr/lib/gdm3/gdm-x-session[2493]: (**) Option "Device" "/dev/input/event11"
/usr/lib/gdm3/gdm-x-session[2493]: (**) Option "_source" "server/udev"
/usr/lib/gdm3/gdm-x-session[2493]: (EE) client bug: Invalid path /dev/input/event11
/usr/lib/gdm3/gdm-x-session[2493]: (EE) libinput: (unnamed): Failed to create a device for /dev/input/event11
/usr/lib/gdm3/gdm-x-session[2493]: (EE) PreInit returned 2 for "(unnamed)"
/usr/lib/gdm3/gdm-x-session[2493]: (II) UnloadModule: "libinput"
/usr/lib/gdm3/gdm-x-session[3808]: (II) config/udev: removing device Eruption Virtual Keyboard
/usr/lib/gdm3/gdm-x-session[3808]: (II) UnloadModule: "libinput"
/usr/lib/gdm3/gdm-x-session[3808]: (II) systemd-logind: releasing fd for 13:74
systemd[1]: eruption.service: Main process exited, code=exited, status=5/NOTINSTALLED
systemd[1]: eruption.service: Failed with result 'exit-code'.
```
I don't know if it's related but during normal use I have a lot of this in my log:
```eruption[12950]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
eruption[12950]:  WARN  eruption                               > This means that we dropped at least one frame
eruption[12950]:  WARN  eruption                               > Loop took: 155 milliseconds, goal: 50
eruption[12950]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
eruption[12950]:  WARN  eruption                               > This means that we dropped at least one frame
eruption[12950]:  WARN  eruption                               > Loop took: 161 milliseconds, goal: 50
```

After `systemctl restart eruption.service` everything is back to normal.

 * Eruption version 0.1.21
 * ROCCAT ROCCAT Vulcan 100 AIMO - ID 1e7d:307a

X3n0m0rph59, 2021-04-23:
Hi @poldolot, thank you for reporting this issue! I will build fixed packages soon...

X3n0m0rph59, 2021-04-24:
@poldolot I just published new packages (`eruption-git`) that _should_ fix the issue!

I could not easily reproduce the problem in a Ubuntu virtual machine. Will have to set up Ubuntu on bare-metal if the problem persists.

poldolot, 2021-04-26:
Hi, unfortunately version 0.1.22 didn't fixed the issue and now even replugging isn't working, only service restart helps.

New log:

```
kernel: [76807.925336] Freezing user space processes ... (elapsed 0.002 seconds) done.
kernel: [76807.927770] OOM killer disabled.
kernel: [76807.927770] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
kernel: [76807.929086] printk: Suspending console(s) (use no_console_suspend to debug)
kernel: [76808.024186] e1000e: EEE TX LPI TIMER: 00000011
kernel: [76810.094919] ACPI: EC: interrupt blocked
kernel: [76920.258000] ACPI: EC: interrupt unblocked
kernel: [76921.530240] nvidia 0000:01:00.0: Enabling HDA controller
kernel: [76921.850542] pcieport 0000:00:1c.0: PME: Spurious native interrupt!
kernel: [76921.850577] pcieport 0000:00:1c.0: PME: Spurious native interrupt!
kernel: [76921.891663] pcieport 0000:00:1c.0: PME: Spurious native interrupt!
kernel: [76922.312351] ata3: SATA link down (SStatus 4 SControl 300)
kernel: [76922.473445] hid-generic 0003:1E7D:307A.003D: offset (77) exceeds report_count (64)
kernel: [76922.473446] hid-generic 0003:1E7D:307A.003D: offset (76) exceeds report_count (64)
kernel: [76922.473448] hid-generic 0003:1E7D:307A.003D: offset (78) exceeds report_count (64)
kernel: [76923.343625] OOM killer enabled.
kernel: [76923.343627] Restarting tasks ... done.
kernel: [76923.371283] thermal thermal_zone9: failed to read out thermal zone (-61)
kernel: [76923.439251] PM: suspend exit
systemd-sleep[29681]: System resumed.
gdm3: GLib: Source ID 77 was not found when attempting to remove it
systemd[1]: Starting Eruption device hotplug notification daemon...
systemd[1]: Started Eruption device hotplug notification daemon.
eruption-hotplug-helper[29943]:  INFO  eruption_hotplug_helper > Waking up from system sleep...
eruption-hotplug-helper[29943]:  INFO  eruption_hotplug_helper > Waiting for the devices to settle...
eruption-hotplug-helper[29943]:  ERROR eruption_hotplug_helper > udevadm settle has failed: No such file or directory (os error 2)
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:76
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:75
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:71
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:74
systemd[1]: systemd-suspend.service: Succeeded.
systemd[1]: Finished Suspend.
systemd[1]: Stopped target Sleep.
systemd[1]: Reached target Suspend.
systemd[1]: Stopped target Suspend.
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:77
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:68
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:81
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:89
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:69
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:70
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:79
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:82
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:90
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:73
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:78
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:91
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:65
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 13:66
/usr/lib/gdm3/gdm-x-session[4353]: (II) systemd-logind: got resume for 226:0
/usr/lib/gdm3/gdm-x-session[4353]: (II) AIGLX: Resuming AIGLX clients after VT switch
/usr/lib/gdm3/gdm-x-session[4353]: (II) modeset(0): EDID vendor "LGD", prod id 1554
/usr/lib/gdm3/gdm-x-session[4353]: (II) modeset(0): Printing DDC gathered Modelines:
/usr/lib/gdm3/gdm-x-session[4353]: (II) modeset(0): Modeline "1920x1080"x0.0  138.60  1920 1968 2000 2080  1080 1083 1088 1111 +hsync -vsync (66.6 kHz eP)
/usr/lib/gdm3/gdm-x-session[4353]: (II) modeset(0): Modeline "1920x1080"x0.0  110.88  1920 1968 2000 2080  1080 1083 1088 1111 +hsync -vsync (53.3 kHz e)
/usr/lib/gdm3/gdm-x-session[4353]: (II) modeset(0): EDID vendor "LGD", prod id 1554
/usr/lib/gdm3/gdm-x-session[4353]: (II) modeset(0): Printing DDC gathered Modelines:
/usr/lib/gdm3/gdm-x-session[4353]: (II) modeset(0): Modeline "1920x1080"x0.0  138.60  1920 1968 2000 2080  1080 1083 1088 1111 +hsync -vsync (66.6 kHz eP)
/usr/lib/gdm3/gdm-x-session[4353]: (II) modeset(0): Modeline "1920x1080"x0.0  110.88  1920 1968 2000 2080  1080 1083 1088 1111 +hsync -vsync (53.3 kHz e)
/usr/lib/gdm3/gdm-x-session[4353]: (II) NVIDIA(G0): Setting mode "NULL"
/usr/lib/gdm3/gdm-x-session[4353]: (II) event9  - Video Bus: is tagged by udev as: Keyboard
/usr/lib/gdm3/gdm-x-session[4353]: (II) event9  - Video Bus: device is a keyboard
/usr/lib/gdm3/gdm-x-session[4353]: (II) event1  - Power Button: is tagged by udev as: Keyboard
/usr/lib/gdm3/gdm-x-session[4353]: (II) event1  - Power Button: device is a keyboard
/usr/lib/gdm3/gdm-x-session[4353]: (II) event2  - Sleep Button: is tagged by udev as: Keyboard
[... 29 lines cut ...]
/usr/lib/gdm3/gdm-x-session[4353]: (II) event14 - ROCCAT ROCCAT Vulcan 100 AIMO Consumer Control: device is a keyboard
[... cut ...]
eruption-hotplug-helper[29943]:    #033[35meruption-hotplug-helper/src/main.rs#033[0m:#033[35m155#033[0m
eruption-hotplug-helper[29943]: Backtrace omitted.
eruption-hotplug-helper[29943]: Run with RUST_BACKTRACE=1 environment variable to display it.
eruption-hotplug-helper[29943]: Run with RUST_BACKTRACE=full to include source snippets.
systemd[1]: eruption-hotplug-helper.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: eruption-hotplug-helper.service: Failed with result 'exit-code'.
```

X3n0m0rph59, 2021-09-13:
@poldolot Sorry, for not getting back to you earlier.

For hotplug and suspend/resume to work, the system needs to be in a state were the following command is able to run successfully:
```sh
 $ sudo eruption-hotplug-helper hotplug
```

Could you please try to run the following commands:
```sh
 $ sudo systemctl unmask eruption.service
 $ sudo systemctl enable --now eruption.service

 $ sudo systemctl unmask eruption-hotplug-helper.service
 $ sudo systemctl enable --now eruption-hotplug-helper.service
```

And the please run:
```sh
 $ sudo eruption-hotplug-helper hotplug
```

If eruption restarts successfully then suspend/resume has a great chance of working.

To debug the code that gets run during a suspend/resume cycle you could run these commands:

To simulate a suspend:
```sh
 $ sudo /lib/systemd/system-sleep/eruption pre
```

And to simulate a resume afterwards:
```sh
 $ sudo /lib/systemd/system-sleep/eruption
```

This should restart the Eruption daemon.

poldolot, 2021-09-13:
Ok, thx for the reply.
After running hotplug I get this:
```sudo eruption-hotplug-helper hotplug
 INFO  eruption_hotplug_helper > A hotplug event has been triggered, notifying the Eruption daemon...
 ERROR eruption_hotplug_helper > Could not determine whether the system is still booting: No such file or directory (os error 2)
```

X3n0m0rph59, 2021-11-20:
@poldolot Sorry, for taking so long to reply!

This issue should be fixed by commit 6d3a2b43b05ee8b05e1a601236a4c58c9afdbfb7. It will ship with Eruption `0.1.23`.

poldolot, 2021-12-20:
@X3n0m0rph59 no worries!
I'll wait :-)

X3n0m0rph59, 2022-02-25:
Hi @poldolot! 

I just published a new snapshot release. Please [find it here](https://github.com/X3n0m0rph59/eruption/releases/tag/v0.1.23-alpha). Could you please verify that it fixes the issue for you?

poldolot, 2022-02-25:
Hi @X3n0m0rph59!
This pre-release version didn't fixed my issue. Below is a log from unplugging and plugging in the keyboard.

```
kernel: [  615.117494] usb 3-2.3.2: USB disconnect, device number 13
message repeated 141 times: [ eruption[20054]: Error during notification of observers [keyboard_event]: timed out waiting on send operation]
acpid: input device has been disconnected, fd 10
eruption[20054]: Fatal: Keyboard device went away: No such device (os error 19)
/usr/lib/gdm3/gdm-x-session[9058]: (II) config/udev: removing device ROCCAT ROCCAT Vulcan 100 AIMO
/usr/lib/gdm3/gdm-x-session[9058]: (**) Option "fd" "94"
/usr/lib/gdm3/gdm-x-session[9058]: (II) event14 - ROCCAT ROCCAT Vulcan 100 AIMO: device removed
/usr/lib/gdm3/gdm-x-session[9058]: (II) UnloadModule: "libinput"
/usr/lib/gdm3/gdm-x-session[9058]: (II) systemd-logind: releasing fd for 13:78
acpid: input device has been disconnected, fd 13
/usr/lib/gdm3/gdm-x-session[9058]: (II) config/udev: removing device ROCCAT ROCCAT Vulcan 100 AIMO
/usr/lib/gdm3/gdm-x-session[9058]: (**) Option "fd" "102"
/usr/lib/gdm3/gdm-x-session[9058]: (II) event18 - ROCCAT ROCCAT Vulcan 100 AIMO: device removed
/usr/lib/gdm3/gdm-x-session[9058]: (II) UnloadModule: "libinput"
/usr/lib/gdm3/gdm-x-session[9058]: (II) systemd-logind: releasing fd for 13:82
eruption[20054]: Write error
/usr/lib/gdm3/gdm-x-session[9058]: (EE) systemd-logind: failed to release device: Device not taken
eruption[20054]: Could not finalize LEDs configuration: Write error
gnome-shell[9512]: Source ID 34277 was not found when attempting to remove it
gnome-shell[9512]: Could not set property Brightness on remote object /org/eruption/config: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.eruption was not provided by any .service files
/usr/lib/gdm3/gdm-x-session[9058]: (II) config/udev: removing device Eruption Virtual Mouse
/usr/lib/gdm3/gdm-x-session[9058]: (**) Option "fd" "136"
/usr/lib/gdm3/gdm-x-session[9058]: (II) event29 - Eruption Virtual Mouse: device removed
/usr/lib/gdm3/gdm-x-session[9058]: (II) UnloadModule: "libinput"
/usr/lib/gdm3/gdm-x-session[9058]: (II) systemd-logind: releasing fd for 13:93
acpid: input device has been disconnected, fd 29
/usr/lib/gdm3/gdm-x-session[9058]: (II) config/udev: removing device Eruption Virtual Keyboard
/usr/lib/gdm3/gdm-x-session[9058]: (**) Option "fd" "120"
/usr/lib/gdm3/gdm-x-session[9058]: (II) event28 - Eruption Virtual Keyboard: device removed
/usr/lib/gdm3/gdm-x-session[9058]: (II) UnloadModule: "libinput"
/usr/lib/gdm3/gdm-x-session[9058]: (II) systemd-logind: releasing fd for 13:92
systemd[20429]: eruption.service: Executable /usr/bin/rm missing, skipping: No such file or directory
systemd[1]: eruption.service: Succeeded.
kernel: [  619.689862] usb 3-2.3.2: new full-speed USB device number 14 using xhci_hcd
kernel: [  619.790341] usb 3-2.3.2: not running at top speed; connect to a high speed hub
kernel: [  619.791590] usb 3-2.3.2: New USB device found, idVendor=1e7d, idProduct=307a, bcdDevice= 1.00
kernel: [  619.791595] usb 3-2.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
kernel: [  619.791599] usb 3-2.3.2: Product: ROCCAT Vulcan 100 AIMO
kernel: [  619.791601] usb 3-2.3.2: Manufacturer: ROCCAT
kernel: [  619.811846] input: ROCCAT ROCCAT Vulcan 100 AIMO as /devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:02.0/0000:07:00.0/usb3/3-2/3-2.3/3-2.3.2/3-2.3.2:1.0/0003:1E7D:307A.0029/input/input99
kernel: [  619.871009] hid-generic 0003:1E7D:307A.0029: input,hidraw5: USB HID v1.11 Keyboard [ROCCAT ROCCAT Vulcan 100 AIMO] on usb-0000:07:00.0-2.3.2/input0
kernel: [  619.873321] input: ROCCAT ROCCAT Vulcan 100 AIMO Mouse as /devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:02.0/0000:07:00.0/usb3/3-2/3-2.3/3-2.3.2/3-2.3.2:1.1/0003:1E7D:307A.002A/input/input100
kernel: [  619.930373] input: ROCCAT ROCCAT Vulcan 100 AIMO Consumer Control as /devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:02.0/0000:07:00.0/usb3/3-2/3-2.3/3-2.3.2/3-2.3.2:1.1/0003:1E7D:307A.002A/input/input101
kernel: [  619.930650] input: ROCCAT ROCCAT Vulcan 100 AIMO System Control as /devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:02.0/0000:07:00.0/usb3/3-2/3-2.3/3-2.3.2/3-2.3.2:1.1/0003:1E7D:307A.002A/input/input102
kernel: [  619.931209] hid-generic 0003:1E7D:307A.002A: input,hiddev2,hidraw6: USB HID v1.11 Mouse [ROCCAT ROCCAT Vulcan 100 AIMO] on usb-0000:07:00.0-2.3.2/input1
kernel: [  619.932231] input: ROCCAT ROCCAT Vulcan 100 AIMO as /devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:02.0/0000:07:00.0/usb3/3-2/3-2.3/3-2.3.2/3-2.3.2:1.2/0003:1E7D:307A.002B/input/input104
kernel: [  619.990608] hid-generic 0003:1E7D:307A.002B: input,hidraw7: USB HID v1.11 Keypad [ROCCAT ROCCAT Vulcan 100 AIMO] on usb-0000:07:00.0-2.3.2/input2
kernel: [  619.991912] input: ROCCAT ROCCAT Vulcan 100 AIMO as /devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:02.0/0000:07:00.0/usb3/3-2/3-2.3/3-2.3.2/3-2.3.2:1.3/0003:1E7D:307A.002C/input/input105
kernel: [  620.051257] hid-generic 0003:1E7D:307A.002C: input,hidraw8: USB HID v1.11 Device [ROCCAT ROCCAT Vulcan 100 AIMO] on usb-0000:07:00.0-2.3.2/input3
mtp-probe: checking bus 3, device 14: "/sys/devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:02.0/0000:07:00.0/usb3/3-2/3-2.3/3-2.3.2"
[... cut ...]
/usr/lib/gdm3/gdm-x-session[9058]: (II) event15 - ROCCAT ROCCAT Vulcan 100 AIMO Mouse: device is a pointer
systemd[1]: Starting Eruption device hotplug notification daemon...
systemd[1]: Started Eruption device hotplug notification daemon.
systemd[1]: eruption-hotplug-helper.service: Succeeded.
boltd[2211]: probing: timeout, done: [2910091] (2000000)
```

and the message from hotplug:
```
INFO  eruption_hotplug_helper > A hotplug event has been triggered, notifying the Eruption daemon...
INFO  eruption_hotplug_helper > System is still booting, skipping restart of Eruption
```

X3n0m0rph59, 2022-02-25:
@poldolot, thanks for the quick reply!

It seems like the `eruption-hotplug-helper.service` is being executed now, but there is another problem happening:

> INFO  eruption_hotplug_helper > System is still booting, skipping restart of Eruption

This normally only happens due to the system exhibiting some hanging or failed Systemd units. Could you please post the output of this command:

```sh
sudo systemctl list-units --failed
```

I believe that as soon as the failed units are fixed, there is a great chance for suspend/resume to now work correctly!

poldolot, 2022-02-25:
@X3n0m0rph59, here is an output for failed services:
```
  UNIT                   LOAD   ACTIVE SUB    DESCRIPTION              
● plymouth-start.service loaded failed failed Show Plymouth Boot Screen
```
and You'r right, after `sudo systemctl reset-failed` the hotplug works!
But unfortunately after waking system from sleep the eruption.service failed:
```
  UNIT             LOAD   ACTIVE SUB    DESCRIPTION                                                               
● eruption.service loaded failed failed Linux user-mode input and LED driver for keyboards, mice and other devices
```

and corresponding systemlog:
```
systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
eruption[45339]: LPP-pburcza eruption[45339]: No supported devices found, exiting now
systemd[1]: eruption.service: Main process exited, code=exited, status=5/NOTINSTALLED
systemd[45371]: eruption.service: Executable /usr/bin/rm missing, skipping: No such file or directory
systemd[1]: eruption.service: Failed with result 'exit-code'.
```

X3n0m0rph59, 2022-02-25:
@poldolot, this looks strange to me. Not completely sure what is going on...

> systemd[1]: eruption.service: Main process exited, code=exited, status=5/NOTINSTALLED
> systemd[45371]: eruption.service: Executable /usr/bin/rm missing, skipping: No such file or directory

Could you please try to run:
```sh
sudo systemctl stop eruption.service

sudo systemctl daemon-reload
systemctl --user daemon-reload

sudo systemctl restart eruption.service
```

And then please test again whether device hotplug is working now.

poldolot, 2022-02-25:
@X3n0m0rph59, no change.
Result of `systemctl status eruption.service`
```
● eruption.service - Linux user-mode input and LED driver for keyboards, mice and other devices
     Loaded: loaded (/lib/systemd/system/eruption.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-02-25 14:09:51 CET; 4s ago
       Docs: man:eruption(8)
             man:eruption.conf(5)
             man:eruptionctl(1)
             man:eruption-netfx(1)
    Process: 53196 ExecStart=/usr/bin/eruption -c /etc/eruption/eruption.conf (code=exited, status=5)
    Process: 53256 ExecStopPost=/usr/bin/rm -f /run/eruption/eruption.pid (code=exited, status=0/SUCCESS)
   Main PID: 53196 (code=exited, status=5)

lut 25 14:09:49 systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
lut 25 14:09:49 systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
lut 25 14:09:51 eruption[53196]: eruption[53196]: No supported devices found, exiting now
lut 25 14:09:51 systemd[1]: eruption.service: Main process exited, code=exited, status=5/NOTINSTALLED
lut 25 14:09:51 systemd[53256]: eruption.service: Executable /usr/bin/rm missing, skipping: No such file or directory
lut 25 14:09:51 systemd[1]: eruption.service: Failed with result 'exit-code'.
```

X3n0m0rph59, 2022-02-25:
@poldolot, maybe the Eruption daemon is racing against system wakeup and hardware initialization. It seems to me that the USB device is not ready yet, when Eruption tries to enumerate and subsequently bind to it.

Could you please temporarily add the line `sleep 3` to ` /usr/lib/systemd/system-sleep/eruption` so that it looks like this:

```bash
#!/bin/sh

if [ "$1" = "pre" ] ; then
    # prepare Eruption for system sleep
    touch /run/lock/eruption-hotplug-helper.lock

    systemctl stop eruption-hotplug-helper.service
	systemctl stop eruption.service

    touch /run/lock/eruption-sleep.lock
else
    sleep 3

    # wake up Eruption after system sleep
    rm /run/lock/eruption-hotplug-helper.lock
    systemctl start eruption-hotplug-helper.service
    rm /run/lock/eruption-sleep.lock
fi
```

and then please try again if it works now. Thanks for your patience!

poldolot, 2022-02-28:
@X3n0m0rph59, I've tested every scenario I could think off, and it works in most of them but when the system is waked up without keyboard connected it doesn't work. The error is the same as above.
btw. I don't have file `/usr/lib/systemd/system-sleep/eruption` in my system but there is `/lib/systemd/system-sleep/eruption` and I've added `sleep 3` command there.

X3n0m0rph59, 2022-02-28:
@poldolot, thanks for testing this so thoroughly!
So did I understand correctly, that resume from suspend with the keyboard connected works now as intended?

Do you use some kind of KVM switch, so that the keyboard could be disconnected during resume?
If the keyboard is __disconnected__ during resume from suspend, the `eruption` daemon aborts with error code 5 (no devices detected) and subsequently Systemd marks the unit as failed. This in turn prohibits `eruption-hotplug-helper` from restarting the daemon on the next hotplug event.

So hotplug after resume should work when we do something like this:

```sh
#!/bin/sh

if [ "$1" = "pre" ] ; then
    # prepare Eruption for system sleep
    touch /run/lock/eruption-hotplug-helper.lock

    systemctl stop eruption-hotplug-helper.service
	systemctl stop eruption.service

    touch /run/lock/eruption-sleep.lock
else
    sleep 3

    # wake up Eruption after system sleep
    rm /run/lock/eruption-hotplug-helper.lock
    systemctl start eruption-hotplug-helper.service
    rm /run/lock/eruption-sleep.lock

    sleep 3
    systemctl reset-failed
fi
```

This line should cause the system to be in a state where the `eruption-hotplug-helper` is able to restart the `eruption` daemon on the next hotplug event:

```sh
    systemctl reset-failed
```

To test whether this theory could be correct, could you please resume from suspend with the keyboard __disconnected__ and then hotplug the device. Please be sure to wait for at least 10 seconds after the resume operation completed, before hotplugging the device, so that everything has settled.

poldolot, 2022-02-28:
@X3n0m0rph59, yes, it helps!
I'm using Dell WD19TB docking station but it doesn't matter where the keyboard is connected, it behaves the same.
Now everything is working, but there is a repeating message in log file:
```
eruption[47383]: Error during notification of observers [keyboard_event]: timed out waiting on send operation
```

X3n0m0rph59, 2022-02-28:
@poldolot, okay great! I will think about this some more and then try to come up with a robust solution to this issue.
Maybe I have to change the exit code of the `eruption` daemon in case no device is found to signal success.

Regarding the log messages: This problem should be corrected by enabling the user session daemons (the audio-proxy) that ship with eruption:

```sh
systemctl --user enable --now eruption-audio-proxy.service eruption-process-monitor.service 
```

And please be sure to install the latest version of the package/update your system.

poldolot, 2022-03-01:
@X3n0m0rph59, I've installed newest version but this command gave me an error:
```
Failed to enable unit: Unit file eruption-audio-proxy.service does not exist.
```

X3n0m0rph59, 2022-03-01:
@poldolot Did you try without `sudo` in front of the command and with `--user` parameter? 

```sh
systemctl --user daemon-reload
systemctl --user enable --now eruption-audio-proxy.service
```

If it still wont work could you please locate the file `eruption-audio-proxy.service` on your system. It should be placed in `/usr/lib/systemd/user/eruption-audio-proxy.service`

poldolot, 2022-03-01:
@X3n0m0rph59, yes I've run it without `sudo`. I've this file here:
```
/lib/systemd/user/eruption-audio-proxy.servic
```
maybe I need to uninstall eruption and install it from scratch?

X3n0m0rph59, 2022-03-05:
@poldolot I have published a new `eruption-git` package for Ubuntu. It ships with multiple bugfixes that are relevant to your issue. Could you please try out whether suspend / resume now works even without the keyboard being detected?

X3n0m0rph59, 2022-03-05:
@poldolot Yes please backup your profiles and then purge and reinstall `eruption-git`.

```sh
sudo apt purge eruption-git
sudo apt update
sudo apt install eruption-git
```

poldolot, 2022-03-07:
@X3n0m0rph59, I've purged and install newest version but still no success without keyboard connected and eruption files are still in /lib/... and not /usr/lib/...

Cybugs, 2022-03-19:
I had a similar problem on my Manjaro Box with kernel 5.16.
After resuming the eruption daemon failed to start. Adding
    sleep 3
    systemctl reset-failed
to /usr/lib/systemd/system-sleep/eruption solved the issue.

Just for your information. Perhaps it helps to solve the issue.

X3n0m0rph59, 2023-04-16:
@poldolot @Cybugs

I have pushed a number of fixes to the `develop` branch that should hugely improve the behavior of Eruption on suspend/resume. Eruption no longer depends on a custom Systemd suspend script. All devices are initialized now via Udev hotplug events, during system resume. There are still some known problems: The `ROCCAT Kone Pure Ultra` does not get re-initialized correctly at the moment... So it is not ready for `master` just yet.

poldolot, 2023-04-17:
@X3n0m0rph59 that's great news! I'll wait for stable version. Thank You for Your insane work :-)