eruption-project/eruption #63: Double Clicking Mouse

man-exists, 2021-04-25:
I have a roccat kone pure ultra and I can't find a way to create a double click macro.

Is there a way i can do that?

X3n0m0rph59, 2021-07-21:
Hi @man-exists, I am sorry for taking so long to respond!

The requested feature will be shipped with Eruption `0.1.22`.
In the meantime you could extend your local `/usr/share/eruption/scripts/lib/macros/user-macros.lua` with the following example code:

```Lua
function macro_inject_double_click_left()
    debug("UserMacros: macro_inject_double_click_left")

    macro_inject_double_click(1)
end

-- Inject a double click with the specified mouse button
function macro_inject_double_click(button_index)
    if MODIFIER_KEY ~= RIGHT_MENU then
        inject_key(MODIFIER_KEY_EV_CODE, false)  -- modifier key up
    end

    consume_key()

    inject_mouse_button(button_index, true)
    inject_mouse_button(button_index, false)

    delay(50)

    inject_mouse_button(button_index, true)
    inject_mouse_button(button_index, false)
end
```

man-exists, 2021-07-22:
Thank you so much!

closed 2021-08-18 by X3n0m0rph59