eruption-project/eruption #64: Problem setting up audio input in ubuntu 21.04

derEremit, 2021-04-28:
I see no output with
pactl list source-outputs

cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=21.04
DISTRIB_CODENAME=hirsute
DISTRIB_DESCRIPTION="Ubuntu 21.04"

ls -lha /tmp/pulse-server
srwxrwxrwx 1 zen zen 0 dub 28 20:49 /tmp/pulse-server

[default.pa.txt](https://github.com/X3n0m0rph59/eruption/files/6394340/default.pa.txt)
[client.conf.txt](https://github.com/X3n0m0rph59/eruption/files/6394349/client.conf.txt)
[eruption.txt](https://github.com/X3n0m0rph59/eruption/files/6394363/eruption.txt)

dub 28 20:48:59 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:48:59 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:00 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:00 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:00 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:00 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:02 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:02 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:03 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:03 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:03 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:03 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:04 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:04 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:10 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
dub 28 20:49:10 flame eruption[1822]:  ERROR eruption::plugins::audio                         > Pulse Audio error: Could not open Pulse Audio: Connection refused

the-unknown, 2021-05-21:
This could be the same error as described here:
https://www.reddit.com/r/linux_gaming/comments/5tidzc/a_solution_for_pulseaudio_pa_context_connect/

There it says:

> I spent a long time searching and trying different things before determining my issue (and I expect it to be common), came from needing to run qemu-system-x86_64 as root while pulseaudio's daemon was run as my user. Because of this, root is unable to find the pa socket for my user(by default at `/run/user/$ID/pulse/native`. If you only have one user $ID is probably 1000. The root user will also need access to the cookie generated for your user. Mine was located at `~/.config/pulse/cookie`.

The eruption deamon runs as root and that might be the reason why eruption cannot connect to the Pulse Audio of the user.

X3n0m0rph59, 2021-09-02:
Hi @derEremit, sorry for taking so long to reply.

> I see no output with
pactl list source-outputs

I believe I finally found the cause for this problem. You need to switch to a profile that triggers the audio grabber thread to start. The `Eruption Audio Grabber` needs to be running for this to work. Please switch to a profile that uses the audio API of Eruption. 
```sh
 $ eruptionctl switch profile spectrum-analyzer-swirl.profile
```
 
I will update the documentation respectively.

X3n0m0rph59, 2021-09-09:
I will close this issue now. Please reopen it if the problem persists!

closed 2021-09-09 by X3n0m0rph59