eruption-project/eruption #66: Ubuntu 20.04 - Window Name not detected

the-unknown, 2021-05-21:
**Introduction**
Hey, so I recently reinstalled my Ubuntu 20.04. I also reinstalled the newest eruption-git version. 

**Please describe the bug**
The `eruption_process_monitor` does not detect a change of the window name. However that worked well right before the reinstallation of Ubuntu.

**To Reproduce**
1. Setup a fresh Ubuntu 20.0
2. Install eruption-git
3. Enable `eruption process monitor` by using the following command: `systemctl --user enable --now eruption-process-monitor.service`
4. Add a new rule `eruption-process-monitor rules add window-name '.*YouTube.*Google Chrome' /var/lib/eruption/profiles/spectrum-analyzer-swirl.profile`
5. Open YouTube in Chrome Browser - nothing happens to the LEDs of the keyboard.

**Expected behavior**
The profile should switch

**Logs**
Please provide logs: `sudo journalctl -u eruption.service -b`
```
-- Logs begin at Wed 2021-05-12 01:41:01 CEST, end at Fri 2021-05-21 19:45:58 CEST. --
Mai 21 19:20:15 pix3lbox systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Mai 21 19:20:15 pix3lbox systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.22 () (release build)
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption > Loading saved state...
Mai 21 19:20:15 pix3lbox eruption[4972]:  WARN  eruption > Could not parse state file: Could not load global runtime state: configuration file "/var/lib/eruption/eruption.state" not found
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption > Registering plugins...
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Mai 21 19:20:15 pix3lbox eruption[4972]:  WARN  eruption                       > Could not load persisted state: No such file or directory (os error 2)
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption                       > Plugins loaded and initialized successfully
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption                       > Enumerating connected devices...
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2e2c (0003:0006:00) - ROCCAT ROCCAT Kone Aimo 16K
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::hwdevices::roccat_kone_aimo_remastered > Bound driver: ROCCAT Kone Aimo Remastered
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::hwdevices                              > Found supported keyboard device: 0x1e7d:0x3098 (0003:0007:00) - ROCCAT ROCCAT Vulcan AIMO
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption::hwdevices::roccat_vulcan_1xx           > Bound driver: ROCCAT Vulcan 100/12x AIMO
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption                                         > Opening keyboard device...
Mai 21 19:20:15 pix3lbox eruption[4972]:  INFO  eruption                                         > Initializing keyboard device...
Mai 21 19:20:15 pix3lbox systemd[1]: Stopping Linux user-mode input and LED driver for keyboards, mice and other devices...
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Configuring keyboard LEDs...
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Firmware revision: 136
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Spawning keyboard input thread...
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Opening mouse device...
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Now listening on keyboard: /dev/input/event11
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Physical location: usb-0000:00:14.0-14/input0
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Grabbing the keyboard device exclusively
Mai 21 19:20:16 pix3lbox eruption[4972]:  INFO  eruption                                         > Initializing mouse device...
Mai 21 19:20:17 pix3lbox eruption[4972]:  ERROR eruption::hwdevices::roccat_kone_aimo_remastered > Step 1: Invalid result
Mai 21 19:20:17 pix3lbox eruption[4972]:  INFO  eruption                                         > Configuring mouse LEDs...
Mai 21 19:20:17 pix3lbox eruption[4972]:  INFO  eruption                                         > Firmware revision: 103
Mai 21 19:20:17 pix3lbox eruption[4972]:  INFO  eruption                                         > Spawning mouse input thread...
Mai 21 19:20:17 pix3lbox eruption[4972]:  INFO  eruption                                         > Device enumeration completed
Mai 21 19:20:17 pix3lbox eruption[4972]:  INFO  eruption                                         > Performing late initializations...
Mai 21 19:20:17 pix3lbox eruption[4972]:  INFO  eruption                                         > Initializing D-Bus API...
Mai 21 19:20:17 pix3lbox eruption[4972]:  INFO  eruption                                         > Late initializations completed
Mai 21 19:20:17 pix3lbox eruption[4972]:  INFO  eruption                                         > Startup completed
```

**Please complete the following information:**

 * Eruption version = eruption-git - 0.1.22-ubuntu11 
 * Your device(s) (`sudo lsusb`)
 ```
Bus 003 Device 006: ID 1e7d:2e2c ROCCAT ROCCAT Kone Aimo 16K
Bus 003 Device 007: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO
```

the-unknown, 2021-05-21:
This can be closed.
The solution was:

I edit the file `/etc/eruption/process-monitor.conf`:
```
# Eruption - process-monitor
# Main configuration file

[global]

[X11]
display = ":1"
```
and changed the last line to read `display = ":0"`.

Now it works!

X3n0m0rph59, 2021-07-18:
@the-unknown, thanks for posting the solution to this problem!

closed 2021-07-18 by X3n0m0rph59