eruption-project/eruption #67: Compilation error

bthiebault, 2021-06-01:
I'm trying to install eruption but I have a compilation error:
```
$ yay -Syu eruption
[...]
Compiling tiff v0.6.1
error[E0658]: use of unstable library feature 'str_strip': newly added
   --> /home/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/sysinfo-0.17.2/src/linux/system.rs:764:34
    |
764 |     if let Some(name) = parts[1].strip_prefix("(") {
    |                                  ^^^^^^^^^^^^
    |
    = note: see issue #67302 <https://github.com/rust-lang/rust/issues/67302> for more information

error[E0658]: use of unstable library feature 'str_strip': newly added
   --> /home/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/sysinfo-0.17.2/src/linux/system.rs:801:42
    |
801 |             if let Some(stripped) = line.strip_prefix(info_str) {
    |                                          ^^^^^^^^^^^^
    |
    = note: see issue #67302 <https://github.com/rust-lang/rust/issues/67302> for more information

error[E0658]: use of unstable library feature 'str_strip': newly added
   --> /home/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/sysinfo-0.17.2/src/linux/system.rs:818:38
    |
818 |         if let Some(stripped) = line.strip_prefix(info_str) {
    |                                      ^^^^^^^^^^^^
    |
    = note: see issue #67302 <https://github.com/rust-lang/rust/issues/67302> for more information

error: aborting due to 3 previous errors

For more information about this error, try `rustc --explain E0658`.
error: could not compile `sysinfo`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
==> ERROR: A failure occurred in build().
    Aborting...
error making: eruption-git
```

I have the same error with the eruption and the eruption-git providers

X3n0m0rph59, 2021-07-18:
HI @bthiebault, this error may be caused by an outdated version of the Rust compiler. Please try to update it using `rustup update`. You need at least Rust `1.53.0` to successfully compile Eruption.

bthiebault, 2021-07-29:
Thanks, updating rust fixed the compilation error

closed 2021-07-29 by bthiebault