eruption-project/eruption #70: Increasing volume above 100% leads to deactivated backlight

follwersion, 2021-06-17:
Hi, first: I LOVE this! Thanks a bunch!

Im on:
Arch Manajro Linux 5.10.42-1-MANJARO
Cinnamon 4.8.6
with
Eruption 0.1.22 (0.1.21.r26.gb3bd27e) (release build)
and 
Vulcan 121 AIMO (keyboard device: 0x1e7d:0x3098 (0001:0002:00))

The bug appears when i allow the sound output device to be leveled up to 125% volume (see screenshot) - and increase it with the volume control on the keyboard.
The white backlight increases from the left to the right up to 100% and then turns completely off.

Switching the profile takes everything back to normal...

I would expect the volume display to stay active above the 100% volume, or even to scale with the max. level from the output device.

Logs:
https://pastebin.com/vz9zg7Yv

[image]

Not a big thing - but wanted to share!

X3n0m0rph59, 2021-07-18:
Hi @follwersion, thank you very much for reporting this bug! I will try to ship a fix for this issue with Eruption `0.1.22`.

X3n0m0rph59, 2021-08-18:
Version 0.1.22 should fix this issue. Please re-open the bug if the problem persists.

closed 2021-08-18 by X3n0m0rph59

follwersion, 2021-08-18:
Great work thank you very much!!! Beer is on the way ;-)