eruption-project/eruption #75: Dim a specific zone of the canvas (e.g. if the mouse LEDs are too bright)

X3n0m0rph59, 2021-07-18:
### Discussed in https://github.com/X3n0m0rph59/eruption/discussions/74

<div type='discussions-op-text'>

<sup>Originally posted by **Cybugs** July 16, 2021</sup>
Hi,

does someone know how to disable or dim the LED of this mouse? You really need sunglasses if you take your hand of it...

Thanks in advance!</div>

X3n0m0rph59, 2021-07-18:
To dim/darken a specific zone on the canvas, we may create a semi-transparent overlay that simply renders a black area. For this approach to work, it is crucial that the overlay is rendered after the `.lua` scripts that should be dimmed. This solution will be included in the release `0.1.22` of Eruption.

File `dim-zone.lua`:
```lua
require "declarations"
require "debug"

-- global state variables --
ticks = 0
color_map = {}

-- event handler functions --
function on_startup(config)
    for i = 0, canvas_size do
        color_map[i] = 0x00000000
    end

    for i = zone_start, zone_end do
        color_map[i] = rgba_to_color(0, 0, 0, 255 * opacity)
    end

    submit_color_map(color_map)
end

function on_apply_parameter(parameter, value)
	local update_fn = load("" .. parameter .. " = " .. value)

	update_fn()

	-- update state
	on_startup(nil)
end
```

File `dim-zone.lua.manifest`:
```toml
name = "Dim Zone"
description = "Dim (darken) a zone on the canvas"
version = "0.0.1"
author = "The Eruption development team"
min_supported_version = "0.1.22"
tags = ['Background', 'Vendor', 'Filter']

[[config]]
type = 'int'
name = 'zone_start'
description = 'Zone on the canvas to dim (start index)'
min = 0
max = 180
default = 144

[[config]]
type = 'int'
name = 'zone_end'
description = 'Zone on the canvas to dim (end index)'
min = 0
max = 180
default = 180

[[config]]
type = 'float'
name = 'opacity'
description = 'Opacity value in the range 0.0 .. 1.0, where 1.0 is fully dimmed'
min = 0.0
max = 1.0
default = 0.8
``` 

And as an example, a .profile file using this:

```toml
id = '5dc62fa6-e965-45cb-a0da-e87d29713119'
name = 'Color Swirls (Perlin): Rainbow'
description = 'Color Swirl effect'
active_scripts = [
    'swirl-perlin.lua',
    'halo.lua',
    'shockwave.lua',
#   'impact.lua',
#   'water.lua',
#   'raindrops.lua',
#   'sysmon.lua',
    'batique.lua',
# uncomment the following line to dim the mouse LEDs
#   'dim-zone.lua',
    'macros.lua',
#   'stats.lua',
]

[[config."Perlin Swirl"]]
type = 'float'
name = 'color_divisor'
value = 1.0
default = 1.0

[[config."Perlin Swirl"]]
type = 'float'
name = 'color_offset'
value = 0.0
default = 0.0

[[config."Perlin Swirl"]]
type = 'float'
name = 'time_scale'
value = 150.0
default = 150.0

[[config."Perlin Swirl"]]
type = 'float'
name = 'coord_scale'
value = 14.0
default = 14.0

[[config.Shockwave]]
type = 'color'
name = 'color_step_shockwave'
value = 0x05010000
default = 0x05010000

[[config.Shockwave]]
type = 'bool'
name = 'mouse_events'
value = true
default = true

[[config.Shockwave]]
type = 'color'
name = 'color_mouse_click_flash'
value = 0xa0ff0000
default = 0xa0ff0000

[[config.Shockwave]]
type = 'color'
name = 'color_mouse_wheel_flash'
value = 0xd0ff0000
default = 0xd0ff0000

[[config.Raindrops]]
type = 'float'
name = 'opacity'
value = 0.75
default = 0.75

[[config."System Monitor"]]
type = 'color'
name = 'color_cold'
value = 0x0000ff00
default = 0x0000ff00

[[config."System Monitor"]]
type = 'color'
name = 'color_hot'
value = 0xffff0000
default = 0xffff0000

# mouse support
[[config.Batique]]
type = 'int'
name = 'zone_start'
value = 144
default = 144

[[config.Batique]]
type = 'int'
name = 'zone_end'
value = 180
default = 180

[[config.Batique]]
type = 'float'
name = 'coord_scale'
value = 0.5
default = 0.5

# dim a specific zone, e.g. if the mouse LEDs are too bright
[[config."Dim Zone"]]
type = 'int'
name = 'zone_start'
value = 144
default = 144

[[config."Dim Zone"]]
type = 'int'
name = 'zone_end'
value = 180
default = 180

[[config."Dim Zone"]]
type = 'float'
name = 'opacity'
value = 0.95
default = 0.95
```

X3n0m0rph59, 2021-08-18:
The problem should be circumvented in the current release `0.1.22`. Starting with Eruption version `0.1.23` we will most likely ship per-device brightness controls. Please re-open this issue if the problem is not solved for you.

closed 2021-08-18 by X3n0m0rph59

Cybugs, 2021-08-21:
Thanks a lot. Just tried the new feature and dim-zone.lua works great. With opacity=0.95 the LED on my Kone Ultra is finally eye-friendly.