eruption-project/eruption #76: Massive input lag ("Jitter") on Fedora 34 x86_64 (i3/X11)

sellerie98, 2021-07-19:
I am experiencing massive input lag ("jitter") issues on Fedora 34 (5.12.15-300.fc34.x86_64) with the latest locally built master branch (commit ID 98342c75cf3e13188a07a7adb30a7e80e9572559). My device is the 1e7d:3098 (Vulcan 121 AIMO).

Example:
I'm typing "normally" and at some point the typing starts being very laggy and will come to a halt.
After a few seconds it will start putting out the typed characters again in the correct order, but slowly, like a buffer is being flushed.
Also the CPU usage is spiking visibly the faster I type and the more input has to be parsed.
The issue goes away as soon as I stop the eruption process.
This is happening in any kind of text input field e.g. terminal or the webbrowser and I am having the same behavior on both my desktop and laptop.
This issue is *not* occuring on Debian 10 (neither buster kernel, nor with 5.10 kernel from backports) at all, only on Fedora.

Eruption's journalctl log is spammed with messages like these with the jitter value sometimes going up to more than 900ms:

```
Mai 20 19:44:35 Cats eruption[39858]:  WARN  eruption                               > Loop took: 600 milliseconds, goal: 50
Mai 20 19:44:58 Cats eruption[39858]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
Mai 20 19:44:58 Cats eruption[39858]:  WARN  eruption                               > This means that we dropped at least one frame
Mai 20 19:44:58 Cats eruption[39858]:  WARN  eruption                               > Loop took: 237 milliseconds, goal: 50
Mai 20 19:45:01 Cats eruption[39858]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
Mai 20 19:45:01 Cats eruption[39858]:  WARN  eruption                               > This means that we dropped at least one frame
Mai 20 19:45:01 Cats eruption[39858]:  WARN  eruption                               > Loop took: 137 milliseconds, goal: 50
Mai 20 19:45:12 Cats eruption[39858]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
Mai 20 19:45:12 Cats eruption[39858]:  WARN  eruption                               > This means that we dropped at least one frame
Mai 20 19:45:12 Cats eruption[39858]:  WARN  eruption                               > Loop took: 245 milliseconds, goal: 50
Mai 20 19:45:21 Cats eruption[39858]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
Mai 20 19:45:21 Cats eruption[39858]:  WARN  eruption                               > This means that we dropped at least one frame
Mai 20 19:45:21 Cats eruption[39858]:  WARN  eruption                               > Loop took: 200 milliseconds, goal: 50
Mai 20 19:45:21 Cats eruption[39858]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
Mai 20 19:45:21 Cats eruption[39858]:  WARN  eruption                               > This means that we dropped at least one frame
Mai 20 19:45:21 Cats eruption[39858]:  WARN  eruption                               > Loop took: 200 milliseconds, goal: 50
Mai 20 19:45:23 Cats eruption[39858]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
Mai 20 19:45:23 Cats eruption[39858]:  WARN  eruption                               > This means that we dropped at least one frame
Mai 20 19:45:23 Cats eruption[39858]:  WARN  eruption                               > Loop took: 200 milliseconds, goal: 50
Mai 20 19:45:24 Cats eruption[39858]:  WARN  eruption                               > More than 82 milliseconds of jitter detected!
Mai 20 19:45:24 Cats eruption[39858]:  WARN  eruption                               > This means that we dropped at least one frame
Mai 20 19:45:24 Cats eruption[39858]:  WARN  eruption                               > Loop took: 247 milliseconds, goal: 50
```

I tested this on multiple profiles and even on a profile that only consists of the solid colour white and nothing else:

```
id = 'f3ca11a2-4c1d-4c6b-a08b-1ab768e0fa9f'
name = 'solid white'
description = 'Does nothing but display white'
active_scripts = [
    'solid.lua',
]

[[config."Solid Color"]]
type = 'color'
name = 'color_background'
value = 0xffffffff
default = 0xffffffff

[[config."Solid Color"]]
type = 'float'
name = 'opacity'
value = 0.45
default = 0.45

```

sellerie98, 2021-07-19:
BTW: I also created this issue on Gitlab 2 months ago and haven't received any reaction so far sadly, hence my creation of this issue on github now. Sorry if you feel bothered by double-posting.

X3n0m0rph59, 2021-07-20:
Hi @sellerie98, thank you for reporting this issue! I am really sorry that I did not respond to your bug report on GitLab. I somehow missed the notification E-Mail for it.

The problem you describe clearly seems to be a bug. I did go to great lengths to keep the input lag as small as possible, when using Eruption. Being a user-space driver it is not completely possible to have no input lag at all since we need one extra round trip from the kernel to the Eruption daemon and then back to the kernel (via the `uinput` subsystem). But the input-lag should be in the range of microseconds and by no means should you be able to notice it.

Do you see the same buffer flushing issue with your mouse device too?
Could you please show the output of the Eruption daemon running with the DEBUG or even TRACE log-level?

```sh
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 $ sudo RUST_LOG=debug eruption
 $ sudo RUST_LOG=trace eruption
```

does it make a difference if you execute Eruption with realtime scheduling class?

```sh
 $ sudo RUST_LOG=debug chrt 20 eruption
```

In the meantime, I will set up a VM running Fedora 34 / i3 on X11... Hope that I will be able to reproduce this issue.

sellerie98, 2021-07-20:
Hi,

no problem! I just assumed you don't read gitlab issues so I recreated it here.
I saw that on Debian! I was actually surprised that the input delay was *that* low in the debian VM so that I really could not see from the delay whether eruption was running or not. Quite an achievement for a userspace application IMO.

The mouse device doesnt have a change of behaviour at all at least in regards of cursor movement (I havent tested mouse buttons).

Log from debug:
```
root@Cats:~ ! RUST_LOG=debug eruption -c /etc/eruption/eruption.conf

 <I cut out the license notice and disclaimer>

 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.22 (0.1.21.r36.g98342c7) (release build)
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported device: 0x2516:0x51 - Cooler Master Technology Inc. AMD SR4 lamplight Control
 DEBUG eruption::hwdevices            > Found unsupported device: 0x2516:0x51 - Cooler Master Technology Inc. AMD SR4 lamplight Control
 DEBUG eruption::hwdevices            > Found unsupported device: 0x2516:0x51 - Cooler Master Technology Inc. AMD SR4 lamplight Control
 DEBUG eruption::hwdevices            > Found unsupported device: 0x1462:0x7c37 - MSI MYSTIC LIGHT 
 DEBUG eruption::hwdevices            > Found unsupported device: 0x28de:0x1142 - Valve Software Steam Controller
 DEBUG eruption::hwdevices            > Found unsupported device: 0x28de:0x1142 - Valve Software Steam Controller
 DEBUG eruption::hwdevices            > Found unsupported device: 0x28de:0x1142 - Valve Software Steam Controller
 DEBUG eruption::hwdevices            > Found unsupported device: 0x28de:0x1142 - Valve Software Steam Controller
 DEBUG eruption::hwdevices            > Found unsupported device: 0x28de:0x1142 - Valve Software Steam Controller
 DEBUG eruption::hwdevices            > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x3098 (0001:0005:00) - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption::hwdevices::roccat_vulcan_1xx > Bound driver: ROCCAT Vulcan 100/12x AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x41e:0x324d - Creative Technology Ltd Sound Blaster Play! 3
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x1e7d:0x2e4a - ROCCAT ROCCAT Tyon Black
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x1e7d:0x2e4a - ROCCAT ROCCAT Tyon Black
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x1e7d:0x2e4a - ROCCAT ROCCAT Tyon Black
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x1e7d:0x2e4a - ROCCAT ROCCAT Tyon Black
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: 136
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 DEBUG eruption                               > Registering filesystem watcher...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 DEBUG eruption                               > Entering the main loop now...
 INFO  eruption                               > Switching to profile: /home/sellerie/p1mod.profile
[... cut ...]
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 INFO  eruption                               > Closing all devices now...
 DEBUG eruption                               > Saving state...
 INFO  eruption                               > Exiting now
```

I attached the TRACE debug log as a file because I think it was too long for this input field breaking the read-flow of the issue entirely:
[eruption-trace.log](https://github.com/X3n0m0rph59/eruption/files/6846759/eruption-trace.log)

X3n0m0rph59, 2021-07-20:
@sellerie98 Thank you for the logs! Judging from the log dumps, I can see nothing being wrong with your setup.

I just managed to reproduce the behavior in a Fedora 34 VM with a local build of the `master` branch! I can see input lags of up to 500 milliseconds which leads to repeated key events and it just feels horrible. I suspect that Eruption has timing issues that have been introduced relatively recently.

I will try to bisect/debug it now...

sellerie98, 2021-07-20:
After a bisecting session I found that this is the first "bad" commit: 8c5889382511d6228451788660dec4abea3e6099

X3n0m0rph59, 2021-07-21:
@sellerie98 Thank you very much for bisecting this issue! That really helped a lot!

I have looked into it, and can confirm that simply reverting the commit 8c58893 already reduces the input-lag/jitter in my Fedora 34 VM.

So I have converted the code back to not use lockless datastructures anymore, and did some additional changes that may reduce the jitter and input-lag somewhat further. 

Could you please verify that the latest state of the `master` branch fixes the issue for you?

sellerie98, 2021-07-21:
@X3n0m0rph59 Yep this is working flawlessly now on Fedora. Thanks for fixing it <3

closed 2021-07-21 by sellerie98